openwrt/packages #19812: transmission: Couldn't save temporary file: Permission denied (variant.c:1235)

chirunosw, 2022-11-04:
Environment: Openwrt 22.03.2 X86

Description:
Transmission can't create and save stats.json and dht.dat, but setting.json can be created. Each time I restart transmission the session  stats would be cleared, altough it is harmless.

`
Fri Nov  4 22:35:44 2022 user.notice transmission: Starting with 6394528000 virt mem
Fri Nov  4 22:35:45 2022 daemon.err transmission-daemon[5774]: [2022-11-04 14:35:45.160] Couldn't save temporary file "/Home/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
Fri Nov  4 22:35:45 2022 daemon.err transmission-daemon[5774]: [2022-11-04 14:35:45.160] Couldn't save temporary file "/Home/transmission/dht.dat.tmp.XXXXXX": Permission denied (variant.c:1235)
Fri Nov  4 22:35:45 2022 daemon.err transmission-daemon[5774]: [2022-11-04 14:35:45.160] Couldn't save temporary file "/Home/transmission/stats.json.tmp.XXXXXX": Permission denied (variant.c:1235)
Fri Nov  4 22:35:47 2022 daemon.err transmission-daemon[27732]: [2022-11-04 14:35:47.230] Couldn't save temporary file "/Home/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
Fri Nov  4 22:36:09 2022 daemon.err transmission-daemon[27732]: [2022-11-04 14:36:09.230] Unable to create session lock file (2): No such file or directory (session-id.c:96)
`

`
root@OpenWrt:~# ls -l  /Home/transmission
drwxr-xr-x    2 transmis transmis      4096 Oct 31 01:49 blocklists
drwxr-xr-x    2 transmis transmis     12288 Nov  4 22:41 resume
-rw-r--r--    1 transmis transmis      1560 Nov  4 22:35 settings.json
drwxr-xr-x    2 transmis transmis     12288 Nov  3 22:36 torrents
`

M95D, 2022-11-21:
/Home/transmission directory permissions won't allow transmissionto create new files (no write permissions).

paper42, 2022-11-22:
What are the permissions of /Home? `ls -ld /Home`?

chirunosw, 2022-11-22:
> What are the permissions of /Home? `ls -ld /Home`?

```
ls -ld /Home
```

So I am confused.

M95D, 2022-11-22:
Could it be some jail / chroot / seccomp issue?

chirunosw, 2022-11-22:
> No, this should work. You could try switching your user to transmission and trying to touch that file.

How can I switch user on openwrt? I changed the main directory of transmission to /Home/transmission but still not work.

chirunosw, 2022-11-22:
> Could it be some jail / chroot / seccomp issue?

Idk but I saw others related similar problems to those packages. I'm not familiar with them.

shtefko, 2025-05-15:
> Could it be some jail / chroot / seccomp issue?

This is definitely a jail / chroot / seccomp issue. In the `/etc/init.d/transmission` there is jail section for the files and directories.

```shell
	procd_add_jail_mount "$config_file"
	procd_add_jail_mount_rw "$config_dir/resume"
	procd_add_jail_mount_rw "$config_dir/torrents"
	procd_add_jail_mount_rw "$config_dir/blocklists"
	procd_add_jail_mount_rw "$config_dir/stats.json"
	procd_add_jail_mount_rw "$download_dir"
``` 
Since there is no write permission for the `$config_dir`, no new files can be created in it. Transmission daemon would create several .tmp files here in normal operation. Beside those mentioned above, `blocklist.tmp` is also created here, during blocklist update. Perhaps there is no need to have such a stringent jail rules for the transmission config dir itself?

Workaround one could use is to add RW permission for the config dir here (in addition to or instead of existing rules)

```shell
	procd_add_jail_mount_rw "$config_dir"
```