openwrt/packages #19831: openvpn: Segmentation fault on server triggered by specially formed commonName in client certificate

mdmisch, 2022-11-05:
Maintainer: @mkrkn
Environment: mipsel_24kc, Netgear R6220, OpenWrt 22.03.2
```
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
root@OpenWrt:~# opkg list-installed | grep 'openssl\|openvpn'
libopenssl1.1 - 1.1.1q-1
luci-app-openvpn - git-22.025.78197-18b1130
openvpn-openssl - 2.5.7-3
root@OpenWrt:~# openvpn --version
OpenVPN 2.5.7 mipsel-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
library versions: OpenSSL 1.1.1q  5 Jul 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2022 OpenVPN Inc <sales@openvpn.net>
```

Description:
I have noticed that turning on logging at level 4 (`verb 4`) triggers Segmentation fault when one of client devices with specific commonName in its certificate connects. Without `verb 4` everything works as expected.

Steps to reproduce:

1. Create with OpenSSL:
    -  dhparams,
    -  secret shared keys for tls-crypt,
    -  dummy CA,
    -  server key,
    -  corresponding server certificate,
    -  client key,
    -  corresponding client certificate with subject like this: `subject=CN = "abcdef, ab. 1, qwerty"`,
2.  Run server with config like following:
```
persist-key
persist-tun
allow-compression no
ca ca.crt
cert server.crt
dev tun
dh dh4096.pem
ifconfig-pool-persist ipp.txt
keepalive 10 120
key server.key
max-clients 10
port 12345
proto udp
push "route 192.168.3.0 255.255.255.0"
server 10.1.2.0 255.255.255.0
status openvpn-status.log
tls-crypt ta.key
tls-version-min 1.3
user nobody
group nogroup
data-ciphers CHACHA20-POLY1305
remote-cert-tls client
tls-server

topology subnet

explicit-exit-notify 1
fast-io

verb 4
```
3. Make client connection with previously generated client certificate.
4. Stop server.
5. Start server.  

Failed (second) run. (wan IP address edited.)
```
root@OpenWrt:/tmp/openvpn-test/server# openvpn --config ./server.conf
2022-11-05 16:23:52 us=356020 --cipher is not set. Previous OpenVPN version defaulted to BF-CBC as fallback when cipher negotiation failed in this case. If you need this fallback please add '--data-ciphers-fallback BF-CBC' to your configuration and/or add BF-CBC to --data-ciphers.
2022-11-05 16:23:52 us=358329 OpenVPN 2.5.7 mipsel-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
2022-11-05 16:23:52 us=358791 library versions: OpenSSL 1.1.1q  5 Jul 2022, LZO 2.10
2022-11-05 16:23:52 us=360358 net_route_v4_best_gw query: dst 0.0.0.0
2022-11-05 16:23:52 us=361888 net_route_v4_best_gw result: via *.*.*.* dev wan
2022-11-05 16:23:52 us=367789 Diffie-Hellman initialized with 4096 bit key
2022-11-05 16:23:52 us=373649 Outgoing Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
2022-11-05 16:23:52 us=374253 Outgoing Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
2022-11-05 16:23:52 us=375147 Incoming Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
2022-11-05 16:23:52 us=376054 Incoming Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
2022-11-05 16:23:52 us=376986 TLS-Auth MTU parms [ L:1621 D:1156 EF:94 EB:0 ET:0 EL:3 ]
2022-11-05 16:23:52 us=380227 TUN/TAP device tun1 opened
2022-11-05 16:23:52 us=380864 do_ifconfig, ipv4=1, ipv6=0
2022-11-05 16:23:52 us=381414 net_iface_mtu_set: mtu 1500 for tun1
2022-11-05 16:23:52 us=382192 net_iface_up: set tun1 up
2022-11-05 16:23:52 us=387587 net_addr_v4_add: 10.1.2.1/24 dev tun1
2022-11-05 16:23:52 us=391734 Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:406 ET:0 EL:3 ]
2022-11-05 16:23:52 us=394039 Could not determine IPv4/IPv6 protocol. Using AF_INET
2022-11-05 16:23:52 us=394732 Socket Buffers: R=[180224->180224] S=[180224->180224]
2022-11-05 16:23:52 us=396068 UDPv4 link local (bound): [AF_INET]192.168.1.1:12345
2022-11-05 16:23:52 us=396842 UDPv4 link remote: [AF_UNSPEC]
2022-11-05 16:23:52 us=397883 GID set to nogroup
2022-11-05 16:23:52 us=398883 UID set to nobody
2022-11-05 16:23:52 us=399984 MULTI: multi_init called, r=256 v=256
2022-11-05 16:23:52 us=401515 IFCONFIG POOL IPv4: base=10.1.2.2 size=253
2022-11-05 16:23:52 us=402349 ifconfig_pool_read(), in='abcdef, ab. 1, qwerty,10.1.2.2,'
2022-11-05 16:23:52 us=403354 RESOLVE: Cannot parse IP address:  ab. 1: (Name does not resolve)
Segmentation fault
```

P.S.: I am filling this bug report here instead of upstream because I was not able to reproduce it with other builds of OpenVPN on another OS. In particular:
1. On Debian Bullseye:
```
$ dpkg-query -l openvpn
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  openvpn        2.5.1-3      amd64        virtual private network daemon
$ /usr/sbin/openvpn --version
OpenVPN 2.5.1 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on May 14 2021
library versions: OpenSSL 1.1.1n  15 Mar 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2018 OpenVPN Inc <sales@openvpn.net>
Compile time defines: enable_async_push=no enable_comp_stub=no enable_crypto_ofb_cfb=yes enable_debug=yes enable_def_auth=yes enable_dependency_tracking=no enable_dlopen=unknown enable_dlopen_self=unknown enable_dlopen_self_static=unknown enable_fast_install=needless enable_fragment=yes enable_iproute2=no enable_libtool_lock=yes enable_lz4=yes enable_lzo=yes enable_maintainer_mode=no enable_management=yes enable_multihome=yes enable_option_checking=no enable_pam_dlopen=no enable_pedantic=no enable_pf=yes enable_pkcs11=yes enable_plugin_auth_pam=yes enable_plugin_down_root=yes enable_plugins=yes enable_port_share=yes enable_selinux=no enable_shared=yes enable_shared_with_static_runtimes=no enable_silent_rules=no enable_small=no enable_static=yes enable_strict=no enable_strict_options=no enable_systemd=yes enable_werror=no enable_win32_dll=yes enable_x509_alt_username=yes with_aix_soname=aix with_crypto_library=openssl with_gnu_ld=yes with_mem_check=no with_sysroot=no
```
2. OpenVPN v2.5.7, built from [sources](https://github.com/OpenVPN/openvpn/tree/v2.5.7):
```
$ ./openvpn --version
OpenVPN 2.5.7 [git:HEAD/a0f9a3e9404c8321] x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Nov  5 2022
library versions: OpenSSL 1.1.1n  15 Mar 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2022 OpenVPN Inc <sales@openvpn.net>
Compile time defines: enable_async_push=no enable_comp_stub=no enable_crypto_ofb_cfb=yes enable_debug=yes enable_def_auth=yes enable_dlopen=unknown enable_dlopen_self=unknown enable_dlopen_self_static=unknown enable_fast_install=needless enable_fragment=yes enable_iproute2=no enable_libtool_lock=yes enable_lz4=yes enable_lzo=yes enable_management=yes enable_multihome=yes enable_pam_dlopen=no enable_pedantic=no enable_pf=yes enable_pkcs11=no enable_plugin_auth_pam=yes enable_plugin_down_root=yes enable_plugins=yes enable_port_share=yes enable_selinux=no enable_shared=yes enable_shared_with_static_runtimes=no enable_small=no enable_static=yes enable_strict=no enable_strict_options=no enable_systemd=no enable_werror=no enable_win32_dll=yes enable_x509_alt_username=no with_aix_soname=aix with_crypto_library=openssl with_gnu_ld=yes with_mem_check=no with_openssl_engine=auto with_sysroot=no
```

luizluca, 2022-11-06:
It looks like an upstream bug triggered by something special about OpenWrt. It would make it much easier to fix it (upstream) if you could isolated why it is crashing with OpenWrt. My guesses are the arch (mips), libc(musl) or a specific compiler hardening settings. At least for the arch, you could try OpenWrt in a VM.

mdmisch, 2022-11-18:
I have tested it in a x86 VM just now in this particular image: [openwrt-22.03.2-x86-64-generic-ext4-combined-efi.img.gz](https://downloads.openwrt.org/releases/22.03.2/targets/x86/64/openwrt-22.03.2-x86-64-generic-ext4-combined-efi.img.gz). It crashes in the same way with `openvpn-openssl`, `openvpn-mbedtls` and `openvpn-wolfssl`. `openvpn-wolfssl` required to change ciphers in config to start the server, by the way.