openwrt/packages #19856: docker container not accepting traffic from another container

dsl400, 2022-11-07:
# Issue template (remove lines from top till here)

Maintainer: @G-M0N3Y-2503
Environment:  x86/64 OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Description:

Mysql container not accepting connection from another container

    docker network create -d bridge mysql

Db container

    docker run --name mysql_db -d \
        --network mysql \
        -e MYSQL_ROOT_PASSWORD=pass \
        -e MYSQL_USER=user \
        -e MYSQL_PASSWORD=pass \
        -e MYSQL_DATABASE=database \
        -e MYSQL_ROOT_HOST=% \
        mariadb 

    $ docker logs mysql_db
    2022-11-06 21:46:01 0 [Note] mariadbd: ready for connections.
    Version: '10.9.3-MariaDB-1:10.9.3+maria~ubu2204'  socket: '/run/mysqld/mysqld.sock'  port: 3306  mariadb.org binary distribution

the second container is in the same network 

    $ php -r "new mysqli('mysql_db','root', 'pass','database');"

    $ PHP Warning:  mysqli::__construct(): (HY000/2002): Connection refused in Command line code on line 1

from the second container

    # ping mysql_db
    PING mysql_db (172.18.0.2): 56 data bytes
    ....
    --- mysql_db ping statistics ---
    6 packets transmitted, 0 packets received, 100% packet loss

got same results when using other mysql containers
tested same configuration on linux mint and everything works ok

G-M0N3Y-2503, 2022-11-07:
Unfortunatly, if you are using firewall4/nftables, docker doesn't work as easily as it did with firewall3/iptables.  
That is to say, If you are using firewall4, i'd imagine that if you disabled the iptables option in `/etc/config/dockerd` you should be able to configure the firewall and NAT etc within OpenWrt itself.

michaellmonaghan, 2023-08-25:
This is still an issue as of 23.05.0-rc2.

G-M0N3Y-2503, 2023-08-25:
https://github.com/openwrt/packages/issues/16818#issuecomment-1120403722

G-M0N3Y-2503, 2024-07-02:
https://github.com/openwrt/packages/issues/17766