openwrt/packages #19860: mwan3: mwan3rtmon unable to copy nhid route to target route table

kghost, 2022-11-07:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Format code blocks by wrapping them with pairs of ```
```

ACiDGRiM, 2023-01-23:
I'm experiencing this issue as well, I can't use ospf routed networks, however your patch doesn't seem to change anything as of 01/2023.
Are you still using this work around on 22.0.3 openwrt?

My work around is to just NAT traffic using masquerade on the upstream interfaces, not ideal to double nat.

kghost, 2023-01-23:
I'm using  OpenWrt 22.03.2, r19803-9a599fee93