openwrt/packages #19880: libassuan/libksba: build error against libgpg-error

feckert, 2022-11-11:
Maintainer: @neheb
Environment: x86_64, APU3, latest openwrt build

Description:
Since the following [commit](https://github.com/openwrt/packages/commit/b83c56576a79a8be01d5080f72cf540e17feb6a2) I have build error for the libs [libksba](https://github.com/openwrt/packages/blob/master/libs/libksba/Makefile) and [libassuan](https://github.com/openwrt/packages/blob/master/libs/libassuan/Makefile). I have remove the your commit I get an build error for [libgcrypt](https://github.com/openwrt/packages/blob/master/libs/libgcrypt/Makefile).

libksba:
```
...
checking for cc for build... gcc
checking if gcc supports -Wpointer-arith... yes
checking if gcc supports -Wno-pointer-sign... yes
checking whether the visibility attribute is supported... yes
checking for broken visibility attribute... no
checking for broken alias attribute... no
checking if gcc supports -fvisibility=hidden... yes
checking for gpg-error-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/host/bin/gpg-error-config
checking for gpgrt-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/host/bin/gpgrt-config
./configure: line 14388: test: : integer expression expected
./configure: line 14391: test: : integer expression expected
checking for GPG Error - version >= 1.8... no
configure: error: libgpg-error is needed.
                See ftp://ftp.gnupg.org/gcrypt/libgpg-error/ .
make[3]: *** [Makefile:75: /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/build_dir/target-x86_64_musl/libksba-1.6.2/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```
libassuan:
```
...
checking for an ANSI C-conforming const... yes
checking for inline... inline
checking for size_t... yes
checking for sys/socket.h... yes
checking for socklen_t... yes
checking for struct cmsghdr.cmsg_len... yes
checking for gpg-error-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/host/bin/gpg-error-config
checking for gpgrt-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/host/bin/gpgrt-config
./configure: line 14281: test: : integer expression expected
./configure: line 14284: test: : integer expression expected
checking for GPG Error - version >= 1.17... no
configure: error: libgpg-error was not found
make[3]: *** [Makefile:77: /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/build_dir/target-x86_64_musl/libassuan-2.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```
libgcrypt:
```
....
checking whether ARMv8 Crypto Extension support is requested... yes
checking whether PPC crypto support is requested... yes
checking whether a -O flag munging is requested... yes
checking whether a instrumentation (-fprofile, -fsanitize) munging is requested... yes
checking whether to enable AMD64 as(1) feature detection... yes
checking for gpg-error-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/gpg-error-config
checking for gpgrt-config... /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/host/bin/gpgrt-config
./configure: line 14658: /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/gpg-error-config: No such file or directory
./configure: line 14670: test: : integer expression expected
./configure: line 14673: test: : integer expression expected
checking for GPG Error - version >= 1.27... no
configure: error: libgpg-error is needed.
                See ftp://ftp.gnupg.org/gcrypt/libgpg-error/ .
make[2]: *** [Makefile:75: /home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/build_dir/target-x86_64_musl/libgcrypt-1.10.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-x86_64/build/feeds/packages/libs/libgcrypt'
time: package/feeds/packages/libgcrypt/compile#1.50#0.39#1.72
    ERROR: package/feeds/packages/libgcrypt failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/libgcrypt/compile] Error 1
make[1]: Leaving directory '/home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt'
make: *** [/home/feckert/workspace/openwrt/LDM-master-x86_64/build/openwrt/include/toplevel.mk:231: package/feeds/packages/libgcrypt/compile] Error 

```

hnyman, 2022-11-12:
Related discussions in #19828 and #19832 
Cc @neheb @flyn-org

egorenar, 2022-11-19:
And i'm facing this problem:
```
/usr/lib64/libgpg-error.so: file not recognized: file format not recognized
```
For some reason. **libksba** tries to link the host library.

closed 2022-11-19 by hnyman