openwrt/packages #19883: ddns-scripts: Can not detect local IP using network_get_ipaddr6

Eason0729, 2022-11-12:
Maintainer: @feckert 
Environment: ``mips 32bit`` ``TOTOLINK X5000R`` ``OpenWrt 22.03.2``

1. I followed [Option 1](https://openwrt.org/docs/guide-user/services/ddns/client#freednsafraidorg) and afraid.org-v2-token does not work(which always return ``Error 404 : Page not found``).
2. [Option 4 Prefer](https://openwrt.org/docs/guide-user/services/ddns/client#freednsafraidorg) work in ipv4, but it didn't work in ipv6 which log something like below.

```
Sat Nov 12 19:55:29 2022 user.err ddns-scripts[27706]: v6: Can not detect local IP using network_get_ipaddr6 'br-lan' - Error: '1' - TERMINATE
Sat Nov 12 19:55:29 2022 user.warn ddns-scripts[27706]: v6: PID '27706' exit WITH ERROR '1' at 2022-11-12 19:55
```

What should I place in field ``ip_network`` and ``interface`` in ``config.config.afraiddnsv6``?

part of ``/etc/config/ddns``
```
config service 'afraiddnsv6'
        option lookup_host '____'
        option update_url 'https://freedns.afraid.org/dynamic/update.php?[PASSWORD]&address=[ip]'
        option use_ipv6 '1'
        option enabled '1'
        option domain '____'
        option username 'non-empty'
        option password 'example_password'
        option use_https '1'
        option cacert 'IGNORE'
        option ip_source 'network'
        option use_syslog '2'
        option check_unit 'minutes'
        option force_unit 'minutes'
        option retry_unit 'seconds'
        option use_logfile '0'
 # I have changed this line frequently('pppoe-wan','wan','br-lan','wan6'), and none of those work
        option ip_network 'br-lan'
        option interface 'wan'
```

Part of ``ifconfig``
```
root@OpenWrt:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr 40:EE:15:B7:78:46
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2001:b011:8003:3917::1/64 Scope:Global
          inet6 addr: fe80::42ee:15ff:feb7:7846/64 Scope:Link
          inet6 addr: fdb8:ca0e:837e::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4197835 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6165752 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:648891300 (618.8 MiB)  TX bytes:6946856113 (6.4 GiB)

eth0      Link encap:Ethernet  HWaddr 40:EE:15:B7:78:46
          inet6 addr: fe80::42ee:15ff:feb7:7846/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:9993195 errors:62 dropped:0 overruns:0 frame:0
          TX packets:8481025 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7456872863 (6.9 GiB)  TX bytes:4504473807 (4.1 GiB)
          Interrupt:21

pppoe-wan Link encap:Point-to-Point Protocol
          inet addr:118.170.4.173  P-t-P:168.95.98.254  Mask:255.255.255.255
          inet6 addr: 2001:b011:8003:70f:59f1:4c5d:174a:c280/64 Scope:Global
          inet6 addr: fe80::59f1:4c5d:174a:c280/128 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:3718409 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2638308 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:4022118602 (3.7 GiB)  TX bytes:398547908 (380.0 MiB)

wan       Link encap:Ethernet  HWaddr 40:EE:15:B7:78:45
          inet6 addr: fe80::42ee:15ff:feb7:7845/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7264464 errors:0 dropped:490602 overruns:0 frame:0
          TX packets:4758451 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6942538288 (6.4 GiB)  TX bytes:794873812 (758.0 MiB)
```

In addition, I am using ``https://freedns.afraid.org/dynamic/update.php?[PASSWORD]&address=[ip]`` instead of ``afraid.org-keyauth`` ,which should be exactly the same.

Tabiskabis, 2023-02-12:
This is still an issue. ddns uses network_get_ipaddr to get the network's "first" IP6, even if it's not a global unicast address (usually a ULA). I've locally just fixed this by changing it to a new frankenstein function, network_get_ipaddr6gu. 
[image]

Spudz76, 2023-02-18:
Mine works fine as-is.  lookup_host and password (aka token) changed to protect the allegedly innocent.

```
config ddns "global"
        option ddns_dateformat "%F %R"
        option ddns_loglines "250"
        option upd_privateip "0"

config service "afraidOrg_ipv4"
        option service_name     "afraid.org-v2-token"
        option lookup_host      "spudz76.one-of-the-afraid-subdomains.com"
        option domain           "none"
        option username         "none"
        option password         "AAAbbbCCCblahblahblah123"
        option use_ipv6         "0"
        option interface        "wan"
        option ip_source        "network"
        option ip_network       "wan"
        option check_unit       "minutes"
        option force_unit       "minutes"
        option retry_unit       "seconds"
        option force_ipversion  "1"
        option enabled          "1"

config service "afraidOrg_ipv6"
        option service_name     "afraid.org-v2-token"
        option lookup_host      "spudz76.one-of-the-afraid-subdomains.com"
        option domain           "none"
        option username         "none"
        option password         "XXXyyyZZZblahblahblah456"
        option use_ipv6         "1"
        option interface        "wan6"
        option ip_source        "network"
        option ip_network       "wan6"
        option check_unit       "minutes"
        option force_unit       "minutes"
        option retry_unit       "seconds"
        option force_ipversion  "1"
        option enabled          "1"
```

I don't even have a `wan6` listed in the shell, but it appears to be an alias within luci.  Both external ip families are bound to `wan` according to the shell but it still works.

`interface` and `ip_network` appear to need to be the same device (or alias name).  Also don't use `domain` if  you are just using the "subdomains" free feature, only fill the `lookup_host`.

I get complaints about busybox nslookup not having force_ipversion support but it still works fine.

Eason0729, 2023-03-09:
I have switch to another dns provider and solved the problem.

closed 2023-04-18 by feckert