openwrt/packages #19903: Request updating 5 year old packages that are in OpenWRT

nsmith1024, 2022-11-15:
Hello,

The packages gphoto2 and libgphoto2 currently in OpenWRT are 5 years old. 

I wanted to request that someone upgrade these packages to the latest versions, because after 5 years a lot of bug fixes and advancements have been done that are not getting included with OpenWRT right now.

libgphoto2 also contains a bunch of sub packages also for various cameras, that also needs to be updated as well.

Please let me know if someone can do that.

Thanks

stangri, 2022-11-15:
@nsmith1024 you might have faster responses if you tag last contributors to these packages: @DocLM @neheb @champtar.

nsmith1024, 2022-11-15:
How do i tag someone?

BKPepe, 2022-11-16:
Hi @nsmith1024 ,

Anyone can update these packages, including you. Please check OpenWrt documentation on how you can prepare the build system and configure it for your device and then how you can compile packages. From your post, it needs to be clarified which router you are using and which OpenWrt version.

I checked the following packages:
- gphoto2 (in all OpenWrt supported versions), there is version 2.5.26, which was released 2 years ago. Not as old as you described, though.
  -  Same applies to libgphoto2

Because of the lack of details, I'm closing this.

closed 2022-11-16 by BKPepe