openwrt/packages #19923: tvheadend : master branch ramips compile error

sKyissKy, 2022-11-17:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
<tvheadend>
Environment: ramips ,master

make[3]: Entering directory '/home/sky/openwrt/feeds/packages/multimedia/tvheadend'
mkdir -p /home/sky/openwrt/dl
SHELL= flock /home/sky/openwrt/tmp/.tvheadend-2021-11-16.tar.xz.flock -c ' /home/sky/openwrt/scripts/download.pl "/home/sky/openwrt/dl" "tvheadend-2021-11-16.tar.xz" "1645e90b6b8f104f2749fb0911493010f7ae3176253f2a96a4d6094536207c03" "" || ( /home/sky/openwrt/scripts/dl_github_archive.py --dl-dir="/home/sky/openwrt/dl" --url="https://github.com/tvheadend/tvheadend.git" --version="2efe90cdcf74fdc4179692d283cf46c85e1cf681" --subdir="tvheadend-2021-11-16" --source="tvheadend-2021-11-16.tar.xz" --hash="1645e90b6b8f104f2749fb0911493010f7ae3176253f2a96a4d6094536207c03" || ( echo "Checking out files from the git repository..."; mkdir -p /home/sky/openwrt/tmp/dl && cd /home/sky/openwrt/tmp/dl && rm -rf tvheadend-2021-11-16 && [ ! -d tvheadend-2021-11-16 ] && git clone https://github.com/tvheadend/tvheadend.git tvheadend-2021-11-16 && (cd tvheadend-2021-11-16 && git checkout 2efe90cdcf74fdc4179692d283cf46c85e1cf681 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=cd tvheadend-2021-11-16 && git log -1 --format='\''@%ct'\'' && rm -rf tvheadend-2021-11-16/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c tvheadend-2021-11-16 | xz -zc -7e > /home/sky/openwrt/tmp/dl/tvheadend-2021-11-16.tar.xz && mv /home/sky/openwrt/tmp/dl/tvheadend-2021-11-16.tar.xz /home/sky/openwrt/dl/ && rm -rf tvheadend-2021-11-16; ); ) '
rm -f /home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/.built
touch /home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/.built_check
CFLAGS="-pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16:tvheadend-2021-11-16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include -O3 -flto -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/include -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/include -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/fortify -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include " CXXFLAGS="-pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16:tvheadend-2021-11-16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include -flto -O3 -flto -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/include -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/include -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/fortify -I/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/lib -L/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/lib -L/home/sky/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib -znow -zrelro -L/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/lib -L/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/lib -flto " make -C /home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16:tvheadend-2021-11-16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libiconv-stub/include -I/home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libintl-stub/include -O3 -flto" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" OS=Linux ;
make[4]: Entering directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make -f Makefile.webui LANGUAGES="ach ady ar bg cs da de en_US en_GB es et fa fi fr he hr hu it ko lv lt nl no pl pt ro ru sl sk sq sv tr uk zh zh-Hans" all
make[5]: Entering directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make -f Makefile.webui WEBUI=std compile-std
make[6]: Entering directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
WEBUI std finished
make[6]: Leaving directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make -f Makefile.webui WEBUI=debug compile-debug
make[6]: Entering directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
WEBUI debug finished
make[6]: Leaving directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make[5]: Leaving directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make[4]: *** No rule to make target '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/data/dvb-scan/dvb-c/mk-Telekabel-Sveti', needed by '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/build.linux/bundle.c'. Stop.
make[4]: Leaving directory '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16'
make[3]: *** [Makefile:262: /home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/.built] Error 2
make[3]: Leaving directory '/home/sky/openwrt/feeds/packages/multimedia/tvheadend'
time: package/feeds/packages/tvheadend/compile#1.07#0.08#1.12
make[2]: *** [package/Makefile:113: package/feeds/packages/tvheadend/compile] Error 2
make[2]: Leaving directory '/home/sky/openwrt'
make[1]: *** [package/Makefile:107: /home/sky/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sky/openwrt'
make: *** [/home/sky/openwrt/include/toplevel.mk:227：world] 错误 2

```
Format code blocks by wrapping them with pairs of ```
```

M95D, 2022-11-21:
Try ``make dirclean``` and then ```make``` again.
If that doesn't work, then there's probably something wrong with the DVB scan tables. Tvheadend downloads them each time it is built. Try disabling that option in ```make menuconfig```.
If it still doesn't work, please wait. I'm working on an update for the tvheadend package.

sKyissKy, 2022-11-21:
> Try `make dirclean` and then `make` again. If that doesn't work, then there's probably something wrong with the DVB scan tables. Tvheadend downloads them each time it is built. Try disabling that [...]

It looks like a problem with the tvheadend source code, because there is no such problem in other architectures (such as x86_64, aarch64), and the ramips compilation error. And if compiling with tvheadend (4.0.10) replacement for openwrt-19.07 passes.

M95D, 2022-11-21:
When the pull request for the current version was accepted, there was no such error, so the problem probably isn't in the tvheadend source code (https://github.com/tvheadend/tvheadend - that didn't change since), but in dvb scan tables (https://github.com/tvheadend/dtv-scan-tables). In OpenWrt package, we can only select tvheadend sources version. We don't have any control over which version of dvb-tables tvheadend uses when it is being built. It probably downloads the latest.

sKyissKy, 2022-11-21:
> New version ready. While testing it, I didn't encounter any errors such as the one you reported. #19952

some architectures  (such as x86_64, aarch64)  will not report errors, only the RAMIPS (Ralink MIPS) architecture will report this errors.  recommended that you test it carefully and do not blindly update the version, thank you.

sKyissKy, 2022-11-22:
> New version ready. While testing it, I didn't encounter any errors such as the one you reported. #19952

[004-fix-dvbscan-filenames.zip](https://github.com/openwrt/packages/files/10063567/004-fix-dvbscan-filenames.zip)

Even though I have patched dtv-scan-tables, I still get an error when compiling, the architecture is Ralink MIPS.

Error notes:

[Error notes.zip](https://github.com/openwrt/packages/files/10063598/Error.notes.zip)

M95D, 2022-11-22:
That's a very different error. No ideea why that happens.

I see you are using unicode paths: /home/sky/\�\�\�\�\�\�/lede/
Could you please start clean from your home path?
```
cd ~
git clone https://github.com/openwrt/openwrt/
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig (only select your router and tvheadend, no other changes)
make
```

sKyissKy, 2022-11-22:
> That's a very different error. No ideea why that happens.
> 
> I see you are using unicode paths: /home/sky/\�\�\�\�\�\�/lede/ Could you please start clean from your home path?
> 
> ```
> cd ~
> git clone https://github.com/openwrt/openwrt/
> cd openwrt
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> make menuconfig (only select your router and tvheadend, no other changes)
> make
> ```

same error too. lol...

[Error notes.zip](https://github.com/openwrt/packages/files/10065026/Error.notes.zip)

M95D, 2022-11-22:
I can't reproduce it. It builds just fine on my system.
```
marius95@GRAPHIM /build/openwrt $ ls build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/ipkg-mipsel_24kc/tvheadend/usr/bin/
total 12420
drwxr-xr-x 1 marius95 users       18 2022-11-22 08:43 .
drwxr-xr-x 1 marius95 users        6 2022-11-22 08:43 ..
-rwxr-xr-x 1 marius95 users 12715319 2022-11-22 08:43 tvheadend
```

sKyissKy, 2022-11-22:
> I can't reproduce it. It builds just fine on my system.
>
> ```
> marius95@GRAPHIM /build/openwrt $ ls build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/ipkg-mipsel_24kc/tvheadend/usr/bin/
> total [...]

I think my gcc version is too low

M95D, 2022-11-22:
But OpenWrt builds it's own toolchain before it starts building any packages, and it's absolutely normal for a 64bit (x86-64) toolchain to generate executable code for MIPS 32bit. It's called cross-compilation. So it doesn't make any sense.

Are you sure your system meets all the [requirements](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem)?

sKyissKy, 2022-11-24:
> But OpenWrt builds it's own toolchain before it starts building any packages, and it's absolutely normal for a 64bit (x86-64) toolchain to generate executable code for MIPS 32bit. It's called [...]

ubuntu 22.04 LTS

sKyissKy, 2022-11-25:
> make[4]: *** No rule to make target '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/data/dvb-scan/dvb-c/mk-Telekabel-Sveti', needed by '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/build.linux/bundle.c'. Stop.

I checked carefully and selected the 'Include DVB-scan tables' option, and the following error will be reported:
make[4]: *** No rule to make target '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/data/dvb-scan/dvb-c/mk-Telekabel-Sveti', needed by '/home/sky/openwrt/build_dir/target-mipsel_24kc_musl/tvheadend-2021-11-16/build.linux/bundle.c'. Stop.

If I select the 'Optimize for speed' option, the following error will be reported:
tvheadend-2022-11-20/src/atomic.h:66: undefined reference to `__sync_fetch_and_add_8'
tvheadend-2022-11-20/src/atomic.h:238: undefined reference to `__sync_lock_test_and_set_8'
tvheadend-2022-11-20/src/atomic.h:189: undefined reference to `__sync_fetch_and_sub_8'

neheb, 2022-11-30:
@sKyissKy missing -latomic

M95D, 2022-11-30:
If you can fix it better, please make a pull request. I'm ok with it. My developer skills are very minimal.

M95D, 2022-12-05:
Fixed.

closed 2022-12-05 by sKyissKy