openwrt/packages #19925: acme: never renews certificates

brackenhill-mob, 2022-11-17:
Maintainer: @tohojo 
Environment: x86, x86_64, OpenWrt SNAPSHOT r21168-c3322cf04a

I have to use SNAPSHOT due to issue [19718](https://github.com/openwrt/packages/issues/19718) which I raised.

Description:

/usr/lib/acme/hook calls the actual acme.sh script (line 42) before it has processed any of the configuration switches passed to it via environment variables, so none of my settings are used. I checked this by inserting "set -x" as line 2 in the hook script. Below is the output of running ```acme get``` (my domain replaced with hostname.example.com)

```
+ set -u
+ ACME=/usr/lib/acme/client/acme.sh
+ LOG_TAG=acme-acmesh
+ WEBROOT=/var/run/acme/challenge
+ NOTIFY=/usr/lib/acme/notify
+ . /usr/lib/acme/functions.sh
+ export 'CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt'
+ export 'NO_TIMESTAMP=1'
+ set --
+ '[' 1 '=' 1 ]
+ set -- --debug
+ domain_dir=/etc/acme/hostname.example.com
+ log info 'Running ACME for hostname.example.com'
+ prio=info
+ shift
+ '[' info '!=' debug ]
+ logger -t acme-acmesh -s -p daemon.info -- 'Running ACME for hostname.example.com'
acme-acmesh: Running ACME for hostname.example.com
+ '[' -e /etc/acme/hostname.example.com ]
+ '[' 0 '=' 0 ]
+ grep -q acme-staging /etc/acme/hostname.example.com/hostname.example.com.conf
+ set -- --debug --renew --home /etc/acme -d hostname.example.com
+ log info '/usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d hostname.example.com'
+ prio=info
+ shift
+ '[' info '!=' debug ]
+ logger -t acme-acmesh -s -p daemon.info -- '/usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d hostname.example.com'
acme-acmesh: /usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d hostname.example.com
+ trap '$NOTIFY renew-failed;exit 1' INT
+ /usr/lib/acme/client/acme.sh --debug --renew --home /etc/acme -d hostname.example.com
Lets find script dir.
_SCRIPT_='/usr/lib/acme/client/acme.sh'
_script='/usr/lib/acme/client/acme.sh'
_script_home='/usr/lib/acme/client'
Using config home:/etc/acme
Running cmd: renew
Using config home:/etc/acme
default_acme_server
ACME_DIRECTORY='https://acme.zerossl.com/v2/DV90'
DOMAIN_PATH='/etc/acme/hostname.example.com'
Le_API='https://acme-v02.api.letsencrypt.org/directory'
Using config home:/etc/acme
ACME_DIRECTORY='https://acme-v02.api.letsencrypt.org/directory'
_init api for server: https://acme-v02.api.letsencrypt.org/directory
Retrying GET
GET
url='https://acme-v02.api.letsencrypt.org/directory'
timeout=
displayError='1'
_CURL='curl --silent --dump-header /etc/acme/http.header  -L '
ret='0'
_hcode='0'
ACME_KEY_CHANGE='https://acme-v02.api.letsencrypt.org/acme/key-change'
ACME_NEW_AUTHZ
ACME_NEW_ORDER='https://acme-v02.api.letsencrypt.org/acme/new-order'
ACME_NEW_ACCOUNT='https://acme-v02.api.letsencrypt.org/acme/new-acct'
ACME_REVOKE_CERT='https://acme-v02.api.letsencrypt.org/acme/revoke-cert'
ACME_AGREEMENT='https://letsencrypt.org/documents/LE-SA-v1.3-September-21-2022.pdf'
ACME_NEW_NONCE='https://acme-v02.api.letsencrypt.org/acme/new-nonce'
+ status=2
+ trap - INT
+ exit
```

My /etc/config/acme file (anonymized) is

```
config acme
	option state_dir '/etc/acme'
	option account_email 'me@example.com'
	option debug '1'

config cert 'Router'
	option keylength '2048'
	list domains 'hostname.example.com'
	option update_uhttpd '0'
	option validation_method 'dns'
	option dns 'dns_cf'
	list credentials 'CF_Account_ID='\''00000000000000000000000000'\'''
	list credentials 'CF_Token='\''000000000000000000000000000000000'\'''
	option enabled '1'
	option update_nginx '1'
	option use_staging '0'
	option days '60'
```

As you can see from the log above, no call to acme.sh with --days or --dns

Happy to test anything. I've got a bunch of domains that need renewing within the next 30 days or less...

tohojo, 2022-11-17:
@hgl PTAL

hgl, 2022-11-17:
`--days` or `--dns` is only used when issuing a new certificate, that's by acme.sh's design. Check `/etc/acme/<domain>_ecc/<domain>.conf` to see when it should be renewed.

brackenhill-mob, 2022-11-17:
I don't have an /etc/acme/<domain>_ecc/ folder. I've deleted all the acme apps and reinstalled, and it isn't created

The <domain>.conf in /etc/acme/<domain> folder contains
```
Le_Domain='hostname.example.com'
Le_Alt='no'
Le_Webroot='dns_cf'
Le_PreHook='__ACME_BASE64__START_QUNUSU9OPXByZXBhcmUgaG90cGx1Zy1jYWxsIGFjbWU=__ACME_BASE64__END_'
Le_PostHook=''
Le_RenewHook='__ACME_BASE64__START_QUNUSU9OPXJlbmV3ZWQgaG90cGx1Zy1jYWxsIGFjbWU=__ACME_BASE64__END_'
Le_API='https://acme-v02.api.letsencrypt.org/directory'
Le_Keylength='2048'
Le_OrderFinalize='https://acme-v02.api.letsencrypt.org/acme/finalize/721651387/123360168247'
Le_LinkOrder='https://acme-v02.api.letsencrypt.org/acme/order/721651387/123360168247'
Le_LinkCert='https://acme-v02.api.letsencrypt.org/acme/cert/04e335e4b985af50cd5e8bb99b3b6753a7f6'
Le_CertCreateTime='1664752127'
Le_CertCreateTimeStr='2022-10-02T23:08:47Z'
Le_NextRenewTimeStr='2022-11-30T23:08:47Z'
Le_NextRenewTime='1669849727'
```

I've just spotted from this that a) the domain has expired and b) there is no ```Le_RenewalDays=``` line.

I'll delete the domain and start afresh

closed 2022-12-02 by brackenhill-mob