openwrt/packages #19930: ddns-scripts-nsupdate - Script Should Allow for Custom Port Number (but does not)

Terrawide, 2022-11-17:
Maintainer: Michal Vasilek, AKA paper42 
Environment: OpenWRT 22.03.0 r19685-512e76967f (Linksys WRT3200ACM, Marvell mvebu CortexA9)
Source: feeds/packages/net/ddns-scripts
Section: net
Package Version: 2.8.2-25 (It could be solved in other versions, past and future, but this one has the issue)

Description: Neither the LuCI GUI or editing the /etc/config/ddns File allow for including a custom Port Number for the DNS Server.

Why?: Because there is 'sanitation code' in the dynamic_dns_functions.sh file on line 74 in the form of a Regular Expression Variable definition that prevents this from being done.

The LuCI GUI makes no mention of being able to configure a custom port.  However, the dynamic_dns_lucihelper.sh File includes the following line, which implies adding a custom Port Number is allowable or intended or used to be a feature, etc.;

```
 -d DNS-SERVER       => dns_server=SERVER[:PORT]
```

But as noted above, the 'sanitation code' does not allow for a Port Number to be added after the IP Address or Host Name of the DNS Server.

Good News!  The underlying BIND / NAMED command (NSUPDATE) that the script uses, does allow for a custom port number to be used, so this should be easy to implement.  This is from the NSUPDATE help information: server address [port]     (set primary server for zone).

Possible Solution?

The current code for /etc/lib/ddns/dynamic_dns_functions.sh Line 74 is below;
```
DNS_CHARSET="[@a-zA-Z0-9._-]"
```

Suggested Change would be to modify the Regular Expression to allow for a single space between the IP Address OR Host and a Port Number;
```
DNS_CHARSET=[@a-zA-Z0-9]*([ ][0-9._-]*)
```

Important Note: As noted above the dynamic_dns_lucihelper.sh File specifies: SERVER[:PORT]  That should be changed to SERVER [PORT]  The simplest solution seems to be to use a space since NSUPDATE uses a space instead of a colon.  Even though the convention is to use a colon as a separator between the SERVER and [:PORT], there would have to be additional code written to change the colon to a space for the sake of NSUPDATE.

I'm also assuming since it's a script rather than a compile binary file that the package is the same across all distribution of OpenWRT, so is hopefully a quick fix.  This is implied in the package file with this comment in the control section: Architecture: all

And finally: Why make this change? / Why would it be helpful?: There are several internet providers like Comcast X-Finity that impose a 'transparent' DNS proxy service that prevents NSUPDATE, and thus the OpenWRT DDNS Scripts from passing updates to a DNS server like BIND / NAMED.  Without going into detail, the proxy service strips out the TSIG information NSUPDATE sends so when it arrives at the destination DNS server, the update or change request is rejected.

Example(s) of some tests of this 'transparent' DNS proxy service are below;

Attempted DDNS update via NSUPDATE via TCP port 53 through Comcast internet;

;; TSIG PSEUDOSECTION:
ddns.                   0       ANY     TSIG    hmac-md5.sig-alg.reg.int. 1668702339 300 16 vG1mnvKCwArp+t3IFL5NbQ== 32700 NOERROR 0 

; TSIG error with server: expected a TSIG or SIG(0)

Reply from update query:
;; ->>HEADER<<- opcode: UPDATE, status: REFUSED, id:  32700
;; flags: qr ra; ZONE: 0, PREREQ: 0, UPDATE: 0, ADDITIONAL: 0
Answer:
;; ->>HEADER<<- opcode: UPDATE, status: REFUSED, id:  32700
;; flags: qr ra; ZONE: 0, PREREQ: 0, UPDATE: 0, ADDITIONAL: 0

Attempted DDNS update via NSUPDATE via TCP port 5353 through Comcast internet;

;; TSIG PSEUDOSECTION:
ddns.                   0       ANY     TSIG    hmac-md5.sig-alg.reg.int. 1668702432 300 16 U2Zm8vhayWBDSKAB1yJNlg== 50001 NOERROR 0 

Answer:
;; ->>HEADER<<- opcode: UPDATE, status: NOERROR, id:  50001
;; flags: qr; ZONE: 1, PREREQ: 0, UPDATE: 0, ADDITIONAL: 1
;; ZONE SECTION:
;fsddns.us.                     IN      SOA

;; TSIG PSEUDOSECTION:
ddns.                   0       ANY     TSIG    hmac-md5.sig-alg.reg.int. 1668702432 300 16 U2Zm8vhayWBDSKAB1yJNlg== 50001 NOERROR 0 

In conclusion, I believe this small modification of the Regular Expression that 'sanitizes' the DNS Server ( DNS-SERVER / dns_server ) Variable(s) and allowing for a custom DNS Port Number to be included could save a lot people from a lot of wasted time.

LorbusChris, 2023-09-18:
Dupe of https://github.com/openwrt/packages/issues/17652?

feckert, 2023-09-18:
Yes this is a dupe. See my comments I made recently https://github.com/openwrt/packages/issues/17652#issuecomment-1722808190

closed 2023-09-18 by feckert