openwrt/packages #19941: libxslt: Could NOT find Iconv (missing: Iconv_LIBRARY)

M95D, 2022-11-20:
OpenWrt git master. Building for Linksys WRT-1900ACS v2 (mvebu). libiconv-full is selected with **<*>**.
I get this error while building libxslt:
```
touch /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/stamp/.iconv_installed
make[3]: Leaving directory '/build/openwrt.a/package/libs/libiconv-full'
time: package/libs/libiconv-full/compile#0.25#0.03#0.25
make[3]: Entering directory '/build/openwrt.a/feeds/packages/libs/libxml2'
mkdir -p /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/stamp
SHELL= flock /build/openwrt.a/tmp/.root-copy.flock -c 'cp -fpR /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxml2-2.10.3/.pkgdir/libxml2/. /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/'
touch /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/stamp/.libxml2_installed
echo "libxml2" >> /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/pkginfo/libxml2.default.install
make[3]: Leaving directory '/build/openwrt.a/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/compile#0.25#0.02#0.23
make[3]: Entering directory '/build/openwrt.a/feeds/packages/libs/libxslt'
rm -f /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/.configured_*
rm -f /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.libxslt_installed
mkdir -p /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/openwrt-build
(cd /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/openwrt-build; CFLAGS="-O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37=libxslt-1.1.37 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/include -I/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/include " CXXFLAGS="-O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37=libxslt-1.1.37 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/include -I/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/include " LDFLAGS="-L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/usr/lib -L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/lib -znow -zrelro -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=arm -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="/build/openwrt.a/staging_dir/host/bin/ccache" -DCMAKE_C_COMPILER="arm-openwrt-linux-muslgnueabi-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="/build/openwrt.a/staging_dir/host/bin/ccache" -DCMAKE_CXX_COMPILER="arm-openwrt-linux-muslgnueabi-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="/build/openwrt.a/staging_dir/host/bin/ccache" -DCMAKE_ASM_COMPILER="arm-openwrt-linux-muslgnueabi-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/usr/lib -L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/lib -znow -zrelro -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/usr/lib -L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/lib -znow -zrelro -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/usr/lib -L/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/lib -znow -zrelro -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libiconv-full/lib -L/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/libintl-full/lib -Wl,-Bsymbolic-functions" -DCMAKE_AR="/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-ar" -DCMAKE_NM="/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-nm" -DCMAKE_RANLIB="/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr;/build/openwrt.a/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-12.2.0_musl_eabi" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi -DCMAKE_PREFIX_PATH=/build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -DBUILD_SHARED_LIBS=ON -DLIBXSLT_WITH_DEBUGGER=OFF -DLIBXSLT_WITH_CRYPTO=OFF -DLIBXSLT_WITH_MEM_DEBUG=OFF -DLIBXSLT_WITH_MODULES=OFF -DLIBXSLT_WITH_PROFILER=ON -DLIBXSLT_WITH_PYTHON=OFF -DLIBXSLT_WITH_TESTS=OFF -DLIBXSLT_WITH_THREADS=ON -DLIBXSLT_WITH_XSLT_DEBUG=OFF /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37 )
CMake Error at /build/openwrt.a/staging_dir/host/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Iconv (missing: Iconv_LIBRARY)
Call Stack (most recent call first):
  /build/openwrt.a/staging_dir/host/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /build/openwrt.a/staging_dir/host/share/cmake-3.24/Modules/FindIconv.cmake:165 (find_package_handle_standard_args)
  /build/openwrt.a/staging_dir/host/share/cmake-3.24/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /build/openwrt.a/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/cmake/libxml2/libxml2-config.cmake:54 (find_dependency)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/openwrt-build/CMakeFiles/CMakeOutput.log".
See also "/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/openwrt-build/CMakeFiles/CMakeError.log".
make[3]: *** [Makefile:165: /build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libxslt-1.1.37/.configured_889556d2f423f99e091beece9c8d870a] Error 1
```
libxslt maintainer: Jiri Slachta <jiri@slachta.eu>
recently updated by @neheb

Thanks.

neheb, 2022-11-20:
Could have sworn I tested this. The issue is that cmake's FindIconv.cmake needs fixing to be compatible with libiconv-full as implemented by OpenWrt. I may be able to solve this without patching tools/cmake.

M95D, 2022-11-21:
Please undo and return to the previous version, so there won't be any rush to fix it.

neheb, 2022-11-21:
quick fix:

```
--- a/libs/libxslt/Makefile
+++ b/libs/libxslt/Makefile
@@ -88,6 +88,12 @@ CMAKE_OPTIONS += \
        -DLIBXSLT_WITH_THREADS=ON \
        -DLIBXSLT_WITH_XSLT_DEBUG=OFF

+ifeq ($(CONFIG_BUILD_NLS),y)
+       CMAKE_OPTIONS += \
+               -DIconv_INCLUDE_DIR=$(ICONV_PREFIX)/include \
+               -DIconv_LIBRARY=$(ICONV_PREFIX)/lib
+endif
+
 define Build/InstallDev/Xslt
        $(INSTALL_DIR) $(1)/usr/bin $(2)/bin $(1)/usr/include/libxslt \
                $(1)/usr/include/libexslt $(1)/usr/lib \
```

M95D, 2022-11-21:
That was quick! I'll test it tomorrow.

neheb, 2022-11-21:
@Youw do you know of a way to fix FindIconv.cmake ? I looked at it and saw nothing obvious.

Youw, 2022-11-21:
"Fixing" `FindIconv.cmake` would be a dirty and non-trivial workaround on CMake side. The root cause of this "situation" is the fact that `find_file`/`find_library` doesn't respect `CFLAGS`/`LDFLAGS`, and that is not something that could be just easily "fixed" on CMake side (at least I think so).

On the other hand, CMake is very specific on how the cross-compilation should be handled, and I believe in this case a better way would be to have something like (simplified):
```Makefile
CMAKE_OPTIONS += -DCMAKE_PREFIX_PATH="$(ICONV_PREFIX)"
```
I.e.: for CMake builds, instead of (or in addition to) appending `CFLAGS`/`LDFLAGS`, the build system better be appending a semicolon-separated `CMAKE_PREFIX_PATH` CMake variable.
I'd apply this suggestion for all the libraries and it would fix all similar issues with CMake that openwrt has right now.

neheb, 2022-11-22:
@Youw still errors. Would be interesting to add these CMAKE_OPTIONS to nls.mk.

Another potential fix is to switch to the libxslt meson wrap. Although that probably needs updating.

Youw, 2022-11-22:
> still errors

Care to share logs? I don't mind to take a look.

> Would be interesting to add these CMAKE_OPTIONS to nls.mk.

Can't say I know what `nls.mk` is.

neheb, 2022-11-22:
Unrelated. Issue was different.

Anyway,

```
CMAKE_OPTIONS += -DCMAKE_PREFIX_PATH="$(ICONV_PREFIX)"
```

actually seems to have worked.

closed 2022-11-27 by M95D