openwrt/packages #19942: tinyproxy - Configuring upstream causes config file syntax error

unregist, 2022-11-20:
Maintainer: @jow- 
Environment: Clean Install, 22.03.2 built with imagebuilder. SamKnows WB8.
```make image PROFILE=samknows_whitebox-v8 PACKAGES="luci luci-ssl luci-app-uhttpd luci-app-openvpn openvpn-openssl tcpdump luci-app-wireguard wireguard-tools luci-app-mwan3 mwan3 luci-app-tinyproxy tinyproxy" ```

Description: tinyproxy - Configuring upstream causes config file syntax error

When you configure an upsteam proxy in uci/luci the tinyproxy.conf file generated has a syntax error.

UCI Config:
```
root@OpenWRT:~# uci export tinyproxy
package tinyproxy

config tinyproxy
        option User 'nobody'
        option Group 'nogroup'
        option Port '8888'
        option Timeout '600'
        option DefaultErrorFile '/usr/share/tinyproxy/default.html'
        option StatFile '/usr/share/tinyproxy/stats.html'
        option LogFile '/var/log/tinyproxy.log'
        option LogLevel 'Info'
        option MaxClients '100'
        option MinSpareServers '5'
        option MaxSpareServers '20'
        option StartServers '10'
        option MaxRequestsPerChild '0'
        option ViaProxyName 'tinyproxy'
        list ConnectPort '443'
        list ConnectPort '563'
        list Allow '127.0.0.1'
        list Allow '192.168.204.0/22'
        option enabled '1'

config upstream
        option type 'proxy'
        option via '172.20.1.175:3128'
```

Generated config file:
```
root@OpenWRT:~# cat /tmp/etc/tinyproxy.conf
### AUTOGENERATED CONFIGURATION
### DO NOT EDIT
### SEE /etc/config/tinyproxy INSTEAD

User nobody
Group nogroup
Port 8888
Timeout 600
DefaultErrorFile "/usr/share/tinyproxy/default.html"
StatHost "127.0.0.1"
StatFile "/usr/share/tinyproxy/stats.html"
LogFile "/var/log/tinyproxy.log"
Syslog Off
LogLevel Info
XTinyproxy Off
MaxClients 100
MinSpareServers 5
MaxSpareServers 20
StartServers 10
MaxRequestsPerChild 0
Allow 127.0.0.1
Allow 192.168.204.0/22
ViaProxyName "tinyproxy"
FilterURLs Off
FilterExtended Off
FilterCaseSensitive Off
FilterDefaultDeny No
ConnectPort 443
ConnectPort 563
upstream 172.16.1.1:3128 
```

At this point tinyproxy refuses to start:
```
root@OpenWRT:~# tinyproxy -c /tmp/etc/tinyproxy.conf
Syntax error on line 30
Unable to parse config file. Not starting.
```

Correcting the last line from ```upstream 172.16.1.1:3128 ``` to ```upstream http 172.16.1.1:3128 ``` fixes the problem.
Unfortunately the .conf file gets overrwitten in by the /etc/init.d/tinyproxy script so the fix is only useable for testing.

Fix (I think)
``` /etc/init.d/tinyproxy ``` line 27 to be updated from ``` echo "upstream $via$target" ``` to ``` echo "upstream http $via$target" ```

danielkucera, 2024-04-13:
Hi @jow- ,
I am having similar issue, following change did the tricky:
```
        [ "$type" = "http" ] && [ -n "$via" ] && \                    
                echo "upstream http $via$target"  
```
It seems like the upstream section could use some refactoring for current tinyproxy version.

heyzling, 2024-04-27:
"reject" branch is also broken.
Should be
```bash
[ "$type" = "reject" ] && [ -n "$target" ] && \
                echo "upstream none$target"
```

Example is here: https://github.com/tinyproxy/tinyproxy/blob/1.10.0/etc/tinyproxy.conf.in#L158

danielkucera, 2024-04-29:
> "reject" branch is also broken. Should be
> 
> ```shell
> [ "$type" = "reject" ] && [ -n "$target" ] && \
>                 echo "upstream none$target"
> ```
> 
> Example is here: https://github.com/tinyproxy/tinyproxy/blob/1.10.0/etc/tinyproxy.conf.in#L158

Thanks, I've added that.