openwrt/packages #19949: privoxy: postinst script is wrong

M95D, 2022-11-21:
OpenWrt git master, Linksys WRT-1900ACSv2 (mvebu)
```
/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/lib/opkg/info/privoxy.postinst-pkg: line 7: unexpected EOF while looking for matching `"'
/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/lib/opkg/info/privoxy.postinst-pkg: line 8: syntax error: unexpected end of file
postinst script ./usr/lib/opkg/info/privoxy.postinst has failed with exit code 2
```
That file contains this:
```
        #!/bin/sh
        # if exists, update previous version entry missing protocol
        (grep -E "privoxy.*8118 ${IPKG_INSTROOT}/etc/services) > /dev/null && sed -i "s/privoxy.*8118/privoxy\t\t8118\/tcp/" ${IPKG_INSTROOT}/etc/services

        # add missing tcp and udp entries
        (grep -E "privoxy.*8118\/tcp" ${IPKG_INSTROOT}/etc/services) > /dev/null || echo -e "privoxy\t\t8118/tcp" >> ${IPKG_INSTROOT}/etc/services
        (grep -E "privoxy.*8118\/udp" ${IPKG_INSTROOT}/etc/services) > /dev/null || echo -e "privoxy\t\t8118/udp" >> ${IPKG_INSTROOT}/etc/services
```
Note that there is a missing quote in the first grep search pattern.
I think it's caused by a $ instead of $$ in Makefile:
```
(grep -E "privoxy.*8118$" $${IPKG_INSTROOT}/etc/services) > /dev/null \
```
Modified by: @powersj  - please test your pull requests.
Commited by: @neheb  - please don't commit untested pull requests.

There was an error in the github automated tests. I don't understand why they didn't fail.
```
Signature check passed.
Testing package privoxy in version 3.0.33 from privoxy
Installing privoxy (3.0.33-2) to root...
//usr/lib/opkg/info/privoxy.postinst: /usr/lib/opkg/info/privoxy.postinst-pkg: line 8: syntax error: unterminated quoted string
2022-11-01 13:06:26.689 7ff81748ab48 Fatal error: can't open pid file '/var/run/privoxy.pid': Invalid argument
Collected errors:
 * pkg_run_script: package "privoxy" postinst script returned status 2.
 * opkg_configure: privoxy.postinst returned 2.
```

powersj, 2022-11-21:
> There was an error in the github automated tests. I don't understand why they didn't fail.

Having tests not fail is unfortunate as I do not use OpenWRT, I opened the PR to help resolve an issue reported upstream to us with Telegraf. 

>         (grep -E "privoxy.*8118 ${IPKG_INSTROOT}/etc/services) > /dev/null && sed -i "s/privoxy.*8118/privoxy\t\t8118\/tcp/" ${IPKG_INSTROOT}/etc/services

It looks like the first grep's final double quote is missing from what is put down. As you suggested I assume this has to do with how the '$' is handled?

I put up https://github.com/openwrt/packages/pull/19951 as a draft that uses a double `$$`. It looks like the test does not fail in the same way. Can you confirm?

Thanks!

M95D, 2022-11-21:
Doubling the $ works. I already made the change on my system.

M95D, 2022-11-21:
I looked at the automated checks and they still report errors, this time with the preinst script.

PR #19951 (2022-11-26): privoxy: fix postinst missing $
fixes: #19949

Maintainer:
Compile tested: (put here arch, model, OpenWrt version)
Run tested: (put here arch, model, OpenWrt version, tests done)

Description:
commit 46e4def6 (2022-11-26): privoxy: fix postinst missing $

fixes: #19949
commit 6d14306a (2022-12-03): privoxy: fix postinst missing $

fixes: #19949

(cherry picked from commit 46e4def61f6b670daac1725df8f3fc4475fe5450)
closed 2022-11-26 by neheb