openwrt/packages #19953: github automated checks don't fail when they should

M95D, 2022-11-21:
Please have a look at this pull request:
https://github.com/openwrt/packages/pull/19694/files
and it's automated check:
https://github.com/openwrt/packages/actions/runs/3325614527/jobs/5589351307

1)
Notice that the pull request only changed the post-install script. I don't see the package being built by the automated check. It looks like it skipped it. On my system, package/install fails with this error, which the automated check should have detected:
```
/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/lib/opkg/info/privoxy.postinst-pkg: line 7: unexpected EOF while looking for matching `"'
/build/openwrt.a/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/usr/lib/opkg/info/privoxy.postinst-pkg: line 8: syntax error: unexpected end of file
postinst script ./usr/lib/opkg/info/privoxy.postinst has failed with exit code 2
make[2]: *** [package/Makefile:73: package/install] Error 1
```

2)
The automated check tried to install the generated package ("Test via Docker container"). There was an error in that test, but the test didn't fail.
```
Testing package privoxy in version 3.0.33 from privoxy
Installing privoxy (3.0.33-2) to root...
//usr/lib/opkg/info/privoxy.postinst: /usr/lib/opkg/info/privoxy.postinst-pkg: line 8: syntax error: unterminated quoted string
2022-11-01 13:06:26.689 7ff81748ab48 Fatal error: can't open pid file '/var/run/privoxy.pid': Invalid argument
Collected errors:
 * pkg_run_script: package "privoxy" postinst script returned status 2.
 * opkg_configure: privoxy.postinst returned 2.
```

M95D, 2022-11-22:
Another serious error during an automated test that didn't fail:
https://github.com/openwrt/packages/actions/runs/3516729022/jobs/5893660748
```
Collected errors:
 * pkg_run_script: package "privoxy" preinst script returned status 1.
 * preinst_configure: Aborting installation of privoxy.
 * opkg_install_cmd: Cannot install package privoxy.
```

BKPepe, 2022-11-22:
Ping @aparcar