openwrt/packages #19969: shadowsocks-libev: ss-rules need auto-merge flag

4ft35t, 2022-11-24:
when moving from ipset version, got error
```c
/etc/init.d/shadowsocks-libev start
In file included from /tmp/ssrules.nft.chk:1:17-43:
/tmp/ssrules.nft:34:3-16: Error: conflicting intervals specified
                223.119.0.0/16,
                ^^^^^^^^^^^^^^
In file included from /tmp/ssrules.nft.chk:1:17-43:
/tmp/ssrules.nft:29:5-23: Error: Could not process rule: File exists
set ss_rules_dst_bypass {
    ^^^^^^^^^^^^^^^^^^^
root@OpenWrt:/etc/v2ray/direct# ss-
ss-local   ss-redir   ss-server
root@OpenWrt:/etc/v2ray/direct# ss-redir
 2022-11-24 20:00:07 ERROR: Invalid config path.
```
because of
```
223.118.0.0/15                                                                                                                        
223.119.0.0/16
```
`223.118.0.0/15` == `223.118.0.0/16` + `223.119.0.0/16`
### solution
add `auto-merge` in the set declaration
https://lwn.net/Articles/896732/

@yousong

4ft35t, 2022-11-24:
Anther issue for ss-rules. I'm moving from https://openwrt-dist.sourceforge.net/, ss-rules in that package, __auto add remote server ip to bypass list__

Can this package add that feature？

1715173329, 2022-11-25:
See #19872