openwrt/packages #19970: openwrt 21.02.5 python3 error

predators46, 2022-11-24:
```In file included from /usr/include/features.h:461,                                                      
from /usr/include/bits/libc-header-start.h:33,                                         
from /usr/include/limits.h:26,                                                         
from ./Include/Python.h:11,                                                            
from /home/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/Python-3.9.15/Modules/_hashopenssl.c:27:                                                        
/usr/include/bits/stdio2.h:47:4: error: unknown type name '__gnuc_va_list'; did you mean 'va_list'?                                                                               
__gnuc_va_list __ap))                                                                  
^~~~~~~~~~~~~~```

predators46, 2022-11-24:
@jefferyto

BKPepe, 2022-11-24:
Sorry, but your report is so minimalist that is lacks so many details, which are necessary to look into it. You should not remove the issue template.

predators46, 2022-11-24:
```tmp/.config-package.in:50213:warning: ignoring type redefinition of 'LUCI_LANG_sk' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50216:warning: ignoring type redefinition of 'LUCI_LANG_sv' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50219:warning: ignoring type redefinition of 'LUCI_LANG_tr' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50222:warning: ignoring type redefinition of 'LUCI_LANG_uk' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50225:warning: ignoring type redefinition of 'LUCI_LANG_vi' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50228:warning: ignoring type redefinition of 'LUCI_LANG_zh_Hans' from 'bool' to 'tristate'                                                             tmp/.config-package.in:50231:warning: ignoring type redefinition of 'LUCI_LANG_zh_Hant' from 'bool' to 'tristate'                                                             tmp/.config-package.in:50249:warning: ignoring type redefinition of 'PACKAGE_luci-mod-admin-full' from 'bool' to 'tristate'                                                   tmp/.config-package.in:50324:warning: ignoring type redefinition of 'PACKAGE_luci-mod-network' from 'bool' to 'tristate'                                                      tmp/.config-package.in:50352:warning: ignoring type redefinition of 'PACKAGE_luci-mod-status' from 'bool' to 'tristate'                                                       tmp/.config-package.in:50367:warning: ignoring type redefinition of 'PACKAGE_luci-mod-system' from 'bool' to 'tristate'                                                       tmp/.config-package.in:50795:warning: ignoring type redefinition of 'PACKAGE_luci-app-firewall' from 'bool' to 'tristate'                                                     tmp/.config-package.in:51193:warning: ignoring type redefinition of 'PACKAGE_luci-app-opkg' from 'bool' to 'tristate'                                                         tmp/.config-package.in:51794:warning: ignoring type redefinition of 'PACKAGE_luci-theme-bootstrap' from 'bool' to 'tristate'                                                  tmp/.config-package.in:51908:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ipv6' from 'bool' to 'tristate'                                                       tmp/.config-package.in:51974:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ppp' from 'bool' to 'tristate'                                                        tmp/.config-package.in:52082:warning: ignoring type redefinition of 'PACKAGE_luci-lib-base' from 'bool' to 'tristate'                                                         tmp/.config-package.in:52138:warning: ignoring type redefinition of 'PACKAGE_luci-lib-ip' from 'bool' to 'tristate'                                                           tmp/.config-package.in:52201:warning: ignoring type redefinition of 'PACKAGE_luci-lib-jsonc' from 'bool' to 'tristate'                                                        tmp/.config-package.in:52215:warning: ignoring type redefinition of 'PACKAGE_luci-lib-nixio' from 'bool' to 'tristate'                                                        tmp/.config-package.in:90782:warning: ignoring type redefinition of 'PACKAGE_ip6tables' from 'bool' to 'tristate'                                                             tmp/.config-package.in:90848:warning: ignoring type redefinition of 'PACKAGE_iptables' from 'bool' to 'tristate'                                                              tmp/.config-package.in:113160:warning: ignoring type redefinition of 'PACKAGE_cgi-io' from 'bool' to 'tristate'                                                               tmp/.config-package.in:114627:warning: ignoring type redefinition of 'PACKAGE_uhttpd' from 'bool' to 'tristate'                                                               tmp/.config-package.in:114663:warning: ignoring type redefinition of 'PACKAGE_uhttpd-mod-ubus' from 'bool' to 'tristate'                                                      tmp/.config-package.in:115117:warning: ignoring type redefinition of 'PACKAGE_hostapd-common' from 'bool' to 'tristate'                                                       tmp/.config-package.in:115592:warning: ignoring type redefinition of 'PACKAGE_wpad-basic-wolfssl' from 'bool' to 'tristate'                                                   tmp/.config-package.in:117968:warning: ignoring type redefinition of 'PACKAGE_iw' from 'bool' to 'tristate'                                                                   tmp/.config-package.in:119931:warning: ignoring type redefinition of 'PACKAGE_odhcp6c' from 'bool' to 'tristate'                                                              tmp/.config-package.in:119982:warning: ignoring type redefinition of 'PACKAGE_odhcpd-ipv6only' from 'bool' to 'tristate'                                                      tmp/.config-package.in:120300:warning: ignoring type redefinition of 'PACKAGE_ppp' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:120343:warning: ignoring type redefinition of 'PACKAGE_ppp-mod-pppoe' from 'bool [... cut ...] 23#0.75                                ERROR: package/feeds/packages/python3 failed to build.                             make[1]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1    make[1]: Leaving directory '/home/openwrt/openwrt'                                     make: *** [/home/openwrt/openwrt/include/toplevel.mk:223: package/xderm-mini/compile] Error 2                                                                                 openwrt@costum69:~/openwrt$```

predators46, 2022-11-24:
@BKPepe 
Full log

[thWCAQNM.txt](https://github.com/openwrt/packages/files/10087517/thWCAQNM.txt)

predators46, 2022-11-24:
``tmp/.config-package.in:50213:warning: ignoring type redefinition of 'LUCI_LANG_sk' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50216:warning: ignoring type redefinition of 'LUCI_LANG_sv' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50219:warning: ignoring type redefinition of 'LUCI_LANG_tr' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50222:warning: ignoring type redefinition of 'LUCI_LANG_uk' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50225:warning: ignoring type redefinition of 'LUCI_LANG_vi' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:50228:warning: ignoring type redefinition of 'LUCI_LANG_zh_Hans' from 'bool' to 'tristate'                                                             tmp/.config-package.in:50231:warning: ignoring type redefinition of 'LUCI_LANG_zh_Hant' from 'bool' to 'tristate'                                                             tmp/.config-package.in:50249:warning: ignoring type redefinition of 'PACKAGE_luci-mod-admin-full' from 'bool' to 'tristate'                                                   tmp/.config-package.in:50324:warning: ignoring type redefinition of 'PACKAGE_luci-mod-network' from 'bool' to 'tristate'                                                      tmp/.config-package.in:50352:warning: ignoring type redefinition of 'PACKAGE_luci-mod-status' from 'bool' to 'tristate'                                                       tmp/.config-package.in:50367:warning: ignoring type redefinition of 'PACKAGE_luci-mod-system' from 'bool' to 'tristate'                                                       tmp/.config-package.in:50795:warning: ignoring type redefinition of 'PACKAGE_luci-app-firewall' from 'bool' to 'tristate'                                                     tmp/.config-package.in:51193:warning: ignoring type redefinition of 'PACKAGE_luci-app-opkg' from 'bool' to 'tristate'                                                         tmp/.config-package.in:51794:warning: ignoring type redefinition of 'PACKAGE_luci-theme-bootstrap' from 'bool' to 'tristate'                                                  tmp/.config-package.in:51908:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ipv6' from 'bool' to 'tristate'                                                       tmp/.config-package.in:51974:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ppp' from 'bool' to 'tristate'                                                        tmp/.config-package.in:52082:warning: ignoring type redefinition of 'PACKAGE_luci-lib-base' from 'bool' to 'tristate'                                                         tmp/.config-package.in:52138:warning: ignoring type redefinition of 'PACKAGE_luci-lib-ip' from 'bool' to 'tristate'                                                           tmp/.config-package.in:52201:warning: ignoring type redefinition of 'PACKAGE_luci-lib-jsonc' from 'bool' to 'tristate'                                                        tmp/.config-package.in:52215:warning: ignoring type redefinition of 'PACKAGE_luci-lib-nixio' from 'bool' to 'tristate'                                                        tmp/.config-package.in:90782:warning: ignoring type redefinition of 'PACKAGE_ip6tables' from 'bool' to 'tristate'                                                             tmp/.config-package.in:90848:warning: ignoring type redefinition of 'PACKAGE_iptables' from 'bool' to 'tristate'                                                              tmp/.config-package.in:113160:warning: ignoring type redefinition of 'PACKAGE_cgi-io' from 'bool' to 'tristate'                                                               tmp/.config-package.in:114627:warning: ignoring type redefinition of 'PACKAGE_uhttpd' from 'bool' to 'tristate'                                                               tmp/.config-package.in:114663:warning: ignoring type redefinition of 'PACKAGE_uhttpd-mod-ubus' from 'bool' to 'tristate'                                                      tmp/.config-package.in:115117:warning: ignoring type redefinition of 'PACKAGE_hostapd-common' from 'bool' to 'tristate'                                                       tmp/.config-package.in:115592:warning: ignoring type redefinition of 'PACKAGE_wpad-basic-wolfssl' from 'bool' to 'tristate'                                                   tmp/.config-package.in:117968:warning: ignoring type redefinition of 'PACKAGE_iw' from 'bool' to 'tristate'                                                                   tmp/.config-package.in:119931:warning: ignoring type redefinition of 'PACKAGE_odhcp6c' from 'bool' to 'tristate'                                                              tmp/.config-package.in:119982:warning: ignoring type redefinition of 'PACKAGE_odhcpd-ipv6only' from 'bool' to 'tristate'                                                      tmp/.config-package.in:120300:warning: ignoring type redefinition of 'PACKAGE_ppp' from 'bool' to 'tristate'                                                                  tmp/.config-package.in:120343:warning: ignoring type redefinition of 'PACKAGE_ppp-mod-pppoe' from 'bool' [... cut ...]
make[2]: Leaving directory '/home/openwrt/openwrt/feeds/packages/lang/python/python3'  time: package/feeds/packages/python3/compile#0.61#0.23#0.75                                
ERROR: package/feeds/packages/python3 failed to build.                             
make[1]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1    
make[1]: Leaving directory '/home/openwrt/openwrt'                                     
make: *** [/home/openwrt/openwrt/include/toplevel.mk:223: package/xderm-mini/compile] Error 2                                                                                 openwrt@costum69:~/openwrt$

closed 2022-11-24 by predators46