openwrt/packages #19978: simple-adblock: change in function call name dnsmasqOps to dnsmasq broke checks for dnsmasq

parona-source, 2022-11-27:
Maintainer: @stangri 
Environment: MediaTek MT7621 ver:1 eco:3, Netgear R6220, OpenWrt 22.03.2 r19803-9a599fee93

Description:
Noticed an error in the logs when setting ipset on for simple-adblock `ERROR: dnsmasq ipset support is enabled in simple-adblock, but dnsmasq is either not installed or installed dnsmasq does not support ipset!`.
```
root@OpenWrt:~# dnsmasq -v
Dnsmasq version 2.86  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.

root@OpenWrt:~# opkg info dnsmasq-full
Package: dnsmasq-full
Version: 2.86-15
Depends: libc, libubus20220601, libnettle8, kmod-ipt-ipset, libnetfilter-conntrack3
Provides: dnsmasq
Status: install user installed
Architecture: mipsel_24kc
Conffiles:
 /etc/config/dhcp ddd520eb24451a892c9c666d83c10c9ea4fc944efbc34a149bc962c56bd8812f
 /etc/dnsmasq.conf 1e6ab19c1ae5e70d609ac7b6246541d52042e4dee1892f825266507ef52d7dfd
Installed-Time: 1669562387
```

After checking the init script saw the older older commit https://github.com/openwrt/packages/commit/f8c5fd93e7e055e2425729e812fe0d1e4aab8032 which changed the function dnsmasqOps to dnsmasq which would then break any dnsmasq checks like `dnsmasq -v` which were intended to call the program instead of the function.

parona-source, 2022-11-27:
https://github.com/parona-source/packages/commit/eb455263c2948f760b0ab3edd26c39a64ac760ae Made patch for it, for which Ill look into making a pull request for once Im sure following the contributor guidelines. With this ipset was enabled and no error was shown.

stangri, 2022-11-27:
@parona-source Thank you Alfred, if you want to submit a PR, please make sure to bump the version and I'd appreciate if you renamed the problematic function to `resolver` as it may help to load it up with more features in the future.

Let me know if you want to send a PR.

parona-source, 2022-11-28:
@stangri Thanks for the prompt response.

As you are familiar with the required quality requirements with compile and runtime tests then it be ideal that you can make the pull request. Especially as Id be essentially copying the commit that you made for your personal repository instead. Im fine with any resolution really.

closed 2022-11-29 by stangri