openwrt/packages #19981: collectd-mod-interface: Erroneous  parameter name set for interface plugin through luci

szero, 2022-11-28:
Maintainer: @jow- @hnyman 
Environment:
[image]

Firmware: OpenWrt SNAPSHOT r21069-78cbcc77cc / LuCI Master git-22.294.25693-a398ccf

Description:
When you want to monitor specific network interfaces and set the value through luci, its uses wrong key name of `Interfaces` instead of `Interface` and collectd fails with following output: 
[image]

Plugin works when we are watching all interfaces since no `Interface` entry is set through luci.

```Mon Nov 28 10:04:48 2022 daemon.err collectd[6479]: Exiting normally.
Mon Nov 28 10:04:48 2022 daemon.err collectd[6479]: collectd: Stopping 4 read threads.
Mon Nov 28 10:04:48 2022 daemon.err collectd[6479]: collectd: Stopping 5 write threads.
Mon Nov 28 10:04:48 2022 daemon.err collectd[6479]: rrdtool plugin: Shutting down the queue thread.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "cpu" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "interface" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: Plugin `interface' did not register for value `Interfaces'.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: Plugin `interface' failed to handle option `Interfaces', return code: -1
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "iwinfo" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "load" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "memory" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: plugin_load: plugin "rrdtool" successfully loaded.
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Mon Nov 28 10:04:48 2022 daemon.err collectd[17849]: Error: Parsing the config file failed!
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "cpu" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "interface" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: Plugin `interface' did not register for value `Interfaces'.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: Plugin `interface' failed to handle option `Interfaces', return code: -1
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "iwinfo" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "load" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "memory" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: plugin_load: plugin "rrdtool" successfully loaded.
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Mon Nov 28 10:04:53 2022 daemon.err collectd[17886]: Error: Parsing the config file failed!
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "cpu" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "interface" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: Plugin `interface' did not register for value `Interfaces'.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: Plugin `interface' failed to handle option `Interfaces', return code: -1
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "iwinfo" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "load" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "memory" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: plugin_load: plugin "rrdtool" successfully loaded.
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Mon Nov 28 10:04:58 2022 daemon.err collectd[17928]: Error: Parsing the config file failed!
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "cpu" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "interface" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: Plugin `interface' did not register for value `Interfaces'.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: Plugin `interface' failed to handle option `Interfaces', return code: -1
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "iwinfo" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "load" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "memory" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: plugin_load: plugin "rrdtool" successfully loaded.
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Mon Nov 28 10:05:03 2022 daemon.err collectd[17931]: Error: Parsing the config file failed!
```
Collectd page for interface plugin: https://collectd.org/wiki/index.php/Plugin:Interface

hnyman, 2022-11-28:
It works for me in 22.03 and master. 
The uci option "Interfaces" gets converted ok into "Interface" for /etc/collectd.conf. 

For testing, I added a wlan interface via LuCI.

(But I haven't yet figured out where the Interfaces-->Interface conversion really happens.  stat-genconfig should be the place...)

22.03
```
root@router1:~# grep -i -C 2 interf /etc/config/luci_statistics
config statistics 'collectd_dns'
        option enable '0'
        option Interfaces 'br-lan'
        option IgnoreSources '127.0.0.1'

--
        option cmdline '/usr/libexec/collectd/sqm_collectd.sh eth0.2'

config statistics 'collectd_interface'
        option enable '1'
        list Interfaces 'br-lan'
        list Interfaces 'wlan0'

config statistics 'collectd_iptables'
--
        option enable '0'
        option IgnoreSelected '0'
        option VerboseInterfaces 'br-lan'
        option QDiscs 'br-lan'

--
config statistics 'collectd_snmp6'
        option enable '0'
        list Interfaces 'eth0.2'

config statistics 'collectd_splash_leases'
```
```
root@router1:~# grep -i -C 2 interf /etc/collectd.conf
</Plugin>

LoadPlugin interface
<Plugin interface>
        Interface "br-lan"
        Interface wlan0
</Plugin>

```

master:
```

root@router4:~# grep -i -C 2 interf /etc/config/luci_statistics
config statistics 'collectd_dns'
        option enable '0'
        option Interfaces 'br-lan'
        option IgnoreSources '127.0.0.1'

--
        option cmdline '/usr/libexec/collectd/sqm_collectd.sh wan'

config statistics 'collectd_interface'
        option enable '1'
        list Interfaces 'br-lan'
        list Interfaces 'wl1-ap0'

config statistics 'collectd_ipstatistics'
--
        option enable '0'
        option IgnoreSelected '0'
        option VerboseInterfaces 'br-lan'
        option QDiscs 'br-lan'

--
config statistics 'collectd_snmp6'
        option enable '0'
        option Interfaces 'br-lan'
        option IgnoreSelected '0'

root@router4:~#  grep -i -C 2 interf /etc/collectd.conf
LoadPlugin cpufreq

LoadPlugin interface
<Plugin interface>
        Interface "br-lan"
        Interface "wl1-ap0"
</Plugin>

```

szero, 2023-04-21:
Closing this, this some janky misconfiguration of my snapshot install on my part ...

closed 2023-04-21 by szero