openwrt/packages #20000: acme: deprecating state_dir

hgl, 2022-12-01:
# Issue template (remove lines from top till here)

Maintainer: @tohojo 
Environment: (put here arch, model, OpenWrt version)

Description:
I wonder if we should make `/etc/acme` the standard location to keep ceritificates and not let users customize it? The main reason being that it's specified in `acme-common`'s `conffiles`. Users who simply change the value in the config, but not specified it to be preserved (which is very likely),  run the risk of lose the certificates upon upgrading OpenWrt, which could potentially be serious.

Official packages like dropbear don't allow users to customize the directory to keep the generated host ssh keys and authorized keys. I think acme pacakges should follow suit.

What do you think?

tohojo, 2022-12-01:
Well the difference to something like dropbear is that the keys directory is only consumed by the package itself; whereas for acme, certificates can be used for things outside our control, so I think we should give user some flexibility as to where they want to put them...

So I don't think we should block it entirely; there may be users with a valid reason to change it? Adding a warning would probably be appropriate, though. We could also hide the setting from the luci settings to make it squarely an "advanced" setting?

hgl, 2022-12-02:
> whereas for acme, certificates can be used for things outside our control

Technically, certificates are only exposed through `/etc/ssl/acme`. IMHO, the real location should be private.

> Adding a warning would probably be appropriate, though

> We could also hide the setting from the luci settings
Yes, that's what I had in mind too, For backward compatibility, we can't simply eliminate the option. Changes I have in mind:

1. Warn if the option is set
2. Remove the option from the default config
3. Hide the option in LuCI

What do you think?

> Also, yay, issue 20k! :D

This is probably the first lottery I've won in my life, albeit useless, lol. And I guessed the World Cup results all wrong. :)