openwrt/packages #20004: wget: fails to build

rucketeg, 2022-12-01:
Maintainer: @hnyman 
Environment: building on x64 for bcm27xx (RPi 4)

Description: building wget fails

```
make[7]: Entering directory '/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3/lib'
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../src  -DDEFAULT_TEXT_DOMAIN=\"wget-gnulib\" -I/home/openwrt/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.3.0_musl/usr/include -I/home/openwrt/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.3.0_musl/include/fortify -I/home/openwrt/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.3.0_musl/include  -I/home/openwrt/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include   -Wno-cast-qual -Wno-conversion -Wno-float-equal -Wno-sign-compare -Wno-undef -Wno-unused-function -Wno-unused-parameter -Wno-float-conversion -Wimplicit-fallthrough -Wno-pedantic -Wno-sign-conversion -Wno-type-limits -Wno-unsuffixed-float-constants -I/home/openwrt/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include  -I/home/openwrt/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include  -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3=wget-1.21.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT libgnu_a-base32.o -MD -MP -MF .deps/libgnu_a-base32.Tpo -c -o libgnu_a-base32.o `test -f 'base32.c' || echo './'`base32.c
In file included from ./idx.h:26,
                 from base32.h:22,
                 from base32.c:43:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
make[7]: *** [Makefile:3036: libgnu_a-base32.o] Error 1
make[7]: Leaving directory '/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3/lib'
make[6]: *** [Makefile:2528: all] Error 2
make[6]: Leaving directory '/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3/lib'
make[5]: *** [Makefile:1741: all-recursive] Error 1
make[5]: Leaving directory '/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3'
make[4]: *** [Makefile:1693: all] Error 2
make[4]: Leaving directory '/home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3'
make[3]: *** [Makefile:112: /home/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/wget-ssl/wget-1.21.3/.built] Error 2
make[3]: Leaving directory '/home/openwrt/openwrt/feeds/packages/net/wget'
time: package/feeds/packages/wget/ssl/compile#0.21#0.04#0.25
    ERROR: package/feeds/packages/wget failed to build (build variant: ssl).
make[2]: *** [package/Makefile:116: package/feeds/packages/wget/compile] Error 1
make[2]: Leaving directory '/home/openwrt/openwrt'
make[1]: *** [package/Makefile:110: /home/openwrt/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/openwrt'
make: *** [/home/openwrt/openwrt/include/toplevel.mk:231: world] Error 2
```

rucketeg, 2022-12-01:
Building at 04c805b4cad04ac77b484319d8cc3dc4164ab7a0 works correctly so it seems that this issue was introduced by #19992

hnyman, 2022-12-02:
Buildbot has built wget also for aarch64-cortex-a72 (and other targets), so this might be something related to your specific build host or build config.

rucketeg, 2022-12-02:
After resetting the build environment `wget` builds correctly.  Sorry for the churn.

closed 2022-12-02 by rucketeg