openwrt/packages #20008: atlas-probe: version 2.4.1-2 "do_page_fault(): sending SIGSEGV to perd for invalid read access from XXXXXXXX"

KsmoinO, 2022-12-02:
Maintainer: @ja-pa 
Environment: ath79, dlink_dir-842-c3, 22.03.2

Description:
About once a day and up to 5 times a day, perd is killed with following kernel message "do_page_fault(): sending SIGSEGV to perd for invalid read access from XXXXXXXX"

As workaround I schedule this script every minutes to restart atlas each time perd is killed :
```
#!/bin/sh
pgrep /usr/libexec/atlas-probe/bin/perd
if [ $? -ne 0 ] ; then
  /etc/init.d/atlas restart
fi
```

Ansuel, 2022-12-02:
This should be fixed in later release... pls upgrade

BKPepe, 2022-12-02:
Wasn't this related to https://github.com/openwrt/packages/issues/19739 or it happens one more time?

KsmoinO, 2022-12-02:
> This should be fixed in later release... pls upgrade

I'm already at latest atlas-probe version (2.4.1-2) available on latest openwrt stable release (22.03.2)

> Wasn't this related to #19739 or it happens one more time?

No this time, it's not 'eperd' but 'perd' which is killed

Ansuel, 2022-12-02:
@BKPepe both were fixed so this sounds strange to me

KsmoinO, 2024-12-19:
I do not use atlas on this device anymore and do not have this issue on the new one

closed 2024-12-19 by KsmoinO