openwrt/packages #20024: https-dns-proxy: crash loop

Gelated-Thoughts, 2022-12-05:
Maintainer: @stangri 

Environment: OpenWRT 22.03.2 on Linksys WRT1900ACS v1 (arm_cortex-a9_vfpv3-d16), package  https-dns-proxy_2022-10-15-8_arm_cortex-a9_vfpv3-d16.ipk

Description:

Since adding the signal-based reset() in place of the full stop/start in https-dns-proxy_2022-10-15-8, my https-dns-proxy instances occasionally crash at the reload.

```
Sun Dec  4 06:04:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server [X.X.X.X]
Sun Dec  4 06:04:44 2022 daemon.notice netifd: wan (4512): udhcpc: lease of [Y.Y.Y.Y] obtained from [X.X.X.X], lease time 3600
Sun Dec  4 06:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 06:04:46 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3595 seconds since last crash
Sun Dec  4 06:04:46 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3595 seconds since last crash

```

I'm not sure how to find or supply a crash dump or other debugging info, but I am happy to provide any further information, if given some guidance.

Gelated-Thoughts, 2022-12-05:
Yes, I believe you are correct.

My ISP provides a DHCP lease of between 1 or 2 hours for the address on my wan interface, so udhcpc is renewing every 30 - 60 minutes, resulting in a reload of fw4 and triggering a https-dns-proxy restart.

The procd_send_signal reload method for https-dns-proxy seems to work most of the time, but occasionally it will crash instead. Once it crashes, it does not restart until I manually restart it.

To provide a bit better clarity, I modified the https-dns-proxy restart() routine to output a log message when it does the signal-based reload. It also detects if the service has crashed and issues a full stop/start combo. Here's what it looks like [all timestamps are PST (UTC-8)]:

```
Sun Dec  4 20:04:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 20:04:44 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3601
Sun Dec  4 20:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 20:04:46 2022 user.notice https-dns-proxy: Reloading service
Sun Dec  4 20:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 20:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 7200
Sun Dec  4 20:34:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 20:34:46 2022 user.notice https-dns-proxy: Reloading service
Sun Dec  4 21:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 21:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3600
Sun Dec  4 21:34:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 21:34:46 2022 user.notice https-dns-proxy: Reloading service
Sun Dec  4 22:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 22:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 7200
Sun Dec  4 22:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 22:04:46 2022 user.notice https-dns-proxy: Reloading service
Sun Dec  4 23:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 23:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3599
Sun Dec  4 23:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 23:04:46 2022 user.notice https-dns-proxy: Reloading service
Sun Dec  4 23:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Sun Dec  4 23:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 7200
Sun Dec  4 23:34:44 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Sun Dec  4 23:34:45 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 1794 seconds since last crash
Sun Dec  4 23:34:45 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 1794 seconds since last crash
Sun Dec  4 23:34:45 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 00:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Mon Dec  5 00:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3601
Mon Dec  5 00:34:44 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Dec  5 00:34:52 2022 user.notice https-dns-proxy: Stopping service ✓
Mon Dec  5 00:34:59 2022 user.notice https-dns-proxy: Starting service ✓✓
Mon Dec  5 01:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Mon Dec  5 01:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 7200
Mon Dec  5 01:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Dec  5 01:04:46 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 02:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Mon Dec  5 02:04:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3600
Mon Dec  5 02:04:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Dec  5 02:04:46 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 02:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Mon Dec  5 02:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 7200
Mon Dec  5 02:34:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Dec  5 02:34:46 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 03:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
Mon Dec  5 03:34:45 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from X.X.X.X, lease time 3599
Mon Dec  5 03:34:45 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Dec  5 03:34:46 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 04:04:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
[... cut ...]
Mon Dec  5 10:34:46 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3595 seconds since last crash
Mon Dec  5 10:34:46 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3595 seconds since last crash
Mon Dec  5 10:34:46 2022 user.notice https-dns-proxy: Reloading service
Mon Dec  5 10:44:30 2022 user.notice https-dns-proxy: Stopping service ✓
Mon Dec  5 10:44:37 2022 user.notice https-dns-proxy: Starting service ✓✓
```

stangri, 2022-12-06:
> To provide a bit better clarity, I modified the https-dns-proxy restart() routine to output a log message when it does the signal-based reload. It also detects if the service has crashed and issues a full stop/start combo. Here's what it looks like [all timestamps are PST (UTC-8)]:

I'd be curious to see the code.

> Sun Dec  4 23:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: sending renew to server X.X.X.X
> Sun Dec  4 23:34:44 2022 daemon.notice netifd: wan (4512): udhcpc: lease of Y.Y.Y.Y obtained from [...]

Is it normal that the IP is renewed twice within a second?

khanjui, 2022-12-06:
I've got a problem that might be related to this one. When my WAN connection downs (pppoe-bridge), https-dns-proxy doesn't restart itself, it stays as "stopped" so my devices don't have an internet connection unless I restart the service manually. I couldn't see anything on logs though... @stangri I believe the issue started after updating to 2022-10-15-8

Gelated-Thoughts, 2022-12-06:
> Is it normal that the IP is renewed twice within a second?

No, I don't think so, but I also don't see any evidence of that happening. Could you please elaborate a bit on what caused you to ask that question?

> I'd be curious to see the code.

Well, this was a quick hack for troubleshooting purposes, so I did not bother spending the time to make it properly elegant or robust, but here it is:

```
restart() {

        service_running

        if [ "$?" -eq 0 ]; then

                logger -t "$packageName" "Issuing service reload..."
                procd_send_signal "$packageName"
                logger -t "$packageName" "Reload issued"

                for i in $(awk 'BEGIN { for ( i=0; i<60; i++ ) { print i; } }')
                do
                        sleep 1
                        service_running
                        if [ "$?" -eq 0 ]; then
                                return
                        fi
                done

                logger -t "$packageName" "Waited 60 seconds, service appears to have crashed, performing stop and start"
                stop
                start

        else

                logger -t "$packageName" "Service appears to have crashed or stopped, performing stop and start"
                stop
                start

        fi
}
```

The output of this looks like the following. Note that with my change to output a log message both before and after the procd_send_signal is invoked, it's clear that when https-dns-proxy does crash, that's when it crashes. 

```
Tue Dec  6 01:47:40 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 01:47:40 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 3600
Tue Dec  6 01:47:40 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 01:47:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 01:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 01:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 01:47:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 01:48:42 2022 user.notice https-dns-proxy: Waited 60 seconds, service appears to have crashed, performing stop and start
Tue Dec  6 01:48:49 2022 user.notice https-dns-proxy: Stopping service ✓
Tue Dec  6 01:48:55 2022 user.notice https-dns-proxy: Starting service ✓✓
Tue Dec  6 02:17:40 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 02:17:40 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 7200
Tue Dec  6 02:17:40 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 02:17:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 02:17:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 03:17:40 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 03:17:40 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 3600
Tue Dec  6 03:17:41 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 03:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 03:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 03:47:41 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 03:47:41 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 7200
Tue Dec  6 03:47:41 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 03:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 03:47:42 2022 user.notice https-dns-proxy: Reload issued
[... 21 lines cut ...]
Tue Dec  6 06:47:40 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 7200
Tue Dec  6 06:47:40 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 06:47:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 06:47:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 07:47:40 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 07:47:40 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 3601
Tue Dec  6 07:47:41 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 07:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 07:47:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 08:17:41 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 08:17:41 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 7200
Tue Dec  6 08:17:41 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 08:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 08:17:42 2022 user.notice https-dns-proxy: Reload issued
[... cut ...]
Tue Dec  6 09:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 09:47:41 2022 daemon.notice netifd: wan (4453): udhcpc: sending renew to server [DHCP Server]
Tue Dec  6 09:47:41 2022 daemon.notice netifd: wan (4453): udhcpc: lease of [WAN IP Address] obtained from [DHCP Server], lease time 7200
Tue Dec  6 09:47:41 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Tue Dec  6 09:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 09:47:42 2022 user.notice https-dns-proxy: Reload issued
```

Gelated-Thoughts, 2022-12-06:
By just filtering down to just reload messages, it seems there is a clear pattern here. After hard restart, the reload works correctly 5 times. When the 6th reload signal is sent, https-dns-proxy crashes.

```
Tue Dec  6 01:47:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 01:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 01:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 01:47:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 01:48:42 2022 user.notice https-dns-proxy: Waited 60 seconds, service appears to have crashed, performing stop and start
Tue Dec  6 01:48:49 2022 user.notice https-dns-proxy: Stopping service ✓
Tue Dec  6 01:48:55 2022 user.notice https-dns-proxy: Starting service ✓✓
Tue Dec  6 02:17:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 02:17:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 03:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 03:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 03:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 03:47:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 04:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 04:47:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 05:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 05:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 06:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 06:17:42 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 06:17:42 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3595 seconds since last crash
Tue Dec  6 06:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 06:18:43 2022 user.notice https-dns-proxy: Waited 60 seconds, service appears to have crashed, performing stop and start
Tue Dec  6 06:18:49 2022 user.notice https-dns-proxy: Stopping service ✓
Tue Dec  6 06:18:57 2022 user.notice https-dns-proxy: Starting service ✓✓
Tue Dec  6 06:47:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 06:47:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 07:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 07:47:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 08:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 08:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 09:17:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 09:17:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 09:47:42 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 09:47:42 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 10:47:41 2022 user.notice https-dns-proxy: Issuing service reload...
Tue Dec  6 10:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3594 seconds since last crash
Tue Dec  6 10:47:41 2022 daemon.info procd: Instance https-dns-proxy::instance2 s in a crash loop 6 crashes, 3594 seconds since last crash
Tue Dec  6 10:47:41 2022 user.notice https-dns-proxy: Reload issued
Tue Dec  6 10:48:42 2022 user.notice https-dns-proxy: Waited 60 seconds, service appears to have crashed, performing stop and start
Tue Dec  6 10:48:48 2022 user.notice https-dns-proxy: Stopping service ✓
Tue Dec  6 10:48:55 2022 user.notice https-dns-proxy: Starting service ✓✓

```

stangri, 2022-12-08:
> No, I don't think so, but I also don't see any evidence of that happening. Could you please elaborate a bit on what caused you to ask that question?

I quoted the part where I thought it was happening and I didn't see the dates were different so the renewals happened exactly 24 hours apart, nevermind.

I'm testing a potential fix: https://github.com/stangri/source.openwrt.melmac.net/commit/7ab999bc3f76c3f84297b3fd1a00ea9c3f8760bc if you want to update your init file to the file from that commit and test I'd be grateful.

Gelated-Thoughts, 2022-12-08:
> I'm testing a potential fix: [stangri/source.openwrt.melmac.net@7ab999b](https://github.com/stangri/source.openwrt.melmac.net/commit/7ab999bc3f76c3f84297b3fd1a00ea9c3f8760bc) if you want to update your init file to the file from that commit and test I'd be grateful.

Absolutely!  Happy to test anything that's useful. I'll go put that in place right now and give it some soak-time.

Gelated-Thoughts, 2022-12-10:
So far, this change appears to be working for me. The https-dns-proxy service is gracefully restarting each time my wan DHCP lease is renewed, but procd is not having to restart it via the watchdog, so it's no longer detecting it as a crash loop.

stangri, 2023-01-16:
Safe to close with the release of 2022-10-15-10?

Gelated-Thoughts, 2023-01-16:
Good by me.  It's still working fine for me since that update.

closed 2023-01-16 by Gelated-Thoughts