openwrt/packages #20026: nextdns: non-retpoline-compatible errors

trippleflux, 2022-12-05:
Maintainer: @rs 
Environment: OpenWrt snapshot x86_64 glibc

Description:
Getting compilation errors : 
```
/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/share/gocode/src does not exist, skipping symlinks

touch /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.configured_eb3faa8116006abaa5579256904571c0
rm -f /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.built
touch /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="y" GO_BUILD_CACHE_DIR="/home/username/works/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/username/works/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/nextdns/nextdns" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/nextdns/nextdns/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0" GO_BUILD_DIR="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.go_work/build" GO_BUILD_BIN_DIR="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="ccache_cc" CXX="ccache_cxx" CGO_CFLAGS="-mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0=nextdns-1.38.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/include" CGO_CXXFLAGS="-mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0=nextdns-1.38.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/lib -znow -zrelro" GOPATH="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.go_work/build" GOCACHE="/home/username/works/openwrt/tmp/go-build" GOMODCACHE="/home/username/works/openwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1670156506' -linkmode external -extldflags '-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/lib -Wl,-z,now -Wl,-z,relro'" -gcflags "all=-spectre all" -asmflags "all=-spectre all"      -ldflags " -X main.version=1.38.0 -buildid '1670156506' -linkmode external -extldflags '-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets

Building targets
runtime
# runtime
<autogenerated>:1: non-retpoline-compatible: 00200 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/time.go:915)       JMP     (R15)(R12*8)
<autogenerated>:1: non-retpoline-compatible: 00115 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/type.go:614)       JMP     (AX)(SI*8)
<autogenerated>:1: non-retpoline-compatible: 00028 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/time.go:452)       JMP     (R11)(R10*8)
<autogenerated>:1: non-retpoline-compatible: 00021 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/error.go:261)      JMP     (DX)(CX*8)
<autogenerated>:1: non-retpoline-compatible: 00050 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/time.go:691)       JMP     (CX)(R12*8)
<autogenerated>:1: non-retpoline-compatible: 00024 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/debuglog.go:616)   JMP     (CX)(SI*8)
<autogenerated>:1: non-retpoline-compatible: 00079 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/time.go:617)       JMP     (R9)(R8*8)
<autogenerated>:1: non-retpoline-compatible: 00025 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/cgocall.go:453)    JMP     (R9)(DX*8)
<autogenerated>:1: non-retpoline-compatible: 00018 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/type.go:66)        JMP     (DX)(CX*8)
<autogenerated>:1: non-retpoline-compatible: 00020 (/home/username/works/openwrt/staging_dir/hostpkg/lib/go-cross/src/runtime/alg.go:156)        JMP     (SI)(DX*8)
<autogenerated>:1: too many errors

make[2]: *** [Makefile:86: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/nextdns-1.38.0/.built] Error 2
make[2]: Leaving directory '/home/username/works/openwrt/feeds/packages/net/nextdns'
time: package/feeds/packages/nextdns/compile#2.11#1.64#2.44
    ERROR: package/feeds/packages/nextdns failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/nextdns/compile] Error 1
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:231: package/nextdns/compile] Error 2
```

Tried to make clean, delete build_dir, staging_dir, tmp and dl/go-mod-cache

rs, 2022-12-05:
Any clue what is going on?

trippleflux, 2022-12-05:
No clue, nothing useful for me found in google also, only from go compiler? [source ](https://github.com/golang/go/blob/master/src/cmd/internal/obj/x86/asm6.go#L2075-L2085)code .

I am running latest debian sid x86_64 for compilation incase it's useful.

trippleflux, 2022-12-05:
@rs 
Hmm, reverting back golang or compiler into previous [1.18.8](https://github.com/openwrt/packages/tree/c33c2d886918cfbc1af8d60cbb530ba563084b67) is working for me, now nextdns compiled fine. Someone should report these to upstream but as I don't have go related knowledges, I've guess some who has it.

trippleflux, 2022-12-07:
Temporary patches before it [cherry-picked](https://github.com/golang/go/issues/57100) into 1.19.x branch : 
./feeds/packages/lang/golang/golang/patches/002-retpoline-fixes.patch
```
diff -Naur a/src/cmd/compile/internal/walk/switch.go b/src/cmd/compile/internal/walk/switch.go
--- a/src/cmd/compile/internal/walk/switch.go	2022-12-02 01:12:55.000000000 +0700
+++ b/src/cmd/compile/internal/walk/switch.go	2022-12-07 19:32:13.620067400 +0700
@@ -289,7 +289,7 @@
 	const minCases = 8   // have at least minCases cases in the switch
 	const minDensity = 4 // use at least 1 out of every minDensity entries
 
-	if !go119UseJumpTables || base.Flag.N != 0 || !ssagen.Arch.LinkArch.CanJumpTable {
+	if !go119UseJumpTables || base.Flag.N != 0 || !ssagen.Arch.LinkArch.CanJumpTable || base.Ctxt.Retpoline {
 		return false
 	}
 	if len(cc) < minCases {
diff -Naur a/test/codegen/retpoline.go b/test/codegen/retpoline.go
--- a/test/codegen/retpoline.go	2022-11-01 03:17:31.000000000 +0700
+++ b/test/codegen/retpoline.go	2022-12-07 19:31:24.060474800 +0700
@@ -12,3 +12,31 @@
 	// amd64:`CALL\truntime.retpoline`
 	x.M()
 }
+
+// Check to make sure that jump tables are disabled
+// when retpoline is on. See issue 57097.
+func noJumpTables(x int) int {
+	switch x {
+	case 0:
+		return 0
+	case 1:
+		return 1
+	case 2:
+		return 2
+	case 3:
+		return 3
+	case 4:
+		return 4
+	case 5:
+		return 5
+	case 6:
+		return 6
+	case 7:
+		return 7
+	case 8:
+		return 8
+	case 9:
+		return 9
+	}
+	return 10
+}
\ No newline at end of file
```

PR #20177 (2022-12-28): golang: backport an upstream fix for non-retpoline-compatible error
Maintainer: @jefferyto 
Compile tested: x86/64
Run tested: n/a

Description:
Upstream commit: golang/go@156578067111742b55718066c91b8ec66d35e03d
Fixes: #20026