openwrt/packages #20028: tailscale: Fails to start due to mismatch of go version

mschae, 2022-12-05:
Maintainer: @ja-pa
Environment: ARMv7 Processor rev 1 (v7l), Linksys WRT1200AC, OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.335.71542-7e42425

Description:

Starting `tailscale` (or `tailscaled`, both version 1.24.2-3) yields the following error:

```
panic: Something in this program imports go4.org/unsafe/assume-no-moving-gc to declare that it assumes a non-moving garbage collector, but your version of go4.org/unsafe/assume-no-moving-gc hasn't been updated to assert that it's safe against the go1.19 runtime. If you want to risk it, run with environment variable ASSUME_NO_MOVING_GC_UNSAFE_RISK_IT_WITH=go1.19 set. Notably, if go1.19 adds a moving garbage collector, this program is unsafe to use.

goroutine 1 [running]:
go4.org/unsafe/assume-no-moving-gc.init.0()
	go4.org/unsafe/assume-no-moving-gc@v0.0.0-20211027215541-db492cf91b37/untested.go:25 +0x1d8

```

I believe this version of tailscale is meant to run on (/be compiled with) go 1.18, looks like it is being run (/was compiled) with 1.19.

paper42, 2022-12-05:
go creates standalone binaries, so it's not possible to "run on 1.18". This seems to be because it was compiled with go 1.19, but one of the dependencies doesn't work with 1.19.
This can probably be fixed by backporting https://github.com/tailscale/tailscale/commit/22c544bca7edcc2e0c7c2adce99c28751a57cf0d or updating tailscale to at least the first version that contains this commit - 1.28.0.

mschae, 2022-12-05:
Your first sentence isn’t particularly helpful to move the conversation forward but let’s not get hung up on it. 

Looks like there is a PR that will fix this: https://github.com/openwrt/packages/pull/1961

I’ll let them know over there.

mschae, 2022-12-05:
Your first sentence isn’t particularly helpful to move the conversation forward but let’s not get hung up on it.

Looks like there is a PR that will fix this: https://github.com/openwrt/packages/pull/19613

I’ll let them know over there.

closed 2022-12-05 by BKPepe