openwrt/packages #20044: luci-proto-wireguard: WireGuard shows `Error: Network device is not present` on LuCi

marek22k, 2022-12-08:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: MediaTek MT7628AN ver:1 eco:2 / GL-MT300N-V2 / OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.339.35241-4ca7a8d

Description:

Hello,

I want to build a WireGuard tunnel over the Yggdrasil overlay network. The Yggdrasil network is provided by the package `yggdrasil`. Yggdrasil successfully establishes its connection to the network over a WiFi Adhoc network. My endpoint is `218:4feb:a509:9db2:2b34:6e7e:e071:5dee`. I have a route to this endpoint and I can ping it. The WireGuard tunnel works when the device is connected to the Internet via WAN. However, if I disconnect the WAN, the message `Error: Network device is not present` appears and it does not even try to establish the WG tunnel, although it is possible because the endpoint is reachable.

My expected behavior is that OpenWrt always tries to reach the endpoint if there is a route to it in the kernel routing table.

```
# ip r g 218:4feb:a509:9db2:2b34:6e7e:e071:5dee
218:4feb:a509:9db2:2b34:6e7e:e071:5dee from :: dev ygg0  src 21e:295a:a561:adb1:f834:9e20:1757:183a  metric 256 
# ping 218:4feb:a509:9db2:2b34:6e7e:e071:5dee
PING 218:4feb:a509:9db2:2b34:6e7e:e071:5dee (218:4feb:a509:9db2:2b34:6e7e:e071:5dee): 56 data bytes
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=0 ttl=64 time=1347.156 ms
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=1 ttl=64 time=384.689 ms
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=2 ttl=64 time=72.068 ms
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=3 ttl=64 time=69.410 ms
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=4 ttl=64 time=69.935 ms
64 bytes from 218:4feb:a509:9db2:2b34:6e7e:e071:5dee: seq=5 ttl=64 time=69.803 ms
^C
--- 218:4feb:a509:9db2:2b34:6e7e:e071:5dee ping statistics ---
6 packets transmitted, 6 packets received, 0% packet loss
round-trip min/avg/max = 69.410/335.510/1347.156 ms
```
The command `wg` does not output anything.

[image]

diizzyy, 2023-07-30:
Closing as it's not a package issue and wireguard maintained in this repo, you're probably better off trying IRC or the forums.

closed 2023-07-30 by diizzyy

marek22k, 2023-07-30:
This is OpenWrt specific and not WireGuard specific.
Since `wg` did not output anything, no configuration was ever delivered to WireGuard. Therefore WireGuard did not even try to establish a connection. If OpenWrt does not pass the configuration to WireGuard, this is a bug in OpenWrt and not in WireGuard.
Furthermore, the error message "Network device not presend" is OpenWrt specific and not WireGuard specific.
@diizzyy So I would like to ask you to reopen this issue.

marek22k, 2023-08-01:
This is OpenWrt specific and not WireGuard specific.
Since `wg` did not output anything, no configuration was ever delivered to WireGuard. Therefore WireGuard did not even try to establish a connection. If OpenWrt does not pass the configuration to WireGuard, this is a bug in OpenWrt and not in WireGuard.
Furthermore, the error message "Network device not presend" is [OpenWrt specific](https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/network.js#L15) and not WireGuard specific.
@diizzyy So I would like to ask you to reopen this issue.

diizzyy, 2023-08-01:
It's the wrong repository

marek22k, 2023-08-01:
Which one is the right one?