openwrt/packages #20046: igmpproxy: update available (0.4.0

damnms, 2022-12-09:
Maintainer: Jan Pavlinec <jan.pavlinec@nic.cz>@\ja-pa 

Description: There is a newer version of igmpproxy available (0.4.0), would be nice if that could be used in the next releases, guess this is just one line of code to adjust

BKPepe, 2022-12-09:
Any pull request is appreciated, so if you think if it is one line of code to adjust, then it seems to be an easy job, which you can do as well. :)

damnms, 2022-12-09:
the change is probably a one-liner, but i never did anything like that on such a big "platform", and i have no idea how to test if it works or how testing in general is done in openwrt 
https://github.com/openwrt/packages/compare/master...damnms:packages:master

BKPepe, 2022-12-09:
We do have CI here, which will compile the packages for each platform and after that, you can download it, copy it on your router, install it there and test, if everything works there. 

BTW: CI is going to complain about your commit subject, it should be smth like this one: ``igmpproxy: update to version 0.4.0`` and in the commit description, there should be ``Signed-off-by``.

damnms, 2022-12-11:
maybe a stupid question but, how can i create a pull request for another branch? i forked the repo, but it forked only the master branch, not e.g. openwrt-22.03. but i would like to add that change there too. any idea without forking again?

1715173329, 2022-12-22:
Closed due to #20060

closed 2022-12-22 by 1715173329