openwrt/packages #20072: yggdrasil: Yggdrasil on OpenWrt hangs back several versions

marek22k, 2022-12-12:
Maintainer: @wfleurant @zhoreeq @aparcar @JamesVorder @dengqf6

Description:

The current Yggdrasil version for OpenWrt is 0.4.3-1. This is no longer the current version.
see https://openwrt.org/packages/pkgdata/yggdrasil and https://github.com/yggdrasil-network/yggdrasil-go/issues/1000

taggart, 2023-11-23:
Not sure if I am seeing the same issue, but...

Currently:
- pkgdata on the [website](https://openwrt.org/packages/pkgdata/yggdrasil) still shows: 0.4.3-1
- on 23.05.2 if I update package lists the version I see is: 0.4.7-1
-  if I look in [a package dir for 23.05.2](https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/packages/) I see: 0.4.7-1

So maybe it's just the website pkgdata that is out of date?

m7mdcc, 2023-11-27:
We need Yggdrasil  v0.5.x

wfleurant, 2023-12-04:
yggdrasil 0.5.3 is in snapshots please check it out.
for OpenWrt-22.03.0 (Old Stable Release) the version is 0.4.3.
the only solution is to backport the latest to support the new protocol.

hmoffatt, 2024-02-16:
OpenWRT 23.05 has Yggdrasil 0.5.4.

zhoreeq, 2025-07-09:
it's v0.5.12 in master branch, v0.5.9 in 24.10 branch and v0.5.10 in 23.05 branch

@CupIvan can you cherry pick your commits from master to those branches? Or I can do that later this month when on a Linux computer

Saiv46, 2025-07-13:
@zhoreeq Currently all of three versions are compatible with each other, so there's no hurry for that.

zhoreeq, 2025-07-15:
@Saiv46 sure, there are important routing bug fixes though