openwrt/packages #20085: adguardhome won't start - stat etc/hosts: os: DirFS with empty root on SNAPSHOT

Mhisani, 2022-12-13:
Maintainer: @dobo90 Dobroslaw Kijowski
Environment: mediatek/mt7622,  aarch64  OpenWrt SNAPSHOT r21464

Description:
In the few days snapshot adguardhome does not launch properly. Attempting to launch adguardhome mimicing the init script gives the following

```
 /usr/bin/AdGuardHome -c /etc/adguardhome.yaml -w /var/adguardhome/

2022/12/13 22:13:58.400987 [info] AdGuard Home, version v0.107.18
2022/12/13 22:13:58.408282 [info] tls: using default ciphers
2022/12/13 22:13:58.408969 [fatal] initing hosts container: hosts container: path at index 0: stat etc/hosts: os: DirFS with empty root
```

As a work around if I run

``` /usr/bin/AdGuardHome -c /etc/adguardhome.yaml -w /var/adguardhome/  --no-etc-hosts```

Then everything works fine. 

I don't claim to know what's happening but I did wonder if it was something to do with this CVE: [https://www.tenable.com/cve/CVE-2022-41720]([](https://www.tenable.com/cve/CVE-2022-41720))

Thankyou so much!

trippleflux, 2022-12-13:
Afaik It's upstream issue, it's fixed in [this commit](https://github.com/AdguardTeam/AdGuardHome/commit/fa49d74aa8f0eac826331732d4008799dc6b8f1e). Scroll down to the below of that page.

GeorgeSapkin, 2025-04-04:
Is this still relevant?

closed 2025-10-30 by GeorgeSapkin