openwrt/packages #20092: nut-upsmon/nut-upsmon

aqwserf, 2022-12-15:
Maintainer: @neheb @cotequeiroz @EricLuehrsen @robje 
Environment:
```
root@router-1:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.1'
DISTRIB_REVISION='r19777-2853b6d652'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 22.03.1 r19777-2853b6d652'
DISTRIB_TAINTS=''
```

Description:
After reinstalling `nut-upsmon` and `nut-server` on my Linksys WTR3200ACM, I noticed that after setting some timers on `upssched.conf`, I receive some emails from my script when the router just start.

After investigating, it seems that the init script `/etc/init.d/nut-monitor` is called 3 times during a reboot (at least on my router). I've added some `echo <functions>` at the beginning of each:

```
Thu Dec 15 13:32:30 2022: stop_service
Thu Dec 15 13:32:44 2022: start_service
Thu Dec 15 13:32:44 2022: build_config
Thu Dec 15 13:32:44 2022: nut_upsmon_conf
Thu Dec 15 13:32:44 2022: nut_upsmon_add
Thu Dec 15 13:32:44 2022: interface_triggers
Thu Dec 15 13:32:44 2022: service_triggers
Thu Dec 15 13:32:44 2022: interface_triggers
Thu Dec 15 13:32:51 2022: stop_service
Thu Dec 15 13:32:51 2022: start_service
Thu Dec 15 13:32:51 2022: build_config
Thu Dec 15 13:32:51 2022: nut_upsmon_conf
Thu Dec 15 13:32:51 2022: nut_upsmon_add
Thu Dec 15 13:32:51 2022: interface_triggers
Thu Dec 15 13:32:51 2022: service_triggers
Thu Dec 15 13:32:51 2022: interface_triggers
Thu Dec 15 13:33:25 2022: stop_service
Thu Dec 15 13:33:25 2022: start_service
Thu Dec 15 13:33:25 2022: build_config
Thu Dec 15 13:33:25 2022: nut_upsmon_conf
Thu Dec 15 13:33:25 2022: nut_upsmon_add
Thu Dec 15 13:33:25 2022: interface_triggers
Thu Dec 15 13:33:25 2022: service_triggers
Thu Dec 15 13:33:25 2022: interface_triggers
```

The first time, `upsmon` calls `upssched` with a timer, let's say 15s for `commbad`, then `upsmon` restarts but `upssched` keeps running. Finally  `upsmon` restarts again and `upssched` runs the command after the 15s. At least this is my interpretation given the following log (we can see different PID for `upsmon`):

```
Thu Dec 15 13:32:42 2022 daemon.info upsd[4263]: Network UPS Tools upsd 2.7.4
Thu Dec 15 13:32:42 2022 daemon.info upsd[4263]: listening on 192.168.33.254 port 3493
Thu Dec 15 13:32:42 2022 daemon.err upsd[4263]: Can't connect to UPS [eatonellipsepro1200] (usbhid-ups-eatonellipsepro1200): No such file or directory
Thu Dec 15 13:32:42 2022 daemon.info usbhid-ups[4262]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
Thu Dec 15 13:32:42 2022 daemon.info usbhid-ups[4262]: USB communication driver 0.33
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.000000	fopen /var/run/upsmon.pid: No such file or directory
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.004402	Using power down flag file /var/run/killpower
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.004821	UPS: eatonellipsepro1200@192.168.33.254:3493 (master) (power value 1)
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.007053	debug level is '1'
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.010864	Trying to connect to UPS [eatonellipsepro1200@192.168.33.254:3493]
Thu Dec 15 13:32:44 2022 daemon.info upsd[4263]: User upsmonitor@192.168.33.254 logged into UPS [eatonellipsepro1200]
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.014784	Logged into UPS eatonellipsepro1200@192.168.33.254:3493
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.015175	Poll UPS [eatonellipsepro1200@192.168.33.254:3493] failed - Driver not connected
Thu Dec 15 13:32:44 2022 daemon.err upsmon[5420]:    0.015271	Communications with UPS eatonellipsepro1200@192.168.33.254:3493 lost
Thu Dec 15 13:32:44 2022 daemon.info upssched[5462]: Timer daemon started
Thu Dec 15 13:32:44 2022 daemon.info upssched[5462]: New timer: commbad (15 seconds)
Thu Dec 15 13:32:46 2022 daemon.info upsd[4263]: User proxmox@192.168.33.20 logged into UPS [eatonellipsepro1200]
Thu Dec 15 13:32:49 2022 daemon.info upsd[4263]: Connected to UPS [eatonellipsepro1200]: usbhid-ups-eatonellipsepro1200
Thu Dec 15 13:32:49 2022 daemon.err upsmon[5420]:    5.016005	Poll UPS [eatonellipsepro1200@192.168.33.254:3493] failed - Driver not connected
Thu Dec 15 13:32:49 2022 daemon.err upsmon[5420]:    5.016028	UPS eatonellipsepro1200@192.168.33.254:3493 is unavailable
Thu Dec 15 13:32:49 2022 daemon.info upssched[5462]: New timer: nocomm (15 seconds)
Thu Dec 15 13:32:51 2022 daemon.err upsmon[5420]:    6.714892	Signal 15: exiting
[... 218 lines cut ...]
		MAIL_TXT="${DATE}\nCommunication lost with the UPS ${UPS} server ${SERVER}.";;
	shutdown)
		#note : executed on the master only
		#MAIL_TXT="Shutdown master server $SERVER on $DATE";;
		MAIL_TXT="${DATE}\nForce shutdown primary server ${SERVER}.\n\nUPS Status: ${STATUS}\nUPS Battery Charge: ${CHARGE}%\nUPS Runtime: ${RUNTIME} seconds";;
	replbatt)
		#MAIL_TXT="Battery needs to be replaced on $UPS on $DATE";;
		MAIL_TXT="${DATE}\nThe ${UPS}'s battery needs to be replaced.\n\nUPS Status: ${STATUS}\nUPS Battery Charge: ${CHARGE}%\nUPS Runtime: ${RUNTIME} seconds";;
	nocomm)
		#MAIL_TXT="Communication unavailable with $SERVER on $DATE";;
		MAIL_TXT="${DATE}\nCommunication unavailable with server ${SERVER}.";;
	noparent)
		#MAIL_TXT="Process that shuts down the system has died on $DATE. Shutdown impossible";;
		MAIL_TXT="${DATE}\nProcess that shuts down the system has died. Shutdown impossible";;
	*)
	#nothing
		logger_add "error: The state ${STATE} is not managed"
		exit 1
esac

log_state
mail_send

exit 0
```

EricLuehrsen, 2022-12-20:
`procd` has some bugs or features (opinion differs) based being a much smaller reinvention of `systemd.` The interface triggers have a related hypersensitivity in `netifd` to DHCPv6 routing refresh with no material routing change. This all takes tweaking the scripts and for one provided UCI/Luci options to trigger on WAN only and not WAN6. However that being said, this package has no volunteer maintainer. Anyone might push fixes when its seriously bad, but none have stepped up for regular enhancements. Nut itself was stale a few years, and only recently received new life, so that took its toll.

aqwserf, 2022-12-22:
Thanks for your answer @EricLuehrsen and your explanation.
I have indeed both ipv4 and ipv6 and my wan interface, which might cause the 2 restarts.
As `procd` seems to be the culprit, I close the ticket and hopefully someone will volunteer to maintain this package !

closed 2022-12-22 by aqwserf