openwrt/packages #20097: curl: zlib not linked in. Oversight?

IngwiePhoenix, 2022-12-17:
Maintainer: @\[stangri](https://github.com/openwrt/packages/commits?author=stangri)

Environment:
```
root@FriendlyWrt /s/s/nextcloud# cd
root@FriendlyWrt ~# uname -a
Linux FriendlyWrt 5.10.110 #1 SMP Sat Dec 3 01:25:15 CST 2022 aarch64 GNU/Linux
root@FriendlyWrt ~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS='busybox'
```

Device: [FriendlyElec NanoPi R6s](https://www.friendlyelec.com/index.php?route=product/product&product_id=289)

Description:

I ran into problems where the available cURL in the repositories was missing `zlib` and caused numerous problems with the applications I run. Whilst working out the root cause, I eventually figured out that [zlib should have been configured](https://github.com/openwrt/packages/blob/master/net/curl/Makefile#L135) but apparently didn't ship? I am new to how packages work on OpenWrt - so forgive me if I am a little lost here. Because the `aria2c` package actually does have `zlib` compiled in. cURL does not.

```
root@FriendlyWrt ~ [1]# aria2c --version
aria2 version 1.36.0
Copyright (C) 2006, 2019 Tatsuhiro Tsujikawa

This program is free software; you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation; either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

** Configuration **
Enabled Features: BitTorrent, GZip, HTTPS, Message Digest
Hash Algorithms: sha-1, sha-224, sha-256, sha-384, sha-512, md5, adler32
Libraries: zlib/1.2.11 OpenSSL/1.1.1q
Compiler: gcc 11.2.0
  built by  x86_64-pc-linux-gnu
  targeting aarch64-openwrt-linux-gnu
  on        Oct 14 2022 22:44:41
System: Linux 5.10.110 #1 SMP Sat Dec 3 01:25:15 CST 2022 aarch64

Report bugs to https://github.com/aria2/aria2/issues
Visit https://aria2.github.io/
root@FriendlyWrt ~# curl --version
curl 7.86.0 (aarch64-openwrt-linux-gnu) libcurl/7.86.0 mbedTLS/2.28.1 nghttp2/1.44.0
Release-Date: 2022-10-26
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 IPv6 Largefile SSL threadsafe
root@FriendlyWrt ~# curl --compressed --head "https://apps.nextcloud.com/api/v1/apps.json"
curl: option --compressed: the installed libcurl version doesn't support this
curl: try 'curl --help' for more information
```

I run a NextCloud instance and it can not download metadata for available apps or even it's own updates as both of them are, by default apparently, requested as compressed streams. 

Also I verified that FriendlyWrt - which is basically FriendlyElec's fork of OpenWrt with a few packages added and default configs adjusted to their board - did not ship their own package of cURL and that it came directly from OpenWrt. In fact, those are my feeds, just to be double-save:

```
# cat /etc/opkg/distfeeds.conf
src/gz openwrt_base https://downloads.openwrt.org/releases/22.03.2/packages/aarch64_generic/base
src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03.2/packages/aarch64_generic/luci
src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03.2/packages/aarch64_generic/packages
src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03.2/packages/aarch64_generic/routing
src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03.2/packages/aarch64_generic/telephony
src/gz friendlywrt_packages file://opt/packages
# opkg info curl
Package: curl
Version: 7.86.0-2
Depends: libc, libcurl4
Status: install user installed
Section: net
Architecture: aarch64_generic
Size: 58983
Filename: curl_7.86.0-2_aarch64_generic.ipk
Description: A client-side URL transfer utility
Installed-Time: 1670640028
```

So my guess is that something did not work properly whilst compiling/deploying the cURL package. :) Would love if you could look into this perhaps?

Thanks a lot and kind regards,
Ingwie

stangri, 2022-12-17:
@IngwiePhoenix it's a curl compilation-time option and it's disabled by default as most use cases do not need it and enabling it results in a bigger binary/package size). If you really need the zlib support in curl, you should download the SDK for your platform and compile your own version of curl with zlib enabled in a Linux host/VM.

IngwiePhoenix, 2022-12-17:
I do have the SDK installed; but how exactly *do* I enable it? Also when I add packages I compiled myself, how do I add the outputs as repository to opkg then?

Thank you for your reply!

Spudz76, 2022-12-17:
The config option is `LIBCURL_ZLIB` you should be able to find it in menuconfig (`/` to search) otherwise it's buried a few levels deep in `Libraries -> libcurl -> Enable zlib support`

stangri, 2023-01-16:
Closing as the concern seems to have been addressed.

closed 2023-01-16 by stangri