openwrt/packages #20107: python3: recent libressl causing python3 has no ssl?

trippleflux, 2022-12-18:
Maintainer: @jefferyto / Jeffery To
Environment: X86_64 snapshot glibc binutils 2.39 GCC 12.2.0

Description:

Not really sure if it's really related but it seems the recent libressl upgrade causing python has no SSL ?
```
root@echelon:/home/username/works/openwrt/staging_dir/hostpkg/bin# ./python3 -m ssl
Traceback (most recent call last):
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/ssl.py", line 99, in <module>
    import _ssl             # if we can't import it, let the error propagate
ModuleNotFoundError: No module named '_ssl'
```

trippleflux, 2022-12-19:
Confirmed, downgraded into version libressl 3.5.3 makes python3 has SSL again

thuehn, 2022-12-19:
HI all,

I just ran into the same problem and here are the steps the fixed it on my side:

git revert ee0cade0141beb0e24c9734a4305161d570a59b2 #revert tools/libressl: update to 3.7.0
make package/python3/host/clean
make

Greetings Thomas

PolynomialDivision, 2022-12-19:
For completeness: https://github.com/openwrt/openwrt/commit/ee0cade0141beb0e24c9734a4305161d570a59b2

PolynomialDivision, 2022-12-19:
Did you start from a clean state? I see no faillogs in the builder:
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/

dhewg, 2022-12-20:
See the both links of bd6e20542199b4805346059cc8e26f67fff8affd. Skimming over recent commits there I guess that the patches `026-openssl-feature-flags.patch` and `028-host-python-support-ssl-with-libressl.patch` now need to be removed? And if that indeed works there're no libressl specific python patches left \o/ (the part of the 028 patch was known, see https://github.com/openwrt/packages/pull/18981#issuecomment-1189057606)

PolynomialDivision, 2022-12-20:
> See the both links of [bd6e205](https://github.com/openwrt/packages/commit/bd6e20542199b4805346059cc8e26f67fff8affd). Skimming over recent commits there I guess that the patches `026-openssl-feature-flags.patch` and `028-host-python-support-ssl-with-libressl.patch` now need to be removed? And if that indeed works there're no libressl specific python patches left \o/ (the part of the 028 patch was known, see [#18981 (comment)](https://github.com/openwrt/packages/pull/18981#issuecomment-1189057606))

Indeed. :)

trippleflux, 2022-12-20:
@dhewg & @PolynomialDivision 
Great work, confirmed fixed 👍

PR #20137 (2022-12-21): python3: fix ssl support by removing libressl patches
Remove libressl specific patches. With commit
("tools/libressl: update to 3.7.0") they are no longer needed, rather they cause python3 to be compiled without working ssl-support.

Fixes: #20107
Suggested-by: Andre Heider <a.heider@gmail.com>

Maintainer: @jefferyto 
Compile tested: master
Run tested: tbd
commit 4ecd9d67 (2022-12-21): python3: fix ssl support by removing libressl patches

Remove libressl specific patches. With commit
("tools/libressl: update to 3.7.0") they are no longer needed,
rather they cause python3 to be compiled without working ssl-support.

Fixes: #20107
Suggested-by: Andre Heider <a.heider@gmail.com>
closed 2022-12-21 by PolynomialDivision