openwrt/packages #20142: https://forum.openwrt.org/t/adguardhome-v107-6/146081

Concubine, 2022-12-20:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
Based on [https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=268444 4](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=268444)

There is a known issue with adguardhome which has been rectified in later verison, current supported (https://openwrt.org/packages/pkgdata/adguardhome) version 107.6 has this flaw and I am unable to start AdGuard on my openwrt install:

details:
|Model|Linksys WRT3200ACM|
|Architecture|ARMv7 Processor rev 1 (v7l)|
|Target Platform|mvebu/cortexa9|
|Firmware Version|OpenWrt 21.02.5 r16688-fa9a932fdb / LuCI openwrt-21.02 branch git-22.245.77575-63bfee6|

Please update AdGuardHome to latest 107.20 to resolve this issue
https://github.com/AdguardTeam/AdGuardHome

1715173329, 2022-12-28:
can you try if #20168 works?

Concubine, 2022-12-28:
Currently its working with what I set it up by manually installing with curl which is 107.21 while I didn't check with the opkg official it is working with that version; if you merged it to the official openwrt package I think we can close this one.

PR #20168 (2022-12-28): [openwrt-21.02] adguardhome: update to v0.107.21
Maintainer: @dobo90
Compile tested: bcm27xx/bcm2710
Run tested: rpi-3b

Description:
Backport adguardhome changes from the master branch.
Fixes: #20142
closed 2022-12-28 by Concubine