openwrt/packages #20184: apinger: Broken default behavior and daemon status

sarcasticadmin, 2022-12-28:
Maintainer: @samm-git @jempatel 
Environment: 
```
model: wndr3800ch
openwrt (`master` branch): fb15cb4ce9559021d463b5cb3816d8a9eeb8f3f9
opkg (`master` branch): c8fcec9371fff43c0b62a58ec2064e83fe39f403
```

## Description:

Related to: https://github.com/openwrt/packages/pull/18913 https://github.com/openwrt/packages/pull/19073

`apinger` no longer reads from the original default config location: `/etc/apinger.conf` instead it defaults to new instance vars and UCI configuration defined in `/etc/config/apinger` and leverages a default template inside the init script (https://github.com/openwrt/packages/blob/573355dedd1229e62afe2fdee9fa48ef4bbd587f/net/apinger/files/apinger.init#L131-L179) This configuration seems to make assumptions about interface names defaulting to `wan`:

```
~# cat /etc/config/apinger 
config interface 'wan'
        option debug '0'
        option status_interval '5'

#config interface 'wan2'
#       option debug '0'
#       option status_interval '5'

# delay is in ms
#config alarm_delay 'delay200'
#       option delay_low '100'
#       option delay_high '200'

# loss is in %
#config alarm_loss 'loss50'
#       option percent_low '30'
#       option percent_high '50'

#config target 'target1'
#       option interface 'wan'
#       option address '8.8.8.8'
#       option alarm_delay 'a1'
#       option alarm_loss 'loss50'
#       option probe_interval '5'

#config target 'target2'
#       option interface 'wan2'
#       option address '8.8.8.8'
```

The service is also unable to correctly track its status if the process dies:

```
~$ service apinger start
~$ service apinger status
running
```

Then in syslog since I dont have a `wan` interface in my configuration the service logs:

```
~$ tail /var/log/messages
...
Dec 27 11:12:59 OpenWrt apinger[3014]: [Dec 27 11:12:59] Starting Alarm Pinger, apinger(3014)
Dec 27 11:12:59 OpenWrt apinger[3014]: [Dec 27 11:12:59] No usable targets found, exiting
```

service still thinks `apinger` running even though its not:

```
~# ps | grep -i apinger
 3303 root      1316 S    grep -i apinger
~# service apinger status
running
```

It also seems to unable to find a command if the daemon is hard stopped and then `restart` is issued:

```
~# service apinger stop
~# service apinger status
inactive
~# service apinger restart
Command failed: Not found
~# service apinger status
running
```

It seems like the most recent changes are around supporting hotplugs and rrd (https://github.com/openwrt/packages/pull/18913 https://github.com/openwrt/packages/pull/19073) but should this be considered a separate flavor of the package all together instead of the default package and behavior? My concern is it seems like a radical departure from the original service and pkg functionality considering the size of the patches to `apinger` codebase. This was even called out https://github.com/openwrt/packages/pull/19073#issuecomment-1198687893 but the init script seems to be the same between the two packages (`apinger` and `apinger-rdd`). I also get an directory in the root of my filesystem which I have no use for `/apinger/rrd`:

```
~# ls -lah /apinger/
drwxr-xr-x    3 root     root           0 Dec 27 07:40 .
drwxr-xr-x    1 root     root           0 Dec 27 07:40 ..
drwxr-xr-x    2 root     root           0 Dec 27 07:40 rrd
```

## Questions

1. Would it be possible to retain a more simplistic `apinger` service for the non rrd flavor? The current state of apinger breaks the default behavior of previous versions.
2. Is there anyway to pass a configuration file to the `apinger` service in its current state? I dont believe there is since the existing service only provides getting a subset of config.

Any help would be appreciated!

jempatel, 2023-04-14:
> Maintainer: @samm-git @jempatel Environment:
>
> ```
> model: wndr3800ch
> openwrt (`master` branch): fb15cb4ce9559021d463b5cb3816d8a9eeb8f3f9
> opkg (`master` branch): [...]
Yes, that default config was kept `wan` by default because openwrt uses `wan` as its default wan interface name in `/etc/config/network`. 

This behavior can be changed with UCI commands or If it's customized firmware then a custom `/etc/uci-defaults` script can be used to change the interface name.
> ```
> ~# cat /etc/config/apinger
> config interface 'wan'
>         option debug '0'
>         option status_interval '5'
>
> #config interface 'wan2'
> #       option debug '0'
> #       option status_interval [...]
Yes, this is a real bug, Work for that is in progress and it can be tracked from
https://github.com/openwrt/openwrt/pull/11884
> 
> ```
> ~$ service apinger start
> ~$ service apinger status
> running
> ```
> 
> Then in syslog since I dont have a `wan` interface in my configuration the service logs:
> 
> ```
> ~$ tail /var/log/messages
> ...
> Dec 27 11:12:59 OpenWrt apinger[3014]: [Dec 27 11:12:59] Starting Alarm Pinger, apinger(3014)
> Dec 27 11:12:59 OpenWrt apinger[3014]: [Dec 27 11:12:59] No usable targets found, exiting
> ```
> 
> service still thinks `apinger` running even though its not:
> 
> ```
> ~# ps | grep -i apinger
>  3303 root      1316 S    grep -i apinger
> ~# service apinger status
> running
> ```
> 
> It also seems to unable to find a command if the daemon is hard stopped and then `restart` is issued:
> 
> ```
> ~# service apinger stop
> ~# service apinger status
> inactive
> ~# service apinger restart
> Command failed: Not found
> ~# service apinger status
> running
> ```
> 
> It seems like the most recent changes are around supporting hotplugs and rrd (#18913 #19073) but should this be considered a separate flavor of the package all together instead of the default package and behavior? My concern is it seems like a radical departure from the original service and pkg functionality considering the size of the patches to `apinger` codebase. This was even called out [#19073 (comment)](https://github.com/openwrt/packages/pull/19073#issuecomment-1198687893) but the packages + init script seems to be the same between the two packages (`apinger` and `apinger-rdd`). I also get an directory in the root of my filesystem which I have no use for `/apinger/rrd`:
> 
> ```
> ~# ls -lah /apinger/
> drwxr-xr-x    3 root     root           0 Dec 27 07:40 .
> drwxr-xr-x    1 root     root           0 Dec 27 07:40 ..
> drwxr-xr-x    2 root     root           0 Dec 27 07:40 rrd
> ```
> 
> ## Questions
> 
>     1. Would it be possible to retain a more simplistic `apinger` service for the non rrd flavor? The current state of apinger breaks the default behavior of previous versions.

apinger-rrd is a separate package which installs few scripts to generate RRD graphs for loss and delays in the targets. RRD support already exists for apinger, and patches are added to make apinger work with different srcip.

Before apinger worked with one instance only, with new changes, it creates a separate instance for different interfaces, and exposes hotplug hooks when loss/delay alarms are triggered to take actions from hotplug scripts.
> 
>     2. Is there anyway to pass a configuration file to the `apinger` service in its current state? I dont believe there is since the existing service only provides getting a subset of config.
Yes, with new changes, it's not supported but we can add that as well. I will look into that.
> 
> 
> Any help would be appreciated!

sarcasticadmin, 2023-04-15:
@jempatel I really appreciate the follow up!

>  Yes, that default config was kept wan by default because openwrt uses wan as its default wan interface name in /etc/config/network.
> This behavior can be changed with UCI commands or If it's customized firmware then a custom /etc/uci-defaults script can be used to change the interface name.

That's good to know

> The service is also unable to correctly track its status if the process dies:
Yes, this is a real bug, Work for that is in progress and it can be tracked from
https://github.com/openwrt/openwrt/pull/11884

Thank you for spinning this out and noticing its a wider issue for procd

> Before apinger worked with one instance only, with new changes, it creates a separate instance for different interfaces, and exposes hotplug hooks when loss/delay alarms are triggered to take actions from hotplug scripts.

I would checkout [dpinger](https://github.com/dennypage/dpinger) since it seems to be the natural successor to `apinger` and already have these features out of the box. Although its not in opkg yet.
 
> Yes, with new changes, it's not supported but we can add that as well. I will look into that.

Thank you! I also would be perfectly fine just using the original pkg, rather than having to do all the work to re-support the feature set.