openwrt/packages #20189: adblock: can't whitelist single part domain

K2IE, 2022-12-28:
I have an internal DNS domain, let's call it "something".  Attempts to either whitelist everything in "something" or even a particular host, such as "openwrt.something" are failing.

The block count as shown in the DNS report keeps increasing.  I suspect that the issue is around the single part domain name.  This ought to be supported as it is fairly common in well layed out internal home/office situations, such as the famous ".local".

Hopefully my description is clear.

Thank you.

dibdot, 2023-01-04:
Most probably  an issue with your local DNS - nothing that is fixable in adblock reporting,

K2IE, 2023-01-04:
Running unbound, with the local domain being forwarded over to DNSmasq.  Operationally it is working fine as far as I can tell.

Spudz76, 2023-01-04:
Why would your own fake TLD have ads to block?  Why would it ever end up on a blocklist, to be blocked?

"The ads are coming from *inside the house*!!!"

dibdot, 2023-01-04:
> Running unbound, with the local domain being forwarded over to DNSmasq.

When unbound can't resolve your local domains it reports NXDomain ... which you'll see in your adblock report.

K2IE, 2023-01-04:
@dibdot Dirk...

In this case, unbound is resolving the local domain via a call to dnsmasq.  It does return an A record, so there is no NXDomain returned to the DNS resolver.  Unless, it locally logs an NXDomain and then sees that the domain is handled via dnsmasq and makes the call.

Thoughts on this?

dibdot, 2023-01-06:
Yep, the report shows all NXDomain issues, whether they are triggered by adblock or not. The "culprit" is not visible in the dump.
Anyway, domain parts like ".local" or wildcard domains are not supported in RPZ lists.

closed 2023-01-26 by dibdot