openwrt/packages #20195: samba4: TimeMachine does not work with MacOS >= Monterey

ffainelli, 2022-12-28:
Maintainer: @Gingernut1978, @neheb 
Environment: bcm2712, 21.02.3

Description:

With the following configuration for a time machine enabled shared:

```
config sambashare
        option guest_only 'yes'
        option name 'TimeMachine'
        option path '/mnt/backup/TimeMachine'
        option guest_ok 'yes'
        option read_only 'no'
        option timemachine '1'
        #option create_mask '0666'
        #option dir_mask '0777'
        option inherit_owner 'yes'
```

It is possible to use Time Machine from a client Mac device unless the MacOS version is greater than or equal to Monterey.

ffainelli, 2022-12-28:
On a client MAC the error happens when there is a renaming operation of the sparsebundle file initially created with a sort of a has value into the machine's friendly network name.

neheb, 2022-12-28:
I don’t use Samba

ffainelli, 2022-12-28:
Who maintains this package nowadays?

neheb, 2022-12-28:
Nobody really

Neustradamus, 2022-12-29:
@ffainelli: Now there are ksmbd (added in Kernel) and ksmbd-tools.

Tickets are here:

1/ ksmbd:
- https://github.com/namjaejeon/ksmbd/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc
- https://github.com/cifsd-team/ksmbd/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

2/ ksmbd-tools:
- https://github.com/namjaejeon/ksmbd-tools/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc
- https://github.com/cifsd-team/ksmbd-tools/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

trippleflux, 2022-12-29:
@ffainelli 

Doesn't use macOS but Have you tried [this ](https://golem.ph.utexas.edu/~distler/blog/archives/003393.html)and [this ](https://wiki.samba.org/index.php/Configure_Samba_to_Work_Better_with_Mac_OS_X)?.

ffainelli, 2022-12-29:
@trippleflux, I did that, thanks for the pointers. Using the OpenWrt-generated smb.conf file on an Ubuntu 20.04 machine works as is, so there is most likely something we are not building in the openwrt version of samba4 that is required by newer MacOS.

Gingernut1978, 2022-12-30:
Seems it might not be exclusive to OpenWrt:
 
https://community.synology.com/enu/forum/1/post/149543

trippleflux, 2022-12-31:
@ffainelli 
Based on [here](https://packages.ubuntu.com/search?keywords=samba), It seems that Ubuntu 20.04 is using Samba 4.13, and on OpenWrt 21.02.3 is using Samba [4.14](https://openwrt.org/packages/pkgdata/samba4-server), you could try either using Samba 4.17 which is on Ubuntu Lunar or OpenWrt snapshot, perhaps by using Samba 4.17, we could work on it better to troubleshooting it and have better chances they has been fixed if the issues are from upstream samba .

If you are building the OpenWrt yourself you could try to see if OpenWrt samba4 package are already applying patches as your aforementioned working on Ubuntu 20.04 from [here](http://archive.ubuntu.com/ubuntu/pool/main/s/samba/samba_4.13.17~dfsg-0ubuntu1.20.04.1.debian.tar.xz) (check the patches folder).

It's too bad cause I don't have test client (macOS) to troubleshooting it.

ffainelli, 2023-01-03:
@trippleflux let me update my Pi 4 to a newer version of OpenWrt and test Samba 4.17. I believe you may be able to find MacOS virtual machine disks on the internet :)