openwrt/packages #20197: Domoticz 2022.1 in OpenWrt 22.03.2 - Python 3.10 not supported

aerno, 2022-12-29:
After updating to OpenWrt 22.03.2 I noticed my Python plugins in Domoticz don't work any more.

My Domoticz test command line startup:

/usr/bin/domoticz -noupdates -approot /usr/share/domoticz/ -loglevel 10 -userdata /overlay/domoticz -dbase ./domoticz.db -www 9000 -sslwww 0

When starting Domoticz from command line with loglevel high I see this error popping along:

2022-12-29 19:23:20.469  Status: EventSystem - Python: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.

The Domoticz version 2022.1 seems to support a limited range of Python versions:

strace -f /usr/bin/domoticz -noupdates -approot /usr/share/domoticz/ -loglevel 10 -userdata /overlay/domoticz -db
ase ./domoticz.db -www 9000 -sslwww 0 2>&1 | grep -i "\"/lib/libpython"

open("/lib/libpython3.9.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.9m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.8.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.8m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.7.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.7m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.6.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.6m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.5.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.5m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libpython3.4m.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)

It supports version 3.4 upto 3.9. 

Newer Domoticz versions should also support Python 3.10 afaik.

The Domoticz binary starts and runs in the background without a hitch, except none of the Python plugins work.

---

I am testing rebuilding OpenWrt 22.03.2 on CentOS 7 having the latest Domoticz package downloaded during build.

BKPepe, 2022-12-30:
I think in your report there is a small typo as in OpenWrt 22.03, there was domoticz version 2021.1 not 2022.1. But I backported the later version to OpenWrt 22.03.

See: https://github.com/openwrt/packages/commit/60af04fa47a8e2f59371992712be08b76a41e24a

closed 2022-12-30 by BKPepe

aerno, 2022-12-30:
Yes, I made a typo, it should mention Domoticz version 2021.1