openwrt/packages #20200: ddns-scripts: PowerDNS script not able to set root record

JFroch, 2022-12-31:
Maintainer: @LecrisUT (as of https://github.com/openwrt/packages/pull/16429)
Environment: ARMv8 Processor rev 4, Xiaomi AX3600, OpenWrt SNAPSHOT r21587-959484625d

Description:

At the moment it is not possible to set a record at root level (sub.example.com is possible, example.com not)
This could be fixed by using '@' as the root domain identifier in accordance to [RFC 1035](https://datatracker.ietf.org/doc/html/rfc1035) / most DNS management software which could be achieved by replacing line 23 in `net/ddns-scripts/files/usr/lib/ddns/update_pdns.sh:`

```
json_add_string name "$username.$domain"
```

with:

```
if [ $username == '@' ]
then
json_add_string name "$domain"
else
json_add_string name "$username.$domain"
fi
```

Also i would propose to check if the supplied domain ends with a `.` and if not prepend it which would help usability since the PowerDNS API requires it.

LecrisUT, 2022-12-31:
That sounds reasonable. I had to check the APIs to see how they do @ records, and it seems your change would work. I don't have router right now setup to test it, so let me know if it works.

For checking the domain, how about:
```bash
if [ ${domain: -1} != '.' ]; then
  domain="${domain}."
fi
```

If you want to open a PR on it let me know, otherwise I will write those up later in the day.

JFroch, 2022-12-31:
I tested both changes and can confirm they work.

I haven´t done a PR yet so before i mess something up it is best when you do that.
Thank you for your (very fast) help!

PR #20201 (2023-01-08): ddns-scripts: Allow setting root record for pdns script
Maintainer: me
Compile tested: @JFroch manually tested
Run tested: (put here arch, model, OpenWrt version, tests done)

Closes #20200
closed 2023-01-08 by 1715173329