openwrt/packages #20203: vim: after update to version 9 new error on start  E1187: Failed to source defaults.vim

ajayramaswamy, 2023-01-02:
Maintainer: Nick Hainke <vincent@systemli.org>
Environment: openwrt master snapshot for nanopi r5s
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21614+102-425945c313'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT 2023-01-01-slim'
DISTRIB_TAINTS='no-all busybox'
```

Description:
on a clean install with a local build which has CONFIG_PACKAGE_vim-full=y

```
root@gw:~# vim /etc/rc.local 
E1187: Failed to source defaults.vim
Press ENTER or type command to continue

```

the older build with vim8x did not have that

a simple cat /usr/share/vim/vimrc > /usr/shrare/vim/defaults.vim made the problem go away

vndroid, 2023-10-31:
Me too.

Run `cat /usr/share/vim/vimrc | tee /usr/share/vim/defaults.vim` can fix.

mattcen, 2023-11-04:
Surely a better work-around would be a symlink, not a copy:
```sh
ln -s vimrc /usr/share/vim/defaults.vim
```

yhding5618, 2024-02-15:
`touch ~/.vimrc` also fix. The soft link can't fix for non-root users in my case.

newzealandpaul, 2025-01-14:
This remains an issue. Also /opt/share/vim/syntax/syntax.vim is not installed which may be a realted issue.

Try the command in vim

:syntax on

[image]

wwuck, 2025-05-24:
@newzealandpaul You need to install `vim-fuller` package to get the syntax files.

The `vim-fuller` package also contains `/usr/share/vim/vim90/defaults.vim` so this error does not appear when running vim from `vim-fuller`.