openwrt/packages #20219: glib2 host build missing a dependency for libpcre2 and zlib  ???

hnyman, 2023-01-04:
cc recent glib2 committers @neheb @feckert @micmac1 

**Summary:**

glib2 is possibly missing host build dependency  for libpcre2, zlib and libffi ???

I accidentally stumbled on it when I tried a build after "make dirclean" on a buildhost without internet connectivity.
glib2 host build seems to download missing sources in the background, so the error does not materialize normally.
As "make clean" does not clean `build_dir/hostpkg`, this is not visible after a simple "make clean".

**Details:**

glib2 has a host build, which seems to need some additional sources if the host packages are missing.

I accidentally ran into this, when I tried to build E8450/RT3200 after a "make dirclean" with a buildhost that temporarily had no internet connectivity. That caused a build error for me. Took me while to figure it out, as if there is connectivity, the glib2 host build apparently downloads the missing libpcre2 and zlib sources in background from somewhere in upstream...

It took me a while to figure out that it really takes the missing hostpkg packaged combined to no internet connectivity to show the problem.

glib2 does not declare any host build dependencies, but maybe it should...

( @neheb added the "forcefallback" in the last version upgrade. That has an impact here? )

The extract below is from the host build log after manually removing build_dir/hostpkg/glib-2.74.0/ , and shows the downloading of missing libpcre2 and zlib sources in the middle of the host build process.
I am not really making heads or tails from it, is it something about meson/ninja, or just something about missing dependencies.

```
Downloading pcre2 source from https://github.com/PhilipHazel/pcre2/releases/download/pcre2-10.40/pcre2-10.40.tar.bz2
Downloading pcre2 patch from https://wrapdb.mesonbuild.com/v2/pcre2_10.40-3/get_patch
...
Downloading zlib source from https://zlib.net/fossils/zlib-1.2.11.tar.gz
Downloading zlib patch from https://wrapdb.mesonbuild.com/v2/zlib_1.2.11-6/get_patch
```

**logs/package/feeds/packages/glib2/host-compile.txt:**

```
...
Header <pthread.h> has symbol "pthread_getname_np" : YES 
Header <sys/syscall.h> has symbol "SYS_sched_getattr" : YES 
Checking if "pthread_setname_np(const char*)" with dependency threads: links: NO 
Checking if "pthread_setname_np(pthread_t, const char*)" with dependency threads: links: YES 
Checking if "stack grows check" runs: NO (1)
Run-time dependency iconv found: YES
Looking for a fallback subproject for the dependency libpcre2-8 because:
Use of fallback dependencies is forced.
Downloading pcre2 source from https://github.com/PhilipHazel/pcre2/releases/download/pcre2-10.40/pcre2-10.40.tar.bz2
Download size: 1765440
Downloading: ..........
Downloading pcre2 patch from https://wrapdb.mesonbuild.com/v2/pcre2_10.40-3/get_patch
Download size: 2750
Downloading: ..........Cloning into 'libffi'...

Executing subproject pcre2 

pcre2| Project name: pcre2
pcre2| Project version: 10.40
pcre2| C compiler for the host machine: gcc (gcc 12.2.0 "gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0")
pcre2| C linker for the host machine: gcc ld.bfd 2.39
pcre2| Has header "inttypes.h" : YES (cached)
pcre2| Has header "bzlib.h" : NO
pcre2| Has header "sys/stat.h" : YES (cached)
pcre2| Has header "sys/types.h" : YES (cached)
pcre2| Has header "sys/wait.h" : YES (cached)
pcre2| Has header "inttypes.h" : YES (cached)
pcre2| Has header "dirent.h" : YES (cached)
pcre2| Has header "dlfcn.h" : YES
pcre2| Has header "limits.h" : YES (cached)
pcre2| Has header "stdint.h" : YES (cached)
pcre2| Has header "stdlib.h" : YES (cached)
pcre2| Has header "string.h" : YES (cached)
[... 146 lines cut ...]
    default_library    : static
    wrap_mode          : forcefallback
    bsymbolic_functions: true
    dtrace             : false
    force_posix_threads: true
    glib_assert        : false
    glib_checks        : true
    glib_debug         : disabled
    gtk_doc            : false
    installed_tests    : false
    libelf             : disabled
    libmount           : disabled
    man                : false
    nls                : disabled
    oss_fuzz           : disabled
    selinux            : disabled
    sysprof            : disabled
    systemtap          : false
    tests              : false
    xattr              : false

Found ninja-1.11.0 at /Openwrt/e8450/staging_dir/host/bin/ninja
touch /Openwrt/e8450/build_dir/hostpkg/glib-2.74.0/.configured
...
```

neheb, 2023-01-04:
IIRC forcefallback was a hack I applied because glib2's pcre2 handling was subpar. I'll look into it later.

neheb, 2023-01-04:
Btw gvdb will always get downloaded.

neheb, 2023-01-07:
Now I remember why I added this. glib's broken pcre2 handling finds pcre2 even when host headers are missing(aka -dev or -devel packages).

neheb, 2023-05-11:
I believe so.

neheb, 2023-05-11:
Now that there's a pcre2 host build. Could use that and remove forcefallback, Or maybe that's still broken. Not sure.

feckert, 2023-05-11:
@neheb @hnyman So if I build the host build `libffi/host` and `pcre2/hos`t before then I don't see any download of the packages in the log. 

I understand the `forcefallback` to mean that he is trying to download and build the missing things. That's why I've removed the fact that it shouldn't do that.

That would be my changes to fix this issue?
```
diff --git a/libs/glib2/Makefile b/libs/glib2/Makefile
index 076423fb0..aba2411ea 100644
--- a/libs/glib2/Makefile
+++ b/libs/glib2/Makefile
@@ -25,6 +25,7 @@ HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/glib-$(PKG_VERSION)
 PKG_CONFIG_DEPENDS:=CONFIG_BUILD_NLS
 PKG_FORTIFY_SOURCE:=0
 PKG_BUILD_FLAGS:=gc-sections
+PKG_BUILD_DEPENDS:=pcre2/host libffi/host

 include $(INCLUDE_DIR)/host-build.mk
 include $(INCLUDE_DIR)/package.mk
@@ -61,7 +62,7 @@ COMP_ARGS= \
        -Dglib_checks=true \
        -Dlibelf=disabled

-MESON_HOST_ARGS += $(COMP_ARGS) -Dxattr=false -Ddefault_library=static -Dnls=disabled -Dwrap_mode=forcefallback
+MESON_HOST_ARGS += $(COMP_ARGS) -Dxattr=false -Ddefault_library=static -Dnls=disabled
 MESON_ARGS += $(COMP_ARGS) -Dxattr=true -Db_lto=true -Ddefault_library=both -Dnls=$(if $(CONFIG_BUILD_NLS),en,dis)abled

 define Build/InstallDev
```

Could you confirm this?

neheb, 2023-05-11:
That should be HOST_BUILD_DEPENDS

The forcefallback thing is a hack, yes. Should be removed.

PR #20988 (2023-05-12): glib2: fix host build offline build
Maintainer: @neheb
Compile tested: x86_64, openwrt master
Run tested: none

fixes #20219 

Description:
If buildsystem does not have a internet connection, then the build fails. This is because of the fact, that glib2 is downloading and building the missing dependencies during host compilation.

This cannot be allowed to happen. The package must also be built without an internet connection.

Adding the missing host build dependency and removing the 'forcefallback' fixes this issue.
closed 2023-05-12 by neheb