openwrt/packages #20226: bind fail to compile

very20101, 2023-01-05:
### Describe the bug

ARMv8
ubuntu 22.04
bind fail to compile

make[3] -C feeds/packages/net/bind compile
ERROR: package/feeds/packages/bind failed to build.
......

find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/bind-9.18.10/ipkg-aarch64_cortex-a53/bind-libs -name 'CVS' -o -name '.svn' -o -name '.#' -o -name '~'| xargs -r rm -rf
Package bind-libs is missing dependencies for the following libraries:
libmaxminddb.so.0.0.7
make[3]: *** [Makefile:290: /workdir/openwrt/bin/packages/aarch64_cortex-a53/packages/bind-libs_9.18.10-2_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/bind'
time: package/feeds/packages/bind/compile#0.29#0.06#0.32
ERROR: package/feeds/packages/bind failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/bind/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

### OpenWrt version

openwrt-22.03

### OpenWrt target/subtarget

armvirt

### Device

ARMv8

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

myannnascar, 2023-01-06:
Try make dirclean , then retry

oyksoft, 2023-01-06:
The same error. Please check it.

very20101, 2023-01-07:
> Try #20231

Thank you! Solved.

closed 2023-01-07 by very20101