openwrt/packages #20232: How did I build openresty at nginx packages

bboyjunzi, 2023-01-06:
Please make sure that the issue subject starts with `<package-name>: `
Nginx
Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
How did I build luajit within nginx
```
Format code blocks by wrapping them with pairs of ```
```

Spudz76, 2023-01-09:
openresty is a full fork of nginx and is not packaged for OpenWRT.  The luajit extension is not available as just-a-plugin to add on to the normal nginx source.

someone (you?) must make a package for openresty if you want it.  probably easy since most of what it would need is already in the nginx package (mainly swap out the sources and maybe add some extra stuff to the configure phase).

Spudz76, 2023-01-09:
I did find this in the menuconfig, maybe it does what you need?
[image]

it will not be in the prebuilds so to try it you must compile your own

pretty sure it's not quite the same as luajit though

I haven't bothered with openresty in years.  The fact they don't just make it a plugin is stupid (and they claim they aren't a fork when what the heck else would it be called?)

bboyjunzi, 2023-01-09:
I'm working at 
https://github.com/openresty/lua-resty-websocket/tree/master
beacuase the openresty need to run luajit + nginx . but nginx of openwrt package was lua+nginx.
So I raise a question, nobody go forward at openwrt nginx.

Spudz76, 2023-01-09:
yes there must be an openresty package created, because it's a ["not-a-fork" full fork of nginx](https://openresty.org/)

brada4, 2023-01-09:
> I'm working at https://github.com/openresty/lua-resty-websocket/tree/master beacuase the openresty need to run luajit + nginx . but nginx of openwrt package was lua+nginx. So I raise a question, nobody go forward at openwrt nginx.

Seems real stuff, just that uses no-jit system default lua interpreter in place of openresty forked luajit.
https://github.com/openwrt/packages/blob/88b0e30806965a73058d79dd2d8bcbe6a2da88f9/net/nginx/Makefile#L498

closed 2025-11-15 by GeorgeSapkin