openwrt/packages #20247: error compile ath10k-4.19 with latest commit

Oozil, 2023-01-09:
I want to compile ath10k-4.19 with the latest commit, but it gives me an error and it never worked. What should I do?

Oozil, 2023-01-10:
> Please don't spam your problem in unrelated issue trackers.

I want to help and I couldn't find a place to help me

stintel, 2023-01-10:
> > Please don't spam your problem in unrelated issue trackers.
> 
> I want to help and I couldn't find a place to help me

Spamming your problem is only going to annoy people, reducing the chance of anyone wanting to help you. Either way, you should reach out to the OpenWrt community instead of creating a bug, as most likely you're just doing something that is not supported.

See https://openwrt.org/contact. My personal suggestion is to use the forum or IRC.