openwrt/packages #20248: <ocserv>: no longer seems to be found after upgrading to OpenWrt 22.03.3

wrighrc, 2023-01-09:
# Issue template (remove lines from top till here)

Maintainer: @\nmav (find it by checking history of the package Makefile)
Environment: openwrt-22.03.3-ramips-mt7621-netgear_r6350

Description:
On  ocserv doesn't seem available

root@router:~# opkg update
Downloading https://downloads.openwrt.org/releases/22.03.3/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.3/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@router:~# opkg install ocserv
Unknown package 'ocserv'.
Collected errors:
 * opkg_install_cmd: Cannot install package ocserv.
root@router:~# opkg list | grep ocserv
luci-app-ocserv - git-20.110.55046-74da73b - LuCI Support for OpenConnect VPN
luci-i18n-ocserv-ar - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - العربية (Arabic)
luci-i18n-ocserv-bg - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - български (Bulgarian)
luci-i18n-ocserv-bn - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - বাংলা (Bengali)
luci-i18n-ocserv-ca - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Català (Catalan)
luci-i18n-ocserv-cs - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Čeština (Czech)
luci-i18n-ocserv-da - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Dansk (Danish)
luci-i18n-ocserv-de - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Deutsch (German)
luci-i18n-ocserv-el - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Ελληνικά (Greek)
luci-i18n-ocserv-en - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - English
luci-i18n-ocserv-es - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Español (Spanish)
luci-i18n-ocserv-fi - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Suomi (Finnish)
luci-i18n-ocserv-fr - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Français (French)
luci-i18n-ocserv-he - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - עִבְרִית (Hebrew)
luci-i18n-ocserv-hi - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - हिंदी (Hindi)
luci-i18n-ocserv-hu - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Magyar (Hungarian)
luci-i18n-ocserv-it - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Italiano (Italian)
luci-i18n-ocserv-ja - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - 日本語 (Japanese)
luci-i18n-ocserv-ko - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - 한국어 (Korean)
luci-i18n-ocserv-mr - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Marāṭhī (Marathi)
luci-i18n-ocserv-ms - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Bahasa Melayu (Malay)
luci-i18n-ocserv-no - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Norsk (Norwegian)
luci-i18n-ocserv-pl - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Polski (Polish)
luci-i18n-ocserv-pt - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Português (Portuguese)
luci-i18n-ocserv-pt-br - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Português do Brasil (Brazilian Portuguese)
luci-i18n-ocserv-ro - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Română (Romanian)
luci-i18n-ocserv-ru - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Русский (Russian)
luci-i18n-ocserv-sk - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Slovenčina (Slovak)
luci-i18n-ocserv-sv - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Svenska (Swedish)
luci-i18n-ocserv-tr - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Türkçe (Turkish)
luci-i18n-ocserv-uk - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Українська (Ukrainian)
luci-i18n-ocserv-vi - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - Tiếng Việt (Vietnamese)
luci-i18n-ocserv-zh-cn - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - 简体中文 (Chinese Simplified)
luci-i18n-ocserv-zh-tw - git-22.362.26422-d6f8695 - Translation for luci-app-ocserv - 繁體中文 (Chinese Traditional)
openconnect - 9.01-1 - A VPN client compatible with several SSL VPN implementations (ocserv, Cisco AnyConnect, Juniper, Palo Alto)  OpenConnect is an SSL VPN client initially created to support Cisco's AnyConnect SSL VPN. It has since been extended to support the Pulse Connect Secure VPN (formerly known as Juniper Network Connect or Junos Pulse) and the Palo Alto Networks GlobalProtect SSL VPN.  A corresponding OpenConnect VPN server implementation can be found in the ocserv package.
root@router:~#

Spudz76, 2023-01-09:
Definitely no `ocserv` in that `Packages.gz`

Definitely still a [package for it](https://github.com/openwrt/packages/tree/openwrt-22.03/net/ocserv)

Now, why isn't it built and available...

1715173329, 2023-01-09:
https://downloads.openwrt.org/releases/faillogs-22.03/mipsel_24kc/packages/ocserv/compile.txt
it's failing for the same reason as #20226. @nmav could you please take a look?

1715173329, 2023-01-09:
https://downloads.openwrt.org/releases/faillogs-22.03/mipsel_24kc/packages/knot/compile.txt
and knot too @salzmdan

salzmdan, 2023-01-09:
> https://downloads.openwrt.org/releases/faillogs-22.03/mipsel_24kc/packages/knot/compile.txt and knot too @salzmdan

It seems libmaxmind package is incomplete (missing library file).

1715173329, 2023-01-09:
> > https://downloads.openwrt.org/releases/faillogs-22.03/mipsel_24kc/packages/knot/compile.txt and knot too @salzmdan
> 
> It seems libmaxmind package is incomplete (missing library file).

Nah, because of fe018482f83d51b9005c44d25652ea323aa338a2, now knot find it and depends on it. Is there any option to disable it explicitly like #20231

salzmdan, 2023-01-09:
> > > https://downloads.openwrt.org/releases/faillogs-22.03/mipsel_24kc/packages/knot/compile.txt and knot too @salzmdan
> > 
> > 
> > It seems libmaxmind package is incomplete (missing library file).
> 
> Nah, because of [fe01848](https://github.com/openwrt/packages/commit/fe018482f83d51b9005c44d25652ea323aa338a2), now knot finds it and depends on it. Is there any option to disable it explicitly like #20231 (or we can add it to dependencies if necessary)

I don't understand. If pkgconfig says the library is available, why the library file isn't there?!

Knot supports `--disable-maxminddb` but adding it to dependencies sounds better.

1715173329, 2023-01-09:
>> I don't understand. If pkgconfig says the library is available, why the library file isn't there?!

> Package knot is missing dependencies for the following libraries:
> libmaxminddb.so.0.0.7

Simply because `libmaxminddb` is not in the dependency list.

>> Knot supports --disable-maxminddb but adding it to dependencies sounds better.

Could you please create a PR for it? Thanks.

1715173329, 2023-01-11:
should be fixed by 69a8305f7cfd82dc00c0fde79963d5ba6fbf04ab

closed 2023-09-10 by nmav