openwrt/packages #20250: podman: panic: runtime error: invalid memory address or nil pointer dereference

petrkr, 2023-01-09:
Maintainer: Oskari Rauta
Environment: ARMv7 - Turris Omnia - OpenWRT 21.02 (Turris OS 6.2.1)
Description:

Seems same problem like this one https://github.com/containers/podman/issues/10535

Just wonder if update to latest 3.4.x podman version (https://github.com/containers/podman/releases?q=v3.4&expanded=true) would fix that. It is hard for me to set-up whole build environment to rebuild just one package to test it

Package makefile: https://git.openwrt.org/?p=feed/packages.git;a=blob;f=utils/podman/Makefile;h=e8323d4022e0b332a00a0ef4c7ba17e3fae0debf;hb=refs/heads/openwrt-21.02

i've tried overlay and btrfs (as there is Btrfs filesystem) storage engine, both are same.

```
root@turris:~# podman run -it --rm arm32v7/busybox
✔ docker.io/arm32v7/busybox:latest
Trying to pull docker.io/arm32v7/busybox:latest...
Getting image source signatures
Copying blob 46758452d3ee done  
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x200 pc=0x58e0a0]

goroutine 87 [running]:
reflect.mapiterinit(0x1bce3c0, 0x3f2a4c0, 0x416a4b0)
        runtime/map.go:1373 +0x2c
github.com/modern-go/reflect2.(*UnsafeMapType).UnsafeIterate(...)
        github.com/modern-go/reflect2@v1.0.1/unsafe_map.go:112
github.com/json-iterator/go.(*sortKeysMapEncoder).IsEmpty(0x3e88738, 0x3eec388)
        github.com/json-iterator/go@v1.1.11/reflect_map.go:333 +0x34
github.com/json-iterator/go.(*structFieldEncoder).IsEmpty(0x3e88f90, 0x3eec300)
        github.com/json-iterator/go@v1.1.11/reflect_struct_encoder.go:118 +0x50
github.com/json-iterator/go.(*structEncoder).Encode(0x3e89008, 0x3eec300, 0x416a4b0)
        github.com/json-iterator/go@v1.1.11/reflect_struct_encoder.go:148 +0x63c
github.com/json-iterator/go.(*OptionalEncoder).Encode(0x3ee8838, 0x3ee8158, 0x416a4b0)
        github.com/json-iterator/go@v1.1.11/reflect_optional.go:70 +0x104
github.com/json-iterator/go.(*sliceEncoder).Encode(0x4061fc0, 0x4028728, 0x416a4b0)
        github.com/json-iterator/go@v1.1.11/reflect_slice.go:38 +0x37c
github.com/json-iterator/go.(*OptionalEncoder).Encode(0x3ee8860, 0x3ee88a0, 0x416a4b0)
        github.com/json-iterator/go@v1.1.11/reflect_optional.go:70 +0x104
github.com/json-iterator/go.(*onePtrEncoder).Encode(0x3ee8868, 0x4028728, 0x416a4b0)
        github.com/json-iterator/go@v1.1.11/reflect.go:219 +0x8c
github.com/json-iterator/go.(*Stream).WriteVal(0x416a4b0, {0x1b4ac50, 0x4028728})
        github.com/json-iterator/go@v1.1.11/reflect.go:98 +0x1d4
github.com/json-iterator/go.(*frozenConfig).Marshal(0x3f9c540, {0x1b4ac50, 0x4028728})
        github.com/json-iterator/go@v1.1.11/config.go:299 +0xb4
github.com/containers/storage.(*layerStore).saveLayers(0x4028700)
        github.com/containers/storage@v1.36.0/layers.go:484 +0x150
github.com/containers/storage.(*layerStore).Save(0x4028700)
        github.com/containers/storage@v1.36.0/layers.go:467 +0xd8
github.com/containers/storage.(*layerStore).Put(0x4028700, {0x413a327, 0x40}, 0x0, {0x0, 0x0, 0x0}, {0x0, 0x0}, 0x0, ...)
        github.com/containers/storage@v1.36.0/layers.go:799 +0x1834
github.com/containers/storage.(*store).PutLayer(0x3cc80c0, {0x413a327, 0x40}, {0x0, 0x0}, {0x0, 0x0, 0x0}, {0x0, 0x0}, ...)
        github.com/containers/storage@v1.36.0/store.go:1057 +0xc38
github.com/containers/image/v5/storage.(*storageImageDestination).commitLayer(0x409e6e0, {0x1d5f65c, 0x3ec31a0}, {{{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, 0x0}, 0x0, ...}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/storage/storage_image.go:1010 +0xeb0
github.com/containers/image/v5/storage.(*storageImageDestination).queueOrCommit(0x409e6e0, {0x1d5f65c, 0x3ec31a0}, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, 0x0}, 0x0, ...}, ...)
        github.com/containers/image/v5@v5.16.0/storage/storage_image.go:837 +0x1dc
github.com/containers/image/v5/storage.(*storageImageDestination).PutBlobWithOptions(0x409e6e0, {0x1d5f65c, 0x3ec31a0}, {0x1d55dd4, 0x3eddd60}, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, ...}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/storage/storage_image.go:466 +0x11c
github.com/containers/image/v5/copy.(*copier).copyBlobFromStream(0x3eeea00, {0x1d5f65c, 0x3ec31a0}, {0x1d5779c, 0x3df4d20}, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, ...}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/copy/copy.go:1632 +0x1b34
github.com/containers/image/v5/copy.(*imageCopier).copyLayerFromStream(0x3ec2180, {0x1d5f65c, 0x3ec31a0}, {0x1d5779c, 0x3df4d20}, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, ...}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/copy/copy.go:1362 +0x348
github.com/containers/image/v5/copy.(*imageCopier).copyLayer.func3(0x3ec2180, 0x416ab40, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, 0x0}, 0x0, {0x40e0640, ...}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/copy/copy.go:1305 +0x19c
github.com/containers/image/v5/copy.(*imageCopier).copyLayer(0x3ec2180, {0x1d5f65c, 0x3ec31a0}, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, 0x0}, 0x0, ...}, ...)
        github.com/containers/image/v5@v5.16.0/copy/copy.go:1329 +0x7b0
github.com/containers/image/v5/copy.(*imageCopier).copyLayers.func1(0x0, {{0x413a5f0, 0x47}, 0x19aa38, {0x0, 0x0, 0x0}, 0x0, {0x40e0640, 0x31}, ...}, ...)
        github.com/containers/image/v5@v5.16.0/copy/copy.go:891 +0x338
created by github.com/containers/image/v5/copy.(*imageCopier).copyLayers.func2
        github.com/containers/image/v5@v5.16.0/copy/copy.go:928 +0x338
root@turris:~# 

```

dangowrt, 2023-01-29:
@oskarirauta

oskarirauta, 2023-01-29:
©petrkr

Not sure, but didn't 3.4.1 fix this issue? Or some of the newer ones? My webserver (yes, my site's server runs on openwrt) - has been running with podman 3.4.4 ever since I updated to that version. There's weekly reboots on server, but it has nothing to do with podman.... Should update the whole system.

Btw. why is mine using cgroupv1 when you clearly have cgroupv2..?

Otherwise your setup seems just like mine that runs without issues except that is a bit older, and:
 - cgroup version
 - I have slirp ( but as we run these with root it doesn't have anything.. )
 - my storage driver is extfs instead of btrfs, but that's propably more related on your choice of filesystem.

One thing though pops... You are using runc as oci runtime. Would you mind trying first with crun, as it's recommended for podman over runc.

oskarirauta, 2023-01-29:
But try switch to crun first.

petrkr, 2023-01-29:
I did not mentioned it, but I've triee both, runc and crun, same results

oskarirauta, 2023-01-29:
Ok, so you need a updated version then. I can verify that 3.4.4 works fine.

petrkr, 2023-01-29:
> Ok, so you need a updated version then. I can verify that 3.4.4 works fine.

Can not update as package is not compiled. That is what I wrote on original issue.. if someone have working build environment, maybe can put and build 3.4.4 and I can test. But in 21.02 repo is still 3.4.1

oskarirauta, 2023-01-29:
I have, but wrong arch- mine is x86-64.. There's though even newer version available, though I am not sure if other system depencies are filled-

Snapshots: [podman_4.3.1-1](https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9/packages/podman_4.3.1-1_arm_cortex-a9.ipk)

petrkr, 2023-01-30:
```
# opkg install /root/podman_4.3.1-1_arm_cortex-a9.ipk 
Package podman (3.4.1-1) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libncursesw for libreadline
 * pkg_hash_fetch_best_installation_candidate: Packages for libreadline found, but incompatible with the architectures configured
```

yes, there will be dependency hell as this package will be probably from 22/master branch.

oskarirauta, 2023-01-30:
That is possible. I pushed it few weeks ago to master.

paper42, 2023-01-30:
Updating to 3.4.7 helps, but I am still having issues:
```
root@turris:~# podman run -it --rm arm32v7/busybox
✔ docker.io/arm32v7/busybox:latest
Trying to pull docker.io/arm32v7/busybox:latest...
Getting image source signatures
Copying blob 46758452d3ee done  
Copying config 1d57ab16f6 done  
Writing manifest to image destination
Storing signatures
Error: OCI runtime error: prctl: Invalid argument
```
and this on any subsequent runs:
```
Error: container create failed (no logs from conmon): EOF
```

I didn't manage to build 4.3.1 on OpenWrt 21.02 even though gpg-error.pc is in `build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libgpg-error-1.45/ipkg-install/usr/lib/pkgconfig/gpg-error.pc`:
```
# /build/build/staging_dir/host/bin/pkg-config --cflags  -- gpgme
Package gpg-error was not found in the pkg-config search path.
Perhaps you should add the directory containing `gpg-error.pc'
to the PKG_CONFIG_PATH environment variable
Package 'gpg-error', required by 'gpgme', not found
Package 'gpg-error', required by 'libassuan', not found
```

petrkr, 2023-01-31:
> Updating to 3.4.7 helps, but I am still having issues:
> 

Can you please share that ipkg/opkg ? I will try to check this other stuff... But I hope it's not kernel config related

paper42, 2023-01-31:
> > Updating to 3.4.7 helps, but I am still having issues:
> 
> Can you please share that ipkg/opkg ? I will try to check this other stuff... But I hope it's not kernel config related

[podman_3.4.7-1_arm_cortex-a9_vfpv3-d16.zip](https://github.com/openwrt/packages/files/10547019/podman_3.4.7-1_arm_cortex-a9_vfpv3-d16.zip) GitHub didn't allow me to upload an .ipk file, so just change the extension from zip to ipk.

petrkr, 2023-01-31:
Well. this is at-least interesting error message in syslog

```
Jan 31 14:42:03 turris modprobe: no module folders for kernel version 5.15.88 found
```

but what kind of kernel module or folder it looking for ?

petrkr, 2023-01-31:
well, kernel module was just because OpenWRT updated to new kernel without reboot. So ignore that message.

oskarirauta, 2023-01-31:
Latest conmon is 2.1.15 - available at snapshots.

petrkr, 2023-01-31:
> Latest conmon is 2.1.5 - available at snapshots.

Again, seems I can not use snapshot version in 21.02 OpenWRT

```
root@turris:~# conmon --version
Error relocating /usr/bin/conmon: __clock_gettime64: symbol not found
Error relocating /usr/bin/conmon: __localtime64_r: symbol not found
Error relocating /usr/bin/conmon: __dlsym_time64: symbol not found
```

paper42, 2023-02-01:
conmon 2.1.5 and libseccomp 2.5.2 which was required for the build for OpenWrt 21.02:
[conmon_2.1.5-1_arm_cortex-a9_vfpv3-d16.zip](https://github.com/openwrt/packages/files/10555095/conmon_2.1.5-1_arm_cortex-a9_vfpv3-d16.zip)
[libseccomp_2.5.2-1_arm_cortex-a9_vfpv3-d16.zip](https://github.com/openwrt/packages/files/10555096/libseccomp_2.5.2-1_arm_cortex-a9_vfpv3-d16.zip)

I am still getting the `Error: OCI runtime error: prctl: Invalid argument` error

petrkr, 2023-02-01:
There could be missing something in kernel. I check that debug message and it can not get terminal info, but which terminal? Container's or system (turris one). 

I also tried all three runc, crun, uxc... actually uxc ends with "I/O Error" but since they missing lot of debug prints, I really do not know which part of 2000 lines C file ends with return -1

oskarirauta, 2023-02-01:
remember to put uxc to different group, if I remember correctly, it doesn't support json like those 2 others- but nevertheless, uxc won't work anyway, I tested that sometime ago as well..