openwrt/packages #20251: atlas-sw-probe: update to 5080

maurice-w, 2023-01-09:
Maintainer: @ja-pa
Environment: x86_64, generic, 22.03.3

Description:
After many delays, RIPE NCC released the source code of version 5080 in 11/2022. Can we expect an update of the OpenWrt package? Thanks!

BKPepe, 2023-01-16:
Done in https://github.com/openwrt/packages/pull/20329

closed 2023-01-16 by BKPepe

maurice-w, 2023-01-18:
Thank @BKPepe! Does this actually work for you? My probes won't start anymore after updating to 5080:
```
root@atlasprobe:~# /etc/init.d/atlas info
{
        "atlas": {
                "instances": {
                        "instance1": {
                                "running": false,
                                "command": [
                                        "/usr/libexec/atlas-probe-scripts/bin/ATLAS"
                                ],
                                "term_timeout": 5,
                                "exit_code": 2
                        }
                }
        }
}
```
5040 worked fine on OpenWrt, so does 5080 on CentOS.