openwrt/packages #20272: syslog-ng: /usr/lib/libssl.so: file not recognized: file format not recognized

DocMAX, 2023-01-12:
latest archlinux / openwrt / packages:
```
ket-dest.lo modules/afsocket/libafsocket_la-socket-options.lo modules/afsocket/libafsocket_la-transport-mapper.lo modules/afsocket/libafsocket_la-afinet.lo modules/afsocket/libafsocket_la-afinet-source.lo modules/afsocket/libafsocket_la-afinet-dest.lo modules/afsocket/libafsocket_la-afinet-dest-failover.lo modules/afsocket/libafsocket_la-socket-options-inet.lo modules/afsocket/libafsocket_la-socket-options-unix.lo modules/afsocket/libafsocket_la-transport-mapper-inet.lo modules/afsocket/libafsocket_la-afunix-source.lo modules/afsocket/libafsocket_la-afunix-dest.lo modules/afsocket/libafsocket_la-transport-mapper-unix.lo modules/afsocket/libafsocket_la-transport-unix-socket.lo modules/afsocket/libafsocket_la-compat-unix-creds.lo modules/afsocket/libafsocket_la-afsocket-grammar.lo modules/afsocket/libafsocket_la-afsocket-parser.lo modules/afsocket/libafsocket_la-afsocket-plugin.lo modules/afsocket/libafsocket_la-systemd-syslog-source.lo ./lib/libsyslog-ng.la -L/mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib -lssl -lcrypto -inst-prefix-dir /mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/syslog-ng-3.38.1/ipkg-install)
/usr/lib/libssl.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
libtool:   error: error: relink 'modules/afsocket/libafsocket.la' with the above command before installing it
make[6]: *** [Makefile:11159: install-moduleLTLIBRARIES] Error 1
make[5]: *** [Makefile:26764: install-am] Error 2
make[4]: *** [Makefile:24985: install-recursive] Error 1
make[3]: *** [Makefile:26757: install] Error 2
make[3]: Leaving directory '/mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/syslog-ng-3.38.1'
make[2]: *** [Makefile:162: /mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/syslog-ng-3.38.1/.built] Error 2
make[2]: Leaving directory '/mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/feeds/packages/admin/syslog-ng'
time: package/feeds/packages/syslog-ng/compile#3.55#1.26#4.36
    ERROR: package/feeds/packages/syslog-ng failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/syslog-ng/compile] Error 1
make[1]: Leaving directory '/mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530'
make: *** [/mnt/zeus/hdd/sources/openwrt/sources/snapshot_fb7530/include/toplevel.mk:231: package/feeds/packages/syslog-ng/compile] Error 2

```

neheb, 2023-01-17:
pkgconfig bug probably.

aleks-mariusz, 2025-02-28:
i've just hit this same issue (also with syslog-ng):

<details><summary>my errors (am using master branch)</summary>

```
libtool: install: /home/user/workdir/openwrt-pesa/staging_dir/host/bin/install -c modules/affile/.libs/libaffile.soT /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.
1/ipkg-install/usr/lib/syslog-ng/libaffile.so
libtool: install: /home/user/workdir/openwrt-pesa/staging_dir/host/bin/install -c modules/affile/.libs/libaffile.lai /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.
1/ipkg-install/usr/lib/syslog-ng/libaffile.la
libtool: warning: relinking 'modules/afprog/libafprog.la'
libtool: install: (cd /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1; /bin/sh "/home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/
syslog-ng-4.7.1/libtool"  --silent --tag CC --mode=relink arm-openwrt-linux-muslgnueabi-gcc -Wshadow -Wno-stack-protector -Wno-unused-parameter -Wno-variadic-macros -O2 -g -pthread -O2 -pipe -mthumb -fno-caller-sav
es -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1=syslog-ng-4.7.1 -Wformat -Werror=format-security -DPI
C -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -fstack-clash-protection -fwrapv -avoid-version -module -no-undefined -L/home/user/workdir/openwrt-pesa/staging_dir/toolchain-arm_cortex-
a9+vfpv3-d16_gcc-13.3.0_musl_eabi/usr/lib -L/home/user/workdir/openwrt-pesa/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-13.3.0_musl_eabi/lib -fuse-ld=bfd -DPIC -fpic -specs=/home/user/workdir/openwrt-pesa/inc
lude/hardened-ld-pie.specs -znow -zrelro -o modules/afprog/libafprog.la -rpath /usr/lib/syslog-ng modules/afprog/libafprog_la-afprog.lo modules/afprog/libafprog_la-afprog-grammar.lo modules/afprog/libafprog_la-afpr
og-parser.lo modules/afprog/libafprog_la-afprog-plugin.lo ./lib/libsyslog-ng.la -inst-prefix-dir /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1/ipkg-install)
libtool: install: /home/user/workdir/openwrt-pesa/staging_dir/host/bin/install -c modules/afprog/.libs/libafprog.soT /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.
1/ipkg-install/usr/lib/syslog-ng/libafprog.so
libtool: install: /home/user/workdir/openwrt-pesa/staging_dir/host/bin/install -c modules/afprog/.libs/libafprog.lai /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.
1/ipkg-install/usr/lib/syslog-ng/libafprog.la
libtool: warning: relinking 'modules/afsocket/libafsocket.la'
libtool: install: (cd /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1; /bin/sh "/home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/
syslog-ng-4.7.1/libtool"  --silent --tag CC --mode=relink arm-openwrt-linux-muslgnueabi-gcc -Wshadow -Wno-stack-protector -Wno-unused-parameter -Wno-variadic-macros -O2 -g -pthread -O2 -pipe -mthumb -fno-caller-sav
es -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1=syslog-ng-4.7.1 -Wformat -Werror=format-security -DPI
C -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -fstack-clash-protection -fwrapv -avoid-version -module -no-undefined -L/home/user/workdir/openwrt-pesa/staging_dir/toolchain-arm_cortex-
a9+vfpv3-d16_gcc-13.3.0_musl_eabi/usr/lib -L/home/user/workdir/openwrt-pesa/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-13.3.0_musl_eabi/lib -fuse-ld=bfd -DPIC -fpic -specs=/home/user/workdir/openwrt-pesa/inc
lude/hardened-ld-pie.specs -znow -zrelro -o modules/afsocket/libafsocket.la -rpath /usr/lib/syslog-ng modules/afsocket/libafsocket_la-afsocket.lo modules/afsocket/libafsocket_la-afsocket-source.lo modules/afsocket/
libafsocket_la-afsocket-dest.lo modules/afsocket/libafsocket_la-socket-options.lo modules/afsocket/libafsocket_la-transport-mapper.lo modules/afsocket/libafsocket_la-afinet.lo modules/afsocket/libafsocket_la-afinet
-source.lo modules/afsocket/libafsocket_la-afinet-dest.lo modules/afsocket/libafsocket_la-afinet-dest-failover.lo modules/afsocket/libafsocket_la-socket-options-inet.lo modules/afsocket/libafsocket_la-socket-option
s-unix.lo modules/afsocket/libafsocket_la-transport-mapper-inet.lo modules/afsocket/libafsocket_la-afunix-source.lo modules/afsocket/libafsocket_la-afunix-dest.lo modules/afsocket/libafsocket_la-transport-mapper-un
ix.lo modules/afsocket/libafsocket_la-transport-unix-socket.lo modules/afsocket/libafsocket_la-compat-unix-creds.lo modules/afsocket/libafsocket_la-afsocket-grammar.lo modules/afsocket/libafsocket_la-afsocket-parse
r.lo modules/afsocket/libafsocket_la-afsocket-plugin.lo modules/afsocket/libafsocket_la-systemd-syslog-source.lo ./lib/libsyslog-ng.la -L/home/user/workdir/openwrt-pesa/staging_dir/target-arm_cortex-a9+vfpv3-d16_mu
sl_eabi/usr/lib -lssl -lcrypto -inst-prefix-dir /home/user/workdir/openwrt-pesa/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/syslog-ng-4.7.1/ipkg-install)
/usr/lib/libssl.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
[... cut ...]
make[1]: *** [package/Makefile:183: /home/user/workdir/openwrt-pesa/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/workdir/openwrt-pesa'
make: *** [/home/user/workdir/openwrt-pesa/include/toplevel.mk:233: world] Error 2
```
</details>

any idea what would the fix be?

I am building in [my custom docker container](https://github.com/aleks-mariusz/docker-openwrt-builder/tree/alpine), which normally works great to rebuild - and here's the [config.buildinfo](https://divested.dev/unofficial-openwrt-builds/mvebu-linksys/20250213-00/config.buildinfo) i am using.