openwrt/packages #20291: openssh-server: selecting openssh-server-pam does not actually build the server with pam support.

apilleri, 2023-01-13:
Maintainer: OpenWrt team 
Environment: not related to env, Current Version

Description:

I'm trying to use the openssh-server-pam package. When I select the -pam package and build openwrt, the toolchain keeps on building the openssh-server with configure args --without-pam, instead of --with-pam.

I think that there is a problem with the openssh/Makefile:
- the default package (Package/openssh/Default) set the variable VARIANT:=without-pam
- the VARIANT variable is then overrided in the -pam definition
- this override is not actually working, and the env variable BUILD_VARIANT (which is used later to set proper configure args) is set to "without-pam" regardless of the override
- the solution is to remove VARIANT:=without-pam from the default package and set it only in openssh-server and openssh-server-pam

This solution works for me:
```
diff --git a/net/openssh/Makefile b/net/openssh/Makefile
index cc5c27b91..60de25454 100644
--- a/net/openssh/Makefile
+++ b/net/openssh/Makefile
@@ -36,7 +36,6 @@ define Package/openssh/Default
        MAINTAINER:=Peter Wagner <tripolar@gmx.at>
        URL:=http://www.openssh.com/
        SUBMENU:=SSH
-       VARIANT:=without-pam
 endef

 define Package/openssh-moduli
@@ -88,6 +87,7 @@ define Package/openssh-server
        $(call Package/openssh/Default)
        DEPENDS+= +openssh-keygen +OPENSSH_LIBFIDO2:libfido2
        TITLE+= server
+       VARIANT:=without-pam
        USERID:=sshd=22:sshd=22
 endef
 ```

Am I missing something or this is actually a bug?

1715173329, 2023-02-02:
Cc: @SibrenVasse @nehen

neheb, 2023-02-03:
sounds like it.

PR #20438 (2023-02-03): openssh: update to 9.2p1
Maintainer: @tripolar
Compile tested: all architectures / master
Run tested: MIR4A / ramips / master

Server running and accepting connections
Client connecting to servers

Description:
Version bump to 9.2p1
Fix openssh-server-pam build (Fixes #20291)
closed 2023-02-03 by neheb