openwrt/packages #20293: strongswan: plugin 'uci' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: uci_lookup:

DocMAX, 2023-01-14:
```
Jan 14 00:25:26 router ipsec: 00[DMN] Starting IKE charon daemon (strongSwan 5.9.8, Linux 5.15.87, armv7l)
Jan 14 00:25:26 router ipsec: 00[LIB] plugin 'uci' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: uci_lookup: symbol not found
Jan 14 00:25:26 router ipsec: 00[NET] using forecast interface br-lan
Jan 14 00:25:26 router ipsec: 00[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Jan 14 00:25:26 router ipsec: 00[LIB] loaded plugins: charon aes des rc2 sha2 sha1 md4 md5 mgf1 random nonce x509 revocation constraints pubkey pkcs1 pgp dnskey sshkey pem fips-prf gmp xcbc hmac kdf drbg attr kernel-netlink resolve socket-default connmark forecast vici updown eap-identity eap-mschapv2 xauth-generic dhcp led
Jan 14 00:25:26 router ipsec: 00[JOB] spawning 16 worker threads

```

RobsonRojas, 2023-05-24:
There are any patch to fix this?

kevinoid, 2024-12-01:
I believe this issue is a duplicate of https://github.com/openwrt/packages/issues/9842

Looks like there's an (untested) patch in https://github.com/openwrt/packages/issues/9842#issuecomment-2158827260