openwrt/packages #20339: strongswan-ipsec: Build fails

ColinTimBarndt, 2023-01-18:
Maintainer: @pprindeville
Environment: ARMv7 Processor rev 5 (v7l), GL.iNet GL-B1300 (ipq40xx), OpenWRT master branch 1cefb14230

Description: Build fails

`install: cannot stat '/home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-install/etc/ipsec.conf': No such file or directory`

<details><summary>Build Errors: <code>make -j1 V=s</code></summary>

```
...

make[3]: Entering directory '/home/colin/Documents/code/openwrt/feeds/packages/net/strongswan'
mkdir -p /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/colin/Documents/code/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/.pkgdir/strongswan/. /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.strongswan_installed
mkdir -p /home/colin/Documents/code/openwrt/bin/targets/ipq40xx/generic/packages /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/CONTROL /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
install -d -m0755 /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/etc/ /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/usr/sbin
install -m0600 /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-install/etc/ipsec.conf /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/etc/
install: cannot stat '/home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-install/etc/ipsec.conf': No such file or directory
make[3]: *** [Makefile:669: /home/colin/Documents/code/openwrt/bin/packages/arm_cortex-a7_neon-vfpv4/packages/strongswan-ipsec_5.9.7-8.1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[3]: Leaving directory '/home/colin/Documents/code/openwrt/feeds/packages/net/strongswan'
time: package/feeds/packages/strongswan/compile#0.40#0.22#0.54
    ERROR: package/feeds/packages/strongswan failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/strongswan/compile] Error 1
make[2]: Leaving directory '/home/colin/Documents/code/openwrt'
make[1]: *** [package/Makefile:110: /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/colin/Documents/code/openwrt'
make: *** [/home/colin/Documents/code/openwrt/include/toplevel.mk:231: world] Error 2
```

</details>

<details><summary><code>cat .config | grep ^CONFIG_PACKAGE_strongswan</code></summary>

```
CONFIG_PACKAGE_strongswan=y
CONFIG_PACKAGE_strongswan-charon=y
CONFIG_PACKAGE_strongswan-charon-cmd=y
CONFIG_PACKAGE_strongswan-ipsec=y
CONFIG_PACKAGE_strongswan-libtls=y
CONFIG_PACKAGE_strongswan-mod-aes=y
CONFIG_PACKAGE_strongswan-mod-curve25519=y
CONFIG_PACKAGE_strongswan-mod-des=y
CONFIG_PACKAGE_strongswan-mod-eap-identity=y
CONFIG_PACKAGE_strongswan-mod-eap-md5=y
CONFIG_PACKAGE_strongswan-mod-eap-mschapv2=y
CONFIG_PACKAGE_strongswan-mod-eap-tls=y
CONFIG_PACKAGE_strongswan-mod-kernel-libipsec=y
CONFIG_PACKAGE_strongswan-mod-led=y
CONFIG_PACKAGE_strongswan-mod-md4=y
CONFIG_PACKAGE_strongswan-mod-pem=y
CONFIG_PACKAGE_strongswan-mod-pkcs1=y
CONFIG_PACKAGE_strongswan-mod-pkcs11=y
CONFIG_PACKAGE_strongswan-mod-pkcs12=y
CONFIG_PACKAGE_strongswan-mod-pkcs7=y
CONFIG_PACKAGE_strongswan-mod-pkcs8=y
CONFIG_PACKAGE_strongswan-mod-pubkey=y
CONFIG_PACKAGE_strongswan-mod-sha1=y
CONFIG_PACKAGE_strongswan-mod-sha2=y
CONFIG_PACKAGE_strongswan-mod-uci=y
CONFIG_PACKAGE_strongswan-mod-unity=y
CONFIG_PACKAGE_strongswan-mod-vici=y
CONFIG_PACKAGE_strongswan-swanctl=y
```

</details>

Relating the error with the Makefile:

```mk
541| define Package/strongswan-ipsec/install
542| 	$(INSTALL_DIR) $(1)/etc/ $(1)/usr/sbin
#> install -d -m0755 ~openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/etc/ ~openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/usr/sbin
543| 	$(INSTALL_CONF) $(PKG_INSTALL_DIR)/etc/ipsec.conf $(1)/etc/
#> install -m0600 ~openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-install/etc/ipsec.conf ~openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-arm_cortex-a7_neon-vfpv4/strongswan-ipsec/etc/
#> install: cannot stat '~openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.9.7/ipkg-install/etc/ipsec.conf': No such file or directory
544| 	echo -e "\ninclude /var/ipsec/ipsec.conf" >> $(1)/etc/ipsec.conf
545| 	$(INSTALL_CONF) ./files/ipsec.secrets $(1)/etc/
546| 	echo -e "\ninclude /var/ipsec/ipsec.secrets" >> $(1)/etc/ipsec.secrets
547| 	$(INSTALL_CONF) ./files/ipsec.user $(1)/etc/
548| 	$(INSTALL_DIR) $(1)/etc/init.d
549| 	$(INSTALL_BIN) ./files/ipsec.init $(1)/etc/init.d/ipsec
550| 	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/ipsec $(1)/usr/sbin/
551| endef
```

Related issues:
- https://forum.openwrt.org/t/unable-to-build-strongswan-on-19-07-2/62384
- #10879

ColinTimBarndt, 2023-01-18:
<details><summary>Diffconfig</summary>

```
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_glinet_gl-b1300=y
CONFIG_DROPBEAR_ECC=y
CONFIG_DROPBEAR_ECC_FULL=y
CONFIG_EXPERIMENTAL=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_LUA_ECO_DEFAULT_OPENSSL=y
CONFIG_LUCI_SRCDIET=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENVPN_openssl_ENABLE_DEF_AUTH=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_MULTIHOME=y
CONFIG_OPENVPN_openssl_ENABLE_PF=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSans=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSans-Bold=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSans-BoldOblique=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSans-Oblique=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSansMono=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSansMono-Bold=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSerif=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSerif-Bold=y
CONFIG_PACKAGE_dejavu-fonts-ttf-DejaVuSerif-Italic=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_ip-tiny=y
CONFIG_PACKAGE_kmod-crypto-authenc=y
CONFIG_PACKAGE_kmod-crypto-cbc=y
CONFIG_PACKAGE_kmod-crypto-deflate=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-echainiv=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-sha1=y
CONFIG_PACKAGE_kmod-ipsec=y
CONFIG_PACKAGE_kmod-ipsec4=y
CONFIG_PACKAGE_kmod-ipsec6=y
CONFIG_PACKAGE_kmod-iptunnel4=y
CONFIG_PACKAGE_kmod-iptunnel6=y
CONFIG_PACKAGE_kmod-lib-zlib-deflate=y
CONFIG_PACKAGE_kmod-lib-zlib-inflate=y
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-tun=y
[... 99 lines cut ...]
CONFIG_PACKAGE_strongswan-mod-uci=y
CONFIG_PACKAGE_strongswan-mod-unity=y
CONFIG_PACKAGE_strongswan-mod-vici=y
CONFIG_PACKAGE_strongswan-swanctl=y
CONFIG_PACKAGE_sudo=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_vpnc=y
CONFIG_PACKAGE_vpnc-scripts=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_xfrm=y
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
CONFIG_VPNC_GNUTLS=y
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
# CONFIG_LUA_ECO_OPENSSL is not set
CONFIG_LUA_ECO_WOLFSSL=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

</details>

hayzamjs, 2023-04-11:
Same issue `ipkg-install/etc/ipsec.conf': No such file or directory`

pprindeville, 2023-04-14:
@ColinTimBarndt Is this still going on?

pprindeville, 2023-04-14:
And if you add this:

```
define Build/Install
       $(INSTALL_DIR) $(PKG_INSTALL_DIR)/etc
       $(call Build/Install/Default,install)
endef
```

does the problem go away?

ColinTimBarndt, 2023-04-15:
> @ColinTimBarndt Is this still going on?
> 

I am going to rerun the build script today to check

ColinTimBarndt, 2023-04-15:
> And if you add this:
> 
> ```
> define Build/Install
>        $(INSTALL_DIR) $(PKG_INSTALL_DIR)/etc
>        $(call Build/Install/Default,install)
> endef
> ```
> 
> does the problem go away?

Where exactly should this be added?

ColinTimBarndt, 2023-04-16:
I have attempted to build OpenWRT with strongswan but am currently getting an error not related to strongswan (probably earlier).

Last commit: `[72780e3eac] ipq40xx: convert AP-365 to DSA`

<details><summary>Build output<pre>make -j1 V=s</pre></summary>

```
make[2]: Entering directory '/home/colin/Documents/code/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/colin/Documents/code/openwrt/scripts/config'
time: target/linux/prereq#0.30#0.05#0.35
make[1]: Entering directory '/home/colin/Documents/code/openwrt'
make[2]: Entering directory '/home/colin/Documents/code/openwrt'
+ mkdir -p /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi
+ cd /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/colin/Documents/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp
touch /home/colin/Documents/code/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/.prepared
make[3]: Entering directory '/home/colin/Documents/code/openwrt/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/colin/Documents/code/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/colin/Documents/code/openwrt/toolchain/binutils'
make  -C /home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40 all
make[4]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40'
make[5]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40'
make[6]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libiberty'
make[7]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libiberty/testsuite'
make[6]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libiberty'
make[6]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/intl'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/intl'
make[6]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libsframe'
make  all-am
make[7]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libsframe'
make[7]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libsframe'
make[6]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/libsframe'
make[6]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
Making info in po
make[7]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd/po'
make[7]: Nothing to be done for 'info'.
make[7]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd/po'
make[7]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
make[7]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[7]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
make  all-recursive
make[7]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
Making all in po
make[8]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd/po'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd/po'
make[8]: Entering directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
make[8]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[8]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
make[7]: Leaving directory '/home/colin/Documents/code/openwrt/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/binutils-2.40/bfd'
[... cut ...]
    ERROR: toolchain/binutils failed to build.
make[2]: *** [toolchain/Makefile:93: toolchain/binutils/compile] Error 1
make[2]: Leaving directory '/home/colin/Documents/code/openwrt'
make[1]: *** [toolchain/Makefile:89: /home/colin/Documents/code/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-12.2.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/colin/Documents/code/openwrt'
make: *** [/home/colin/Documents/code/openwrt/include/toplevel.mk:231: world] Error 2
```

</details>

pprindeville, 2023-04-17:
> > And if you add this:
> > ```
> > define Build/Install
> >        $(INSTALL_DIR) $(PKG_INSTALL_DIR)/etc
> >        $(call Build/Install/Default,install)
> > endef
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > does the problem go away?
> 
> Where exactly should this be added?

After all of the other define’s should be fine.