openwrt/packages #20340: zerotier: symbol not found

cubercsl, 2023-01-18:
Maintainer: @mwarning 
Environment: 

 --- | Value
---|---
Model | Xiaomi Redmi Router AC2100
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598
Kernel Version | 5.10.161

Description:

After I upgraded my firmware by attendedsysupgrade, zerotier failed to start due to symbol not found.

```console
root@OpenWrt:~# ldd /usr/bin/zerotier-cli
        /lib/ld-musl-mipsel-sf.so.1 (0x77d80000)
        libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0x77d66000)
        libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0x77d53000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x77c09000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77be5000)
        libc.so => /lib/ld-musl-mipsel-sf.so.1 (0x77d80000)
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructEjc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEjjPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEjjRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEaSEOS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13runtime_errorC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEjPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5eraseEN9__gnu_cxx17__normal_iteratorIPKcS4_EES9_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEjjPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13random_device7_M_initERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx117collateIcE9transformEPKcS3_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4copyEPcjj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1EOS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZSt9use_facetINSt7__cxx117collateIcEEERKT_RKSt6locale: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_S_copyEPcPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6substrEjj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1ERKNSt7__cxx1112basic_stringIcS1_SaIcEEESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findERKS4_j: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE14_M_replace_auxEjjjc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE3strEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7replaceEjjPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEjj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13find_first_ofEPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEjc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEjRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_disposeEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcS5_S5_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERjj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9push_backEc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1ERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEcj: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringIS4_S5_T1_EE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEaSEPKc: symbol not found
```

After I download OpenWrt SDK (openwrt-sdk-22.03.3-ramips-mt7621_gcc-11.2.0_musl.Linux-x86_64) and rebuild it from source, it works fine.

Looks like the package in the repositories is out of date。

1715173329, 2023-01-18:
Nah, it's broken because it's "up-to-date".
https://github.com/openwrt/openwrt/commit/c0b4303d2e2f4a9e1d4684fd584e6b6548666f0f changed stdcpp abi, all packages depending on libstdcpp may be broken on 22.03.3 and earlier.

Cc @ynezz

ynezz, 2023-01-18:
Thanks for the report, sorry for the breakage, going to fix it ASAP.

ynezz, 2023-01-18:
Should be fixed:

```shell
$ docker run --rm -it openwrt/rootfs:x86-64-22.03.3 sh -c "mkdir -p /var/lock && opkg update && opkg install zerotier && ldd /usr/bin/zerotier-cli" 2>&1 | tail -5
	libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0x7fbf937a6000)
	libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0x7fbf937a1000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7fbf9366e000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fbf9365a000)
	libc.so => /lib/ld-musl-x86_64.so.1 (0x7fbf937b3000)
```

Can you check?

0anton, 2023-01-18:
Same here for `rockchip/armv8` platform on _NanoPi R4S_.

```
root@gw:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="22.03.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 22.03.2"
VERSION_ID="22.03.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r19803-9a599fee93"
OPENWRT_BOARD="rockchip/armv8"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 22.03.2 r19803-9a599fee93"
```

root@gw:~# opkg list zerotier
zerotier - 1.10.2-1 - ZeroTier creates a global provider-independent virtual private cloud network.

root@gw:~# uname -a
Linux gw 5.15.78 #1 SMP PREEMPT Fri Dec 2 20:34:27 CST 2022 aarch64 GNU/Linux

```
root@gw:~# ldd /usr/bin/zerotier-cli
        /lib/ld-musl-aarch64.so.1 (0xffff9a19b000)
        libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0xffff9a17e000)
        libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0xffff9a16b000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xffff99fd9000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xffff99fb4000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0xffff9a19b000)
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEmRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6substrEmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEaSEOS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13runtime_errorC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5eraseEN9__gnu_cxx17__normal_iteratorIPKcS4_EES9_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt13random_device7_M_initERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx117collateIcE9transformEPKcS3_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEmc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKc: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_S_copyEPcPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1EOS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZSt9use_facetINSt7__cxx117collateIcEEERKT_RKSt6locale: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1ERKNSt7__cxx1112basic_stringIcS1_SaIcEEESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEmmRKS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEPKcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEcm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE3strEv: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendERKS4_: symbol not found
[... 7 lines cut ...]
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9push_backEc: symbol not found
[... cut ...]
Error relocating /usr/bin/zerotier-cli: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4copyEPcmm: symbol not found
Error relocating /usr/bin/zerotier-cli: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEaSEPKc: symbol not found
```

```
root@gw:~# opkg files libstdcpp
Package libstdcpp6 (11.2.0-4) is installed on root and has the following files:
/usr/lib/libstdc++.so.6
/usr/lib/libstdc++.so.6.0.29
root@gw:~# opkg list | grep  libstdcpp
libstdcpp6 - 11.2.0-4 - GNU Standard C++ Library v3
```

Looks like zerotier want _libstdcpp7_, but only _libstdcpp6_ exists in the current stable branch 22.03.2.

cubercsl, 2023-01-19:
> Should be fixed for you, once this [build](https://buildbot.openwrt.org/openwrt-22.03/packages/#/builders/31/builds/249) finishes. I've just tested it on already finished x86/64 build:
> 
> ```shell
> $ docker run --rm -it openwrt/rootfs:x86-64-22.03.3 sh -c "mkdir -p /var/lock && opkg update && opkg install zerotier && ldd /usr/bin/zerotier-cli" 2>&1 | tail -5
> 	libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0x7fbf937a6000)
> 	libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0x7fbf937a1000)
> 	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7fbf9366e000)
> 	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fbf9365a000)
> 	libc.so => /lib/ld-musl-x86_64.so.1 (0x7fbf937b3000)
> ```
> 
> Can you check?

@ynezz After `opkg update && opkg install --force-reinstall zerotier`, it works fine for me.

Feel free to close this issue, or I'll close it until other people's issues are resolved too.

lhbdhr, 2023-01-19:
`opkg update && opkg install --force-reinstall zerotier`

Yesterday, I had the same problem as him.
I tried your code today and it worked.

Xiaomi AC2100,  ramips/mt7621, OpenWrt 22.03.2.

ynezz, 2023-01-19:
> Same here for rockchip/armv8 platform on NanoPi R4S.

@0anton that `opkg update && opkg install --force-reinstall zerotier` should fix it for you.

0anton, 2023-01-19:
> > Same here for rockchip/armv8 platform on NanoPi R4S.
> 
> @0anton that `opkg update && opkg install --force-reinstall zerotier` should fix it for you.

Confirmed. Package reinstall fixes the issue. Executable is again linked to libstd6:

```
root@gw:~# ldd /usr/bin/zerotier-cli
        /lib/ld-musl-aarch64.so.1 (0xffffbf490000)
        libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0xffffbf473000)
        libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0xffffbf460000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xffffbf2ce000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xffffbf2a9000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0xffffbf490000)
```

```
root@gw:~# opkg info zerotier
Package: zerotier
Version: 1.10.2-1
Depends: libc, libpthread, libstdcpp6, kmod-tun, ip, libminiupnpc, libnatpmp1
Status: install user installed
Section: net
Architecture: aarch64_generic
Size: 484655
Filename: zerotier_1.10.2-1_aarch64_generic.ipk
Conffiles:
 /etc/config/zerotier ca06cf1cd6d28a64ffddb1950505d8a9e7523cdd9be0f780e2955c69508005f5
Description: ZeroTier creates a global provider-independent virtual private cloud network.
Installed-Time: 1674120847
```

closed 2023-01-19 by cubercsl